iiab/iiab #433: Temporary failure resolving 'mirrordirector.raspbian.org'

holta, 2017-10-19:
This failure is new and extremely common over the past week or so.  Many different playbooks fail to download essential bits.  A long failure message is pasted in below, that happens to have occurred during [network : Install squid packages] but this download failure occurs in many other Ansible playbooks as well.

When this happens, "ping 8.8.8.8" works but "ping mit.edu" does not, confirming:
```
Temporary failure in name resolution
```
Possibly this flood of new downloads errors from the past few days is exacerbated by the "testing" version of dhcpcd5 that Raspbian has induced us to install in recent days, as a result of "apt upgrade" etc:
```
root@box:/opt/iiab/iiab# apt -a list dhcpcd5
Listing... Done
dhcpcd5/testing,now 1:6.11.5-1+rpt2 armhf [installed]
dhcpcd5/stable 6.10.1-1 armhf
```
```
root@box:/opt/iiab/iiab# apt -a list dhcpcd
Listing... Done
dhcpcd/stable 1:3.2.3-11 armhf
```
_Confusingly, installing IIAB over WiFi (experimentally, per #197) often yields the same error message despite different circumstances, i.e. when WiFi fails completely.  So when diagnosing, it's very important to test BOTH:_
```
ping 8.8.8.8
ping mit.edu
```
Finally, a typical error appears as follows:

> TASK [network : Install squid packages] ****************************************
failed: [127.0.0.1] (item=squid) => {"cache_update_time": 1508400494, "cache_updated": false, "changed": false, "failed": true, "item": "squid", "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"     install 'squid'' failed: E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/libe/libecap/libecap3_1.0.1-3_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'\nE: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/s/squid-langpack/squid-langpack_20150704-1_all.deb  Temporary failure resolving 'mirrordirector.raspbian.org'\nE: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/s/squid3/squid-common_3.5.23-5_all.deb  Temporary failure resolving 'mirrordirector.raspbian.org'\nE: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/s/squid3/squid_3.5.23-5_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'\nE: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?\n", "rc": 100, "stderr": "E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/libe/libecap/libecap3_1.0.1-3_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'\nE: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/s/squid-langpack/squid-langpack_20150704-1_all.deb  Temporary failure resolving 'mirrordirector.raspbian.org'\nE: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/s/squid3/squid-common_3.5.23-5_all.deb  Temporary failure resolving 'mirrordirector.raspbian.org'\nE: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/s/squid3/squid_3.5.23-5_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'\nE: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?\n", "stderr_lines": ["E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/libe/libecap/libecap3_1.0.1-3_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'", "E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/s/squid-langpack/squid-langpack_20150704-1_all.deb  Temporary failure resolving 'mirrordirector.raspbian.org'", "E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/s/squid3/squid-common_3.5.23-5_all.deb  Temporary failure resolving 'mirrordirector.raspbian.org'", "E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/s/squid3/squid_3.5.23-5_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'", "E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?"], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following additional packages will be installed:\n  libecap3 squid-common squid-langpack\nSuggested packages:\n  squidclient squid-cgi squid-purge resolvconf smbclient ufw winbindd\nThe following NEW packages will be installed:\n  libecap3 squid squid-common squid-langpack\n0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.\nNeed to get 2440 kB of archives.\nAfter this operation, 9420 kB of additional disk space will be used.\nErr:1 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libecap3 armhf 1.0.1-3\n  Temporary failure resolving 'mirrordirector.raspbian.org'\nErr:2 http://mirrordirector.raspbian.org/raspbian stretch/main armhf squid-langpack all 20150704-1\n  Temporary failure resolving 'mirrordirector.raspbian.org'\nErr:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf squid-common all 3.5.23-5\n  Temporary failure resolving 'mirrordirector.raspbian.org'\nErr:4 http://mirrordirector.raspbian.org/raspbian stretch/main armhf squid armhf 3.5.23-5\n  Temporary failure resolving 'mirrordirector.raspbian.org'\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "The following additional packages will be installed:", "  libecap3 squid-common squid-langpack", "Suggested packages:", "  squidclient squid-cgi squid-purge resolvconf smbclient ufw winbindd", "The following NEW packages will be installed:", "  libecap3 squid squid-common squid-langpack", "0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.", "Need to get 2440 kB of archives.", "After this operation, 9420 kB of additional disk space will be used.", "Err:1 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libecap3 armhf 1.0.1-3", "  Temporary failure resolving 'mirrordirector.raspbian.org'", "Err:2 http://mirrordirector.raspbian.org/raspbian stretch/main armhf squid-langpack all 20150704-1", "  Temporary failure resolving 'mirrordirector.raspbian.org'", "Err:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf squid-common all 3.5.23-5", "  Temporary failure resolving 'mirrordirector.raspbian.org'", "Err:4 http://mirrordirector.raspbian.org/raspbian stretch/main armhf squid armhf 3.5.23-5", "  Temporary failure resolving 'mirrordirector.raspbian.org'"]}
failed: [127.0.0.1] (item=cadaver) => {"cache_update_time": 1508400494, "cache_updated": false, "changed": false, "failed": true, "item": "cadaver", "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"     install 'cadaver'' failed: E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/n/neon27/libneon27-gnutls_0.30.2-2_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'\nE: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/c/cadaver/cadaver_0.23.3-2+b2_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'\nE: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?\n", "rc": 100, "stderr": "E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/n/neon27/libneon27-gnutls_0.30.2-2_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'\nE: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/c/cadaver/cadaver_0.23.3-2+b2_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'\nE: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?\n", "stderr_lines": ["E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/n/neon27/libneon27-gnutls_0.30.2-2_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'", "E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/c/cadaver/cadaver_0.23.3-2+b2_armhf.deb  Temporary failure resolving 'mirrordirector.raspbian.org'", "E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?"], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following additional packages will be installed:\n  libneon27-gnutls\nThe following NEW packages will be installed:\n  cadaver libneon27-gnutls\n0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.\nNeed to get 203 kB of archives.\nAfter this operation, 438 kB of additional disk space will be used.\nErr:1 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libneo [... cut ...] "0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.", "Need to get 203 kB of archives.", "After this operation, 438 kB of additional disk space will be used.", "Err:1 http://mirrordirector.raspbian.org/raspbian stretch/main armhf libneon27-gnutls armhf 0.30.2-2", "  Temporary failure resolving 'mirrordirector.raspbian.org'", "Err:2 http://mirrordirector.raspbian.org/raspbian stretch/main armhf cadaver armhf 0.23.3-2+b2", "  Temporary failure resolving 'mirrordirector.raspbian.org'"]}

jvonau, 2017-10-19:
root@raspberrypi:/opt/iiab/iiab# ls -la /lib/systemd/system/dhcpcd.service
-rw-r--r-- 1 root root 260 Jun 30 12:44 /lib/systemd/system/dhcpcd.service
root@raspberrypi:/opt/iiab/iiab# cat /lib/systemd/system/dhcpcd.service
[Unit]
Description=dhcpcd on all interfaces
Wants=network.target
Before=network.target

[Service]
Type=forking
PIDFile=/run/dhcpcd.pid
ExecStart=/usr/lib/dhcpcd5/dhcpcd -q -b
ExecStop=/sbin/dhcpcd -x

[Install]
WantedBy=multi-user.target
Alias=dhcpcd5.service
root@raspberrypi:/opt/iiab/iiab# 

root@raspberrypi:/opt/iiab/iiab# ls -la /usr/lib/dhcpcd5/dhcpcd 
-rwxr-xr-x 1 root root 343 Jul 17 10:13 /usr/lib/dhcpcd5/dhcpcd
root@raspberrypi:/opt/iiab/iiab# cat /usr/lib/dhcpcd5/dhcpcd 
#!/bin/sh -e

DHCPCD=/sbin/dhcpcd
INTERFACES=/etc/network/interfaces

if grep -q -E "^[[:space:]]*iface[[:space:]]*.*[[:space:]]*inet[[:space:]]*(dhcp|static)" \
		$INTERFACES; then
	echo "Not running dhcpcd because $INTERFACES"
	echo "defines some interfaces that will use a"
	echo "DHCP client or static address"
	exit 6
fi

exec $DHCPCD $@

root@raspberrypi:/opt/iiab/iiab# ls -la /etc/alternatives/dhcpcd
lrwxrwxrwx 1 root root 13 Sep  7 11:13 /etc/alternatives/dhcpcd -> /sbin/dhcpcd5
binary

root@raspberrypi:/opt/iiab/iiab# ls -la /sbin/dhcpcd
lrwxrwxrwx 1 root root 24 Sep  7 11:13 /sbin/dhcpcd -> /etc/alternatives/dhcpcd
binary

root@raspberrypi:/opt/iiab/iiab# ls -la /sbin/dhcpcd5
-rwxr-xr-x 1 root root 239508 Sep 20 10:18 /sbin/dhcpcd5

holta, 2017-10-21:
New info has come to light: many/most (all?) installs of [BIG-sized local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars_big.yml) fail at Ansible role **[network : Install squid packages]** just as pasted in above.

holta, 2017-10-21:
FYI #433 here is possibly caused by things like named / bind9 &mdash; @jvonau is investigating if the underlying OS [Raspbian] has changed, and may now be shipping with nameserver enabled for 'secure DNS' out of the box?

holta, 2017-10-21:
Solving this might be as straightforward as commenting out Lines 22-24 from roles/network/tasks/named.yml per @jvonau's PR #267 https://github.com/iiab/iiab/pull/267/commits/be739b976468df91ab37a225e6618be5d0241c45
```
# or we have to change the serial number in the config files.
#- name: Stop named before copying files
#  service: name={{ dns_service }} state=stopped
#  when: not installing
```
Really great news if so!  If as @jvonau suspects "secureDNS was configured and enabled out of the box with bind9 and integrated with resolvconf (the package) in the latest raspbian packages so stopping dns server is not a good idea" ?

holta, 2017-10-21:
Confirmed: @jvonau will spin up a PR commenting out the [above 3 lines](https://github.com/iiab/iiab/commit/be739b976468df91ab37a225e6618be5d0241c45)[*] and possibly more.

[*] that I him thoroughly test, repeatedly today

commit 89e3461e (2017-10-22): Merge pull request #442 from jvonau/433

named / DNS fix for #433 Squid install fail

closes #433
closed 2017-10-22 by holta