iiab/iiab #437: [RFC][TravisCI] Testing Ansible Releases

arky, 2017-10-20:
We can now test various release of Ansible ( https://pypi.python.org/pypi/ansible ) using Travis CI.  The goal of this feature enhancement is to ensure that IIAB code works with any latest release of Ansible without breakages. 

In future, we could distribute IIAB through [ansible-galaxy](http://docs.ansible.com/ansible/latest/galaxy.html). This test coverage will bring us closer to that goal.  

Current versions of Ansible on pypi are as follows:  1.0, 1.1, 1.2, 1.2.1, 1.2.2, 1.2.3, 1.3.0, 1.3.1, 1.3.2, 1.3.3, 1.3.4, 1.4, 1.4.1, 1.4.2, 1.4.3, 1.4.4, 1.4.5, 1.5, 1.5.1, 1.5.2, 1.5.3, 1.5.4, 1.5.5, 1.6, 1.6.1, 1.6.2, 1.6.3, 1.6.4, 1.6.5, 1.6.6, 1.6.7, 1.6.8, 1.6.9, 1.6.10, 1.7, 1.7.1, 1.7.2, 1.8, 1.8.1, 1.8.2, 1.8.3, 1.8.4, 1.9.0.1, 1.9.1, 1.9.2, 1.9.3, 1.9.4, 1.9.5, 1.9.6, 2.0.0.0, 2.0.0.1, 2.0.0.2, 2.0.1.0, 2.0.2.0, 2.1.0.0, 2.1.1.0, 2.1.2.0, 2.1.3.0, 2.1.4.0, 2.1.5.0, 2.1.6.0, 2.2.0.0, 2.2.1.0, 2.2.2.0, 2.2.3.0, 2.3.0.0, 2.3.1.0, 2.3.2.0, 2.4.0.0

Please let me know which Ansible versions you would like to automatically test/build IIAB code.

holta, 2017-10-20:
Ansible 2.4.1 hah (it seems it's still a few days away from release, per https://releases.ansible.com/ansible/ :-)

arky, 2017-10-20:
@holta That would be ansible:latest then in few days.

jvonau, 2017-10-21:
Target 2.4.0 and bump the value when upstream releases 2.4.1 or just leave that at 2.4.0 for the time being till the noise calms down in travis and is not so overwhelming so the finer difference between 2.4.0 and 2.4.1 are more apparent.

holta, 2017-10-21:
[Lines 61-91 of scripts/ansible](https://github.com/iiab/iiab/blob/master/scripts/ansible#L61-L91) currently install the highest-numbered (might not be the latest!) Ansible PPA for new IIAB installs on Raspbian/Ubuntu/Debian, be it Ansible 2.4.x or 2.5.x or 3.x.x

Presumably 3.x.x would get installed even if 2.4.9 was released a week later etc?  Anyway that's many moons way, so we can pin to 2.4.x later if it becomes wise/nec/etc.

Meantime 5 weeks have passed since 2.4.0, and it's proved wonderfully reliable.  Hopefully 2.4.1's release notes (in coming days presumably) will elaborate what it offers, and if it's faster than 2.4.0 by some miracle etc :)

holta, 2017-11-21:
FYI Ansible 2.4.2 will probably arrive right _after_ IIAB 6.5 is finalized in coming days...

holta, 2018-06-19:
> Please let me know which Ansible versions you would like to automatically test/build IIAB code.

Ansible 2.5.5 and 2.6.0 RC3, if possible!

arky, 2018-07-01:
Disabling Travis CI as requested by the @holta  Closing this bug as invalid.

closed 2018-07-01 by arky