iiab/iiab #438: no Internet (Ethernet or WiFi) after WiFi installs

holta, 2017-10-20:
WiFi installations of IIAB 6.5 do complete!  But after rebooting, Internet access is lost (via Ethernet and WiFi both).

See details @ #432 where /etc/network/interfaces.d/iiab is pasted in.

Here's one interim workaround to restore Ethernet access to the Internet:
```
rm /etc/network/interfaces.d/iiab
reboot
```
FYI this does not restore WiFi access to the Internet.  But perhaps that part is intentional?  i.e. so the IIAB's Hotspot/AP works &mdash; until we later make progress with #430 (simultaneous hotspot + client-WiFi) during the [IIAB 6.6](https://github.com/iiab/iiab/milestone/3) release cycle.

holta, 2017-10-20:
Related if indirectly:

#424 RFC on moving network last.
#433 Temporary failure resolving 'mirrordirector.raspbian.org'

jvonau, 2017-10-21:
extend [current code](https://github.com/iiab/iiab/blob/master/roles/1-prep/templates/iiab-network-reset)

holta, 2017-11-22:
@jvonau can we close this?  (Or better to test further?)

jvonau, 2017-11-22:
You'll have at least wired internet seemlessly

holta, 2017-11-22:
> You'll have at least wired internet seemlessly

Great.

(As you know my personal vote is for WiFi-to-Internet to hold on until all content & glitches are resolved to the satisfaction of the implementer, but we'll try to work streamline workflows there in due course)

closed 2017-11-22 by holta