iiab/iiab #439: Hotspot/AP not working after WiFi install of IIAB onto RPi3

holta, 2017-10-21:
Installing IIAB via WiFi completes!  But these RPi3's internal WiFi/AP/hotspots do not work unfortunately &mdash; even after reboot.

Apparently because /etc/hostapd/hostapd.conf is not created during ./runansible?  And certainly regardless if I delete /etc/network/interfaces.d/iiab and regardless if I reboot with an Ethernet cable inserted, hostapd.conf does not appear (so far!)

In contrast: when installing a default IIAB onto RPi3 via Ethernet, /etc/hostapd/hostapd.conf appears as follows:
```
# Basic configuration

interface=wlan0
ssid=Internet in a Box
channel=6
bridge=br0

# Hardware configuration
driver=nl80211
country_code=US
# limit emissions to what is legal in country
ieee80211d=1
hw_mode=g
```
Related, if indirectly:
#424 RFC on moving network last
#438 no Internet (Ethernet or WiFi) after WiFi installs

holta, 2017-10-21:
FYI @jvonau has made progress here while working on #433 &mdash; if all goes well, a PR may be available for testing soon in coming days.

holta, 2017-10-21:
Oddly a reboot was required to get the new SSID to show up on the air.

In the past editing /etc/hostapd/hostapd.conf then "systemctl restart hostapd" was sufficient.

But this time, now even "systemctl daemon-reload" following by "systemctl restart hostapd.service" worked &mdash; until reboot.

Rome Wasn't Built In A Day...and until then rebooting is a fine workaround.

jvonau, 2017-10-21:
The auto switch over to AP mode is future, maybe.  What should work is wifi gateway during install, and wifi gateway post-install with a reboot, and the wired interface becoming part of br0, so the LAN is extendible via the wired port. The caveat being the wired interface would no longer be available to use as a gateway. with the workaround in #438 being needed, hence the suggestion to extend existing code to handle that situation.

holta, 2017-10-22:
@jvonau should we be concerned about the following error/warning that occurs when installing IIAB via WiFi?
```
TASK [network : Restart hostapd if slave is inactive] **************************
fatal: [127.0.0.1]: FAILED! => {"failed": true, "msg": "'dict object' has no attribute 'stdout'"}
...ignoring
```

jvonau, 2017-10-22:
No issue, can't use wifi as client and use hostapd at the same time, the
check is better limited to when needed in PR #267

holta, 2017-10-25:
In ~2015 @georgejhunt implemented xs-hotspot-on and xs-hotspot-off which worked extremely well.

Let's strongly consider following this pattern, implementing these commands to toggle the functionality of IIAB's internal hotspot:
```
iiab-hotspot-on
iiab-hotspot-off
```
_Idea: this needs to be communicated by telephone to non-technical operators.  So telling them things like "cd /opt/iiab/iiab; ./runtags hotspot-off" is a recipe for frustration for literally everyone.  What's needed is a CLI toggle that is short+sweet.  And then later the Admin Console can build on this functionality in more sophisticated ways, once we understand the actual implementer/operator/users use cases better._

jvonau, 2017-10-25:
Other code such as iiab-remote-{on-off} follows a similar format and is
easy to use for the end user. Any reason that the code from 2015 that
implements this functionality is not part of the current code base?

holta, 2017-10-25:
> Other code such as iiab-remote-{on-off} follows a similar format and is easy to use for the end user. Any reason that the code from 2015 that implements this functionality is not part of the current code base?

If I recall correctly, @georgejhunt implemented these for XSCE on XO-1.5 and XO-4.  I wish `iiab-remote-on` and `iiab-remote-off` had become part of the larger code base, but ah well for whatever historical reason they did not.

holta, 2017-10-25:
FYI the older `xs-hotspot-on` and `xs-hotspot-off` are documented in http://FAQ.IIAB.IO under Question #13 but let's revise that as soon as we reach a new consensus/implementation here :)

holta, 2017-11-25:
Can we close this or not yet?

(Whether or not #431 -> #543 can offer an even easier install path for implementers that do not have access to Ethernet!)

holta, 2017-12-07:
Prelim Documentation for iiab-hotspot-on and iiab-hotspot-off added to https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch Step 3.

closed 2017-12-07 by holta