iiab/iiab #440: Adding content to kiwix fail on iiab(master) and debian 9.2

germanrs, 2017-10-21:
This issue is on iiab (master branch) and Debian 9.2,I was testing kiwix, but the process of add content failed with some files.

Here is the procees I followed:

**_Copy zim files from my hard drive_**
`
root@box:/media/usb0/iiab# cp -v *.zim /library/zims/content/
'gutenberg_es_all_10_2014.zim' -> '/library/zims/content/gutenberg_es_all_10_2014.zim'
'phet_es_2017-09.zim' -> '/library/zims/content/phet_es_2017-09.zim'
'ted_en_science_2016-06.zim' -> '/library/zims/content/ted_en_science_2016-06.zim'
'vikidia_es_all_2017-08.zim' -> '/library/zims/content/vikidia_es_all_2017-08.zim'
'wikibooks_es_all_2017-02.zim' -> '/library/zims/content/wikibooks_es_all_2017-02.zim'
'wikinews_es_all_2017-04.zim' -> '/library/zims/content/wikinews_es_all_2017-04.zim'
'wikipedia_es_all_2017-05.zim' -> '/library/zims/content/wikipedia_es_all_2017-05.zim'
'wikipedia_es_medicine_2017-08.zim' -> '/library/zims/content/wikipedia_es_medicine_2017-08.zim'
'wikiquote_es_all_2017-03.zim' -> '/library/zims/content/wikiquote_es_all_2017-03.zim'
'wikisource_es_all_2017-02.zim' -> '/library/zims/content/wikisource_es_all_2017-02.zim'
'wikiversity_es_all_2017-03.zim' -> '/library/zims/content/wikiversity_es_all_2017-03.zim'
'wikivoyage_es_all_2017-03.zim' -> '/library/zims/content/wikivoyage_es_all_2017-03.zim'
'wiktionary_es_all_2016-12.zim' -> '/library/zims/content/wiktionary_es_all_2016-12.zim'
`

**_Regenerate library.xml file_**
`
root@box:/media/usb0/iiab# iiab-make-kiwix-lib
terminate called after throwing an instance of 'int'
skipping phet_es_2017-09
terminate called after throwing an instance of 'int'
skipping vikidia_es_all_2017-08
terminate called after throwing an instance of 'int'
skipping wikibooks_es_all_2017-02
terminate called after throwing an instance of 'int'
skipping wikinews_es_all_2017-04
terminate called after throwing an instance of 'int'
skipping wikipedia_es_all_2017-05
terminate called after throwing an instance of 'int'
skipping wikipedia_es_medicine_2017-08
terminate called after throwing an instance of 'int'
skipping wikiquote_es_all_2017-03
terminate called after throwing an instance of 'int'
skipping wikisource_es_all_2017-02
terminate called after throwing an instance of 'int'
skipping wikiversity_es_all_2017-03
terminate called after throwing an instance of 'int'
skipping wikivoyage_es_all_2017-03
root@box:/media/usb0/iiab#
`
[image]

[image]

**_Here is the output of systemctl status kiwix-serve_**
`root@box:/media/usb0/iiab# systemctl status kiwix-serve
● kiwix-serve.service - Provides the Kiwix Zim File Server
   Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2017-10-21 11:16:57 EDT; 3s ago
  Process: 4218 ExecStart=/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml (code=exited, status=0/SUCCESS)
 Main PID: 4219 (kiwix-serve)
    Tasks: 5 (limit: 4915)
   CGroup: /system.slice/kiwix-serve.service
           └─4219 /opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml

Oct 21 11:16:57 box.olpc systemd[1]: Starting Provides the Kiwix Zim File Server...
Oct 21 11:16:57 box.olpc systemd[1]: Started Provides the Kiwix Zim File Server.
root@box:/media/usb0/iiab# systemctl status kiwix-serve
● kiwix-serve.service - Provides the Kiwix Zim File Server
   Loaded: loaded (/etc/systemd/system/kiwix-serve.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Sat 2017-10-21 11:17:05 EDT; 7s ago
  Process: 4218 ExecStart=/opt/iiab/kiwix/bin/kiwix-serve --daemon --port 3000 --nolibrarybutton --library /library/zims/library.xml (code=exited, status=0/SUCCESS)
 Main PID: 4219 (code=killed, signal=ABRT)

Oct 21 11:16:57 box.olpc systemd[1]: Starting Provides the Kiwix Zim File Server...
Oct 21 11:16:57 box.olpc systemd[1]: Started Provides the Kiwix Zim File Server.
Oct 21 11:17:05 box.olpc kiwix-serve[4218]: terminate called after throwing an instance of 'int'
Oct 21 11:17:05 box.olpc systemd[1]: kiwix-serve.service: Main process exited, code=killed, status=6/ABRT
Oct 21 11:17:05 box.olpc systemd[1]: kiwix-serve.service: Unit entered failed state.
Oct 21 11:17:05 box.olpc systemd[1]: kiwix-serve.service: Failed with result 'signal'.
root@box:/media/usb0/iiab#
`
This process was working fine on Ubuntu 16.04 and IIAB -6.4

holta, 2017-10-21:
@tim-moody do you have time to look before Monday's community/team call?  (http://minutes.iiab.io)

tim-moody, 2017-10-21:
on Linux iiab-demo.iiab.me 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u3 (2017-08-06) x86_64 GNU/Linux

wget http://download.kiwix.org/zim/wikibooks/wikibooks_es_all_2017-02.zim

iiab-make-kiwix-lib

no complaints

tim-moody, 2017-10-21:
wget http://download.kiwix.org/zim/phet/phet_es_2017-09.zim

iiab-make-kiwix-lib

no complaints

holta, 2017-10-21:
Clarif: the server Tim mentions above is using kiwix-tools_linux64_2017-09-25.tar.gz (similar to IIAB 6.4) rather than the http://download.kiwix.org/nightly/2017-10-11/ version of kiwix-tools.

germanrs, 2017-10-22:
I found something strange with my zim files.

**_Test 1:_
Ubuntu 16.04 
IIAB-6.4
phet from hard drive**
`root@box:/library/zims/content# cp /media/usb0/iiab/phet_es_2017-09.zim .
root@box:/library/zims/content# iiab-make-kiwix-lib
root@box:/library/zims/content# md5sum phet_es_2017-09.zim
c9cddb7df6d483610c8894016d4c0fb1  phet_es_2017-09.zim
root@box:/library/zims/content#`
http://10.0.0.10:3000/phet_es_2017-09/
Work fine

**_Test 2:_
Debian 9.2
IIAB-master
phet from HDD (same on ubuntu)**
`root@box:/library/zims/content# cp /media/usb0/iiab/phet_es_2017-09.zim .
root@box:/library/zims/content# iiab-make-kiwix-lib
terminate called after throwing an instance of 'int'
skipping phet_es_2017-09
root@box:/library/zims/content# md5sum phet_es_2017-09.zim
48da957228dabb73b616f2120f4ad5d6  phet_es_2017-09.zim`

**_Test 3:_
Debian 9.2
IIAB-master
phet from internet**
`root@box:/library/zims/content# wget http://download.kiwix.org/zim/phet/phet_es_2017-09.zim
--2017-10-21 22:03:58--  http://download.kiwix.org/zim/phet/phet_es_2017-09.zim
Resolving download.kiwix.org (download.kiwix.org)... 62.210.143.55
Connecting to download.kiwix.org (download.kiwix.org)|62.210.143.55|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://ftp.acc.umu.se/mirror/kiwix.org/zim/phet/phet_es_2017-09.zim [following]
--2017-10-21 22:03:59--  https://ftp.acc.umu.se/mirror/kiwix.org/zim/phet/phet_es_2017-09.zim
Resolving ftp.acc.umu.se (ftp.acc.umu.se)... 194.71.11.165, 194.71.11.173, 2001:6b0:19::165, ...
Connecting to ftp.acc.umu.se (ftp.acc.umu.se)|194.71.11.165|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://saimei.ftp.acc.umu.se/mirror/kiwix.org/zim/phet/phet_es_2017-09.zim [following]
--2017-10-21 22:03:59--  https://saimei.ftp.acc.umu.se/mirror/kiwix.org/zim/phet/phet_es_2017-09.zim
Resolving saimei.ftp.acc.umu.se (saimei.ftp.acc.umu.se)... 194.71.11.138, 2001:6b0:19::138
Connecting to saimei.ftp.acc.umu.se (saimei.ftp.acc.umu.se)|194.71.11.138|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 27653132 (26M)
Saving to: ‘phet_es_2017-09.zim’

phet_es_2017-09.zim                         100%[=========================================================================================>]  26.37M   827KB/s    in 42s

2017-10-21 22:04:42 (645 KB/s) - ‘phet_es_2017-09.zim’ saved [27653132/27653132]

root@box:/library/zims/content# md5sum phet_es_2017-09.zim
c9cddb7df6d483610c8894016d4c0fb1  phet_es_2017-09.zim

root@box:/library/zims/content# iiab-make-kiwix-lib
root@box:/library/zims/content#`

md5sum for phet changes on Debian, using phet from internet works, also the md5sum is the same on Ubuntu

germanrs, 2017-10-22:
Same ka-lite zip file, first one is on my iiab server, copied from my USB HDD
`root@box:/home/rsc# md5sum ka-lite-0.17-resized-videos-spanish.zip
8945a96ea61669c73bff935e255ebd47  ka-lite-0.17-resized-videos-spanish.zip`

Second is from USB HDD
`root@box:/home/rsc# md5sum /media/usb0/iiab/ka-lite-0.17-resized-videos-spanish.zip
7c4da4805f9fbf751d4bcc45ebad5ad7  /media/usb0/iiab/ka-lite-0.17-resized-videos-spanish.zip`

As you can see md5sum is different :/

Tomorrow, I will re-download Debian iso file.

germanrs, 2017-10-22:
I hit the problem, installing **`ntfs-3g/stable`** fix the problem, now i can copy and use my zims files from my ntfs USB HDD.

I guess i can close this bug or open another one to add ntfs support for debian.

holta, 2017-10-22:
@germanrs keep a close eye on #395 in the coming week as well, until this gets merged with master for [IIAB 6.5 Preview 2](https://github.com/iiab/iiab/releases/) in the coming week we hope!

Specifically: don't hesitate to grab/test the latest nightly (e.g. http://download.kiwix.org/nightly/2017-10-20/kiwix-tools_linux64_2017-10-20.tar.gz for the moment) and...
1. save that to /opt/iiab/download
2. unpack that into /opt/iiab/kiwix/bin
3. run /usr/bin/iiab-make-kiwix-lib
4. _restest all your ZIM's at http://box:3000_

PS if you have older/legacy ZIM's (those chunked into 2GB pieces) these can be concatenated into 1 single file (i.e. do "cat old.zimaa old.zimab old.zimac old.zimad > new.zim") that still reside in /library/zims/content, remembering to get rid of your old ZIM files when you're done.

PPS note also that the directory /library/zims/index will be increasingly unused as newer ZIM files (should, increasingly in 2017) all contain the full-text search index within the ZIM itself.

tim-moody, 2017-10-22:
also keep in mind that not all zims have a built in index yet, so by copying only the zim file you may not have a full text index.

holta, 2017-10-24:
@germanrs is this #440 under control / resolvable?

holta, 2017-10-25:
Resolving for now — plz re-open as nec.

closed 2017-10-25 by holta