iiab/iiab #446: WiFi-installed IIAB on Raspbian Desktop takes 40-95sec to shut down

holta, 2017-10-22:
In all cases, this appeared on the screen during these slow shutdowns:
```
plymouth-reboot.service
```
On the bright side, Raspbian Desktop still boots (to its LXDE-pi screen) almost as fast as Raspbian Lite.  See these 4 attempts of each kind of OS &mdash; with 2 attempts in each configuration:

WiFi-installed [BIG-sized](http://wiki.laptop.org/go/IIAB/local_vars_big.yml) IIAB on Raspbian Desktop:
- 95 sec / 95 sec - shutdown
- 30 sec / 47 sec - boot to LXDE-pi

WiFi-installed [BIG-sized](http://wiki.laptop.org/go/IIAB/local_vars_big.yml) IIAB on Raspbian Desktop AFTER deletiong of /etc/network/interfaces.d/iiab:
- 40 sec / 39 sec - shutdown
- 36 sec / 41 sec - boot to LXDE-pi

<hr>

Ethernet-install [MIN-sized](http://wiki.laptop.org/go/IIAB/local_vars_min.yml) IIAB on Raspbian Lite:
- 8 sec / 7 sec - shutdown
- 19 sec / 23 sec - boot to login prompt

Ethernet-install [MEDIUM-sized](http://wiki.laptop.org/go/IIAB/local_vars.yml) IIAB on Raspbian Lite:
- 6 sec / 9 sec - shutdown
- 24 sec / 22 sec - boot to login prompt

_Corollary: adding Calibre, Elgg, Nextcloud, OSM, Sugarizer, WordPress (when moving from MIN-sized to MEDIUM-sized local_vars.yml) did not materially change shutdown and bootup times!_

holta, 2017-10-22:
Possibly Related:
#438 no Internet (Ethernet or WiFi) after WiFi installs
#439 Hotspot/AP not working after WiFi install of IIAB onto RPi3

holta, 2017-10-22:
I'll keep this open as a QA placeholder/reminder but good news: @jvonau clarifies this (#446) is not a concern, as these slow shutdown issues will likely disappear when #438 and #439 are solved in coming weeks.

holta, 2017-10-23:
@georgejhunt / @jvonau remind us during today's community/team call (http://minutes.iiab.io) that MySQL, PostgreSQL Squid can also cause this!

jvonau, 2017-11-10:
I'll go with squid as the source of the issue as there is a noted slowness shutting down squid to relace files during the install routine.

jvonau, 2017-11-12:
Any new reports with the revised networking?

holta, 2017-11-23:
> Any new reports with the revised networking?

No.  If this re-occurs we can re-open if nec-

jvonau, 2017-11-26:
[seen elsewhere](https://www.mail-archive.com/debian-bugs-dist@lists.debian.org/msg1558061.html)

holta, 2017-11-27:
> [seen elsewhere](https://www.mail-archive.com/debian-bugs-dist@lists.debian.org/msg1558061.html)

Above refers to "Bug#879639: squid takes 30 seconds to restart, should be < 1 second"

Am re-opening this just in case: should it be part of IIAB 6.5 or 6.6?

holta, 2018-06-12:
This tkt is a great reference but let's focus on solving this on lead tkt #822

closed 2018-06-12 by holta