iiab/iiab #448: Internet-in-a-box is not installed on Debian 9.2.1, even if osm is True in config file

germanrs, 2017-10-22:
This is the config present in `vars/local_vars`
`# OpenStreetMap: renamed from {iiab_install, iiab_enabled} in June 2017`
`osm_install: True`
`osm_enabled: True`

I've copied my OSM data from my local hdd with `cp -rv /media/knowledge/modules /library/knowledge`

Accesing http://10.0.0.3/osm/ doesn't show any map, here is the error from `/var/log/apache2/error.log`

[Sun Oct 22 15:09:11.533053 2017] [wsgi:error] [pid 19099] [client 10.0.0.6:54097] Traceback (most recent call last):, referer: http://10.0.0.3/maps/
[Sun Oct 22 15:09:11.533067 2017] [wsgi:error] [pid 19099] [client 10.0.0.6:54097]   File "/library/www/html/osm.wsgi", line 19, in <module>, referer: http://10.0.0.3/maps/
[Sun Oct 22 15:09:11.533086 2017] [wsgi:error] [pid 19099] [client 10.0.0.6:54097]     import iiab, referer: http://10.0.0.3/maps/
[Sun Oct 22 15:09:11.533100 2017] [wsgi:error] [pid 19099] [client 10.0.0.6:54097] ImportError: No module named iiab, referer: http://10.0.0.3/maps/
[Sun Oct 22 15:09:11.865133 2017] [wsgi:error] [pid 19103] [client 10.0.0.6:54104] mod_wsgi (pid=19103): Target WSGI script '/library/www/html/osm.wsgi' cannot be loaded as P
ython module., referer: http://10.0.0.3/maps/
[Sun Oct 22 15:09:11.865167 2017] [wsgi:error] [pid 19103] [client 10.0.0.6:54104] mod_wsgi (pid=19103): Exception occurred processing WSGI script '/library/www/html/osm.wsgi
'., referer: http://10.0.0.3/maps/

`pip install Internet-in-a-box ` fix the issue

Accesing http://10.0.0.3/osm/ and http://10.0.0.3/maps both works fine, there is a worldmap in there.

holta, 2017-10-22:
Is [roles/osm/tasks/main.yml lines 24-42](https://github.com/iiab/iiab/blob/master/roles/osm/tasks/main.yml#L24-L42) to blame, possibly relating to --no-cache-dir or some such?

germanrs, 2017-10-22:
This also affects Ubuntu 16.04 LTS

[Sun Oct 22 17:06:49.039341 2017] [wsgi:error] [pid 11798] [client 10.0.0.6:55392] mod_wsgi (pid=11798): Target WSGI script '/library/www/html/osm.wsgi' cannot be loaded as Python module., referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.039341 2017] [wsgi:error] [pid 11795] [client 10.0.0.6:55393] mod_wsgi (pid=11795): Target WSGI script '/library/www/html/osm.wsgi' cannot be loaded as Python module., referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.039374 2017] [wsgi:error] [pid 11798] [client 10.0.0.6:55392] mod_wsgi (pid=11798): Exception occurred processing WSGI script '/library/www/html/osm.wsgi'., referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.039378 2017] [wsgi:error] [pid 11795] [client 10.0.0.6:55393] mod_wsgi (pid=11795): Exception occurred processing WSGI script '/library/www/html/osm.wsgi'., referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.039400 2017] [wsgi:error] [pid 11795] [client 10.0.0.6:55393] Traceback (most recent call last):, referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.039401 2017] [wsgi:error] [pid 11798] [client 10.0.0.6:55392] Traceback (most recent call last):, referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.039426 2017] [wsgi:error] [pid 11798] [client 10.0.0.6:55392]   File "/library/www/html/osm.wsgi", line 19, in <module>, referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.039426 2017] [wsgi:error] [pid 11795] [client 10.0.0.6:55393]   File "/library/www/html/osm.wsgi", line 19, in <module>, referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.039463 2017] [wsgi:error] [pid 11798] [client 10.0.0.6:55392]     import iiab, referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.039463 2017] [wsgi:error] [pid 11795] [client 10.0.0.6:55393]     import iiab, referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.039488 2017] [wsgi:error] [pid 11798] [client 10.0.0.6:55392] ImportError: No module named iiab, referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.039488 2017] [wsgi:error] [pid 11795] [client 10.0.0.6:55393] ImportError: No module named iiab, referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.050972 2017] [wsgi:error] [pid 11800] [client 10.0.0.6:55394] mod_wsgi (pid=11800): Target WSGI script '/library/www/html/osm.wsgi' cannot be loaded as Python module., referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.050993 2017] [wsgi:error] [pid 11800] [client 10.0.0.6:55394] mod_wsgi (pid=11800): Exception occurred processing WSGI script '/library/www/html/osm.wsgi'., referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.051013 2017] [wsgi:error] [pid 11800] [client 10.0.0.6:55394] Traceback (most recent call last):, referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.051032 2017] [wsgi:error] [pid 11800] [client 10.0.0.6:55394]   File "/library/www/html/osm.wsgi", line 19, in <module>, referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.051069 2017] [wsgi:error] [pid 11800] [client 10.0.0.6:55394]     import iiab, referer: http://10.0.0.8/osm/
[Sun Oct 22 17:06:49.051089 2017] [wsgi:error] [pid 11800] [client 10.0.0.6:55394] ImportError: No module named iiab, referer: http://10.0.0.8/osm/

holta, 2017-10-23:
Discussed during our community/team call (http://minutes.iiab.io) ...let's all plz try to get to the bottom of this in coming days, Thanks!

jvonau, 2017-10-23:
> cat /library/www/html/osm.wsgi
 WSGI interface file for Apache's mod_wsgi
 If you are using a virtualenv, set this pathpath_to_virtualenv = None      << overlooked   osm.wsgi is provided by iiab's pip install`

The virtual environment path should be "/usr/local/osm"

holta, 2017-10-24:
Resolved by #453

closed 2017-10-24 by holta