iiab/iiab #449: "ignore_errors" takes precedence over "include when" in ansible 2.4

georgejhunt, 2017-10-23:
The OS is ubuntu, with current master (10/22), running ./runansible. The debug statement (in the following log) was inserted at line 63 of roles/1-prep/tasks/detected_network.yml, RIGHT BEFORE the "include detect_redhat.yml" (line 64 in current master), (which is supposed to be skipped). 
Nothing in the detect_redhat.yml should have any way to affect the outcome, since it should only be included when the "is_redhat" is true. But apparently the "ignore_errors" directive causes a clause in the "skipped" include, to error out.

I think it is an ansible bug.

Moving to ansible 2.4 happened over my objection.

2017-10-22 17:34:13,723 p=6759 u=root |  TASK [1-prep : debug] **********************************************************
2017-10-22 17:34:13,744 p=6759 u=root |  ok: [127.0.0.1] => {
    "is_redhat": false
}
2017-10-22 17:34:13,760 p=6759 u=root |  TASK [1-prep : pause] **********************************************************
2017-10-22 17:34:13,779 p=6759 u=root |  [1-prep : pause]
Press enter to continue:
2017-10-22 17:34:31,034 p=6759 u=root |  ok: [127.0.0.1]
2017-10-22 17:34:31,071 p=6759 u=root |  TASK [1-prep : Checking for ifcfg-WAN file - Can Fail] *************************
2017-10-22 17:34:31,085 p=6759 u=root |  skipping: [127.0.0.1]
2017-10-22 17:34:31,100 p=6759 u=root |  TASK [1-prep : Setting ifcfg-WAN True] *****************************************
2017-10-22 17:34:31,114 p=6759 u=root |  skipping: [127.0.0.1]
2017-10-22 17:34:31,129 p=6759 u=root |  TASK [1-prep : Get a list of slaves from previous config - Can Fail] ***********
2017-10-22 17:34:31,143 p=6759 u=root |  skipping: [127.0.0.1]
2017-10-22 17:34:31,158 p=6759 u=root |  TASK [1-prep : Find gateway config based on device] ****************************
2017-10-22 17:34:31,172 p=6759 u=root |  skipping: [127.0.0.1]
2017-10-22 17:34:31,186 p=6759 u=root |  TASK [1-prep : Setting has ifcfg gw based on device if found] ******************
2017-10-22 17:34:31,198 p=6759 u=root |  fatal: [127.0.0.1]: FAILED! => {"failed": true, "msg": "'dict object' has no attribute 'stdout_lines'"}
2017-10-22 17:34:31,199 p=6759 u=root |  ...ignoring
2017-10-22 17:34:31,213 p=6759 u=root |  TASK [1-prep : Find active gateway config based on macaddress - Can Fail] ******
2017-10-22 17:34:31,227 p=6759 u=root |  skipping: [127.0.0.1]
2017-10-22 17:34:31,242 p=6759 u=root |  TASK [1-prep : Set has ifcfg gw based on on macaddress if found] ***************
 2017-10-22 17:34:31,256 p=6759 u=root |  skipping: [127.0.0.1]
2017-10-22 17:34:31,271 p=6759 u=root |  TASK [1-prep : Find wifi gateway config if present - Can Fail] *****************
2017-10-22 17:34:31,285 p=6759 u=root |  skipping: [127.0.0.1]
2017-10-22 17:34:31,300 p=6759 u=root |  TASK [1-prep : Setting has_wifi_gw based on ESSID if found - Can Fail] *********
2017-10-22 17:34:31,311 p=6759 u=root |  fatal: [127.0.0.1]: FAILED! => {"failed": true, "msg": "'dict object' has no attribute 'stdout_lines'"}
2017-10-22 17:34:31,313 p=6759 u=root |   to retry, use: --limit @/opt/iiab/iiab/iiab.retry

2017-10-22 17:34:31,313 p=6759 u=root |  PLAY RECAP *********************************************************************
2017-10-22 17:34:31,313 p=6759 u=root |  127.0.0.1                  : ok=31   changed=7    unreachable=0    failed=1

holta, 2017-10-23:
Current master's ./runansible completes for me on Ubuntu 16.04 LTS but I'm trying to better understand the situation.

@georgejhunt can you post the failing code to http://pastebin.mozilla.org to shed more light?

holta, 2017-10-23:
Let's not bash Ansible 2.4 when this occurs in Ansible 2.2 and 2.3 etc &mdash; it's part of Ansible's confusing spec [*] which is what we should be bashing!

In short: **skipped includes get evaluated** as explained within https://github.com/ansible/ansible/issues/20287#issuecomment-272916520 :
> the problem is that your include is static, which means the "when" is applied to the included tasks (the include itself is NOT skipped). This "when" fails as it executes per item as per docs http://docs.ansible.com/ansible/playbooks_conditionals.html#loops-and-conditionals
>
> IF you want the when to apply to the include itself, try setting static: no.

[*] hence the reason "include:" is being deprecated in Ansible 2.8 &mdash; as everyone falls for this:
http://docs.ansible.com/ansible/latest/include_module.html

Short-Term Solution for such situations &mdash; works since Ansible 2.1: insert "static: no" between [lines 64 and 65 in roles/1-prep/tasks/detected_network.yml](https://github.com/iiab/iiab/blob/master/roles/1-prep/tasks/detected_network.yml#L64-L65)

Long-Term Solution for such situations &mdash; works since Ansible 2.4: make it dynamic instead of static by changing [line 64](https://github.com/iiab/iiab/blob/master/roles/1-prep/tasks/detected_network.yml#L64) from:
```
include: detected_redhat.yml
```
...to:
```
include_tasks: detected_redhat.yml
```

holta, 2017-10-23:
@georgejhunt change the subject line to anything that works for you &mdash; and most important thanks for opening our eyes to this super-annoying problem that (apparently) is part of all versions of Ansible &mdash; hence the reason "include...when" is officially deprecated & on its way out in coming couple years.

holta, 2017-10-23:
My quick tests indicate Ansible 2.0.x seems to be the *one and only version* that interprets detected_network.yml's "include...when" dynamically (as per new syntax "include_tasks...when").

All other versions of Ansible (2.1, 2.2, 2.3, 2.4...and those prior to 2.0!) seem to evaluate skipped includes by default, when using the "include...when" legacy syntax.

georgejhunt, 2017-10-23:
When I tried "include_tasks" (as Adam's research suggests) the error is 
- name: RedHat Network detection
  ^ here

ERROR! no action detected in task. This often indicates a misspelled module name, or incorrect module path.

-- which is correct if the "when" clause is false. So I'm not understanding how to proceed.  The raw "include" is deprecated, and scheduled to be removed in 2.8.

georgejhunt, 2017-10-23:
One thought I just had. The RedHat Network detection is an include, which is defined within an included yml file.  Maybe "include_tasks" needs to be used up the calling tree.  
I'm not in favor of more than one level of includes -- it makes it much more difficult to find stuff when reading through the code, and trying to guess where something has gone astray.

georgejhunt, 2017-10-23:
using include_tasks, and "static: no" does not solve the problem:

root@box:/opt/iiab/iiab# git diff
diff --git a/roles/1-prep/tasks/detected_network.yml b/roles/1-prep/tasks/detected_network.yml
index 502f3dc..d81dbf3 100644
--- a/roles/1-prep/tasks/detected_network.yml
+++ b/roles/1-prep/tasks/detected_network.yml
@@ -61,7 +61,8 @@
   when: discovered_wan_iface != "none"
 
 - name: RedHat Network detection
-  include: detected_redhat.yml
+  include_tasks: detected_redhat.yml
+  static: no
   when: is_redhat
 
 # WIRELESS -- if any wireless is detected as gateway, it becomes WAN
diff --git a/roles/1-prep/tasks/main.yml b/roles/1-prep/tasks/main.yml
index dc9672c..a58d81b 100644
--- a/roles/1-prep/tasks/main.yml
+++ b/roles/1-prep/tasks/main.yml
@@ -87,7 +87,8 @@
 
 - include: computed_vars.yml
 
-- include: detected_network.yml
+- include_tasks: detected_network.yml
+  static: no
   when: not installing
 
 # Put conditional actions for hardware platforms here

georgejhunt, 2017-10-23:
> In short: *skipped includes get evaluated* as explained within ansible/ansible#20287
> (comment)
> ansible/playbooks_conditionals.html#loops-and-conditionals
> [*] hence the reason "include:" is being deprecated in Ansible 2.8 — as
> Short-Term Solution for such situations — works since Ansible 2.1: insert
> Long-Term Solution for such situations — works since Ansible 2.4: make it

abort on all my testing I have reported.  I'm on ansible 2.2!!!

georgejhunt, 2017-10-23:
Changing to "include_tasks" will probably work.  My failures were on ansible version 2.2

holta, 2017-10-23:
Ref #451 for another way to solve the orig issue.

Also I tested "include_tasks...when" last night just to verify the execution path.

But let's try to decide during our call (http://minutes.iiab.io) on any relatively-permanent solution(s) that works for everybody.

jvonau, 2017-10-23:
Related #319

holta, 2017-10-23:
Clarif: both PR #451 & PR #319 were merged and we will try to solve the larger problem of short-term vs. long-term solution across all 70 instances of "include...when" starting on our community/team call Thursday (http://minutes.iiab.io)

holta, 2017-10-24:
In the spirit of #454 (eliminating unintended side-effects, idempotency) I converted our 56 instances of "include:" to "include_tasks:" on a clean WIFI INSTALL on Raspbian Desktop:
```
cd /opt/iiab/iiab/roles
find ./ -type f -exec sed -i 's/include:/include_tasks:/g' {} \;
```
These 56 instances occur in these 21 files: (run "grep -rc include_tasks: * | grep -v :0")
```
1-prep/tasks/detected_network.yml:1
1-prep/tasks/main.yml:5
2-common/tasks/main.yml:8
ajenti/tasks/main.yml:1
awstats/tasks/main.yml:1
dokuwiki/tasks/main.yml:1
httpd/tasks/main.yml:1
iiab-admin/tasks/main.yml:2
kalite/tasks/main.yml:5
kiwix/tasks/main.yml:1
mysql/tasks/main.yml:2
network/tasks/ifcfg_mods.yml:4
network/tasks/squid.yml:1
network/tasks/static.yml:1
network/tasks/main.yml:16
nextcloud/tasks/main.yml:1
owncloud/tasks/main.yml:1
rachel/tasks/main.yml:1
sugar-stats/tasks/main.yml:1
teamviewer/tasks/main.yml:1
wordpress/tasks/main.yml:1
```
All functional tests worked, on this basic install:
* http://box
* http://box/usb
* http://box/info
* http://box/maps
* http://box/admin
* http://box/sugarizer

PS since the above I've expanded local_vars.yml and am currently running another "./runansible" so all possible above code is run through & tested (for more complete confirmation that "include_tasks:" is demonstrably solid).

holta, 2017-10-24:
Larger local_vars.yml worked passing all functional tests (Kiwix, KA Lite, Nextcloud, DokuWiki, AWStats, etc) confirming Ansible 2.4's "include_tasks:" works across the board!

holta, 2017-10-24:
@tim-moody let's not forget this 57th instance of the deprecated "include:" syntax in Admin Console -> CMDSRV we'll have to swap out eventually &mdash; [Line 2 of roles/cmdsrv/tasks/main.yml](https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/tasks/main.yml#L2) here:
```
include: packages.yml
```

holta, 2017-11-08:
We've upgrade all instances of "include:" to "include_tasks:" as of #490.

closed 2017-11-08 by holta