iiab/iiab #454: [Travis CI] Commands should not change things if nothing needs doing

arky, 2017-10-24:
Idempotence is an important concept in Ansible. It seems we have large number of tasks that are violating this principle.

Here are some of the Travis CI "[EANSIBLE0012] Commands should not change things if nothing needs doing" warnings: 

````shell
[0;34m/home/travis/build/arky/iiab/roles/1-prep/tasks/detected_network.yml[0m:[0;36m98[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/1-prep/tasks/detected_redhat.yml[0m:[0;36m50[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/1-prep/tasks/main.yml[0m:[0;36m41[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/centos.yml[0m:[0;36m1[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/net_mods.yml[0m:[0;36m18[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/net_mods.yml[0m:[0;36m25[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/udev.yml[0m:[0;36m33[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/xo.yml[0m:[0;36m1[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/3-base-server/tasks/main.yml[0m:[0;36m5[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/4-server-options/tasks/main.yml[0m:[0;36m1[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/5-xo-services/tasks/main.yml[0m:[0;36m1[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/6-generic-apps/tasks/main.yml[0m:[0;36m1[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/7-edu-apps/tasks/main.yml[0m:[0;36m1[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/8-mgmt-tools/tasks/main.yml[0m:[0;36m1[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/9-local-addons/tasks/main.yml[0m:[0;36m1[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/activity-server/tasks/main.yml[0m:[0;36m18[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/authserver/tasks/main.yml[0m:[0;36m35[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/dokuwiki/tasks/install.yml[0m:[0;36m8[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/iiab-admin/tasks/admin-user.yml[0m:[0;36m35[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/iiab-admin/tasks/admin-user.yml[0m:[0;36m54[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/kalite/tasks/install-f18.yml[0m:[0;36m39[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/kalite/tasks/setup-f18.yml[0m:[0;36m3[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/kalite/tasks/setup-f18.yml[0m:[0;36m8[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/kalite/tasks/setup-f18.yml[0m:[0;36m13[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/kalite/tasks/setup.yml[0m:[0;36m10[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/kiwix/tasks/kiwix_install.yml[0m:[0;36m74[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/computed_network.yml[0m:[0;36m9[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/computed_network.yml[0m:[0;36m139[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/enable_services.yml[0m:[0;36m113[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/hostapd.yml[0m:[0;36m22[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/redetect.yml[0m:[0;36m72[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/restart.yml[0m:[0;36m54[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/nextcloud/tasks/nextcloud_enabled.yml[0m:[0;36m30[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/nextcloud/tasks/nextcloud_enabled.yml[0m:[0;36m55[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/owncloud/tasks/owncloud_enabled.yml[0m:[0;36m29[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/phpmyadmin/tasks/main.yml[0m:[0;36m14[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/sugarizer/tasks/main.yml[0m:[0;36m5[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/vnstat/tasks/main.yml[0m:[0;36m17[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/wordpress/tasks/install.yml[0m:[0;36m10[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/wordpress/tasks/install.yml[0m:[0;36m22[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;34m/home/travis/build/arky/iiab/roles/wordpress/tasks/install.yml[0m:[0;36m41[0m: [[1;31mEANSIBLE0012[0m] [0;31mCommands should not change things if nothing needs doing[0m
[0;33mroles/1-prep/tasks/detected_network.yml:98: [ANSIBLE0012] Commands should not change things if nothing needs doing[0m
[0;33mroles/1-prep/tasks/detected_redhat.yml:50: [ANSIBLE0012] Commands should not change things if nothing needs doing[0m
[0;33mroles/1-prep/tasks/main.yml:41: [ANSIBLE0012] Commands should not change things if nothing needs doing[0m
[0;33mroles/2-common/tasks/centos.yml:1: [ANSIBLE0012] Commands should not change things if nothing needs doing[0m
[0;33mroles/2-common/tasks/net_mods.yml:18: [ANSIBLE0012] Commands should not change things if nothing needs doing[0m
[0;33mroles/2-common/tasks/net_mods.yml:25: [ANSIBLE0012] Commands should not change things if nothing needs doing[0m
[... cut ...]
[0;33mroles/9-local-addons/tasks/main.yml:1: [ANSIBLE0012] Commands should not change things if nothing needs doing[0m
[0;33mroles/activity-server/tasks/main.yml:18: [ANSIBLE0012] Commands should not change things if nothing needs doing[0m
[0;33mroles/authserver/tasks/main.yml:35: [ANSIBLE0012] Commands should not change things if nothing needs doing[0m
[0;33mroles/debian_schooltool/tasks/main.yml:45: [ANSIBLE0012] Commands should not change things if nothing needs doing[0m
[0;33mroles/dokuwiki/tasks/install.yml:8: [ANSIBLE0012] Commands should not change things if nothing needs doing[0m
````

jvonau, 2017-10-24:
[root@box iiab]# cat roles/9-local-addons/tasks/main.yml
- name: Addon services installed
  command: echo Addon services installed

What is being 'changed' here? What is the best practice to return feedback to the user? related #441

holta, 2017-11-21:
@arky is any further action required here?

Or can we consider closing?

holta, 2017-11-23:
@jvonau should we close this?  Or defer to IIAB 6.6?

closed 2018-06-19 by holta