iiab/iiab #455: [Travis CI] Shell should only be used when essential warnings

arky, 2017-10-24:
There are numerous shell warnings in Travis CI tests. In general using commands is not a best practice in Ansible.  Perhaps it would good idea to use this code to turn off these warnings.  

`args:
    warn: no` 

Refer: http://willthames.github.io/2016/09/21/using-command-and-shell-in-ansible.html

```
[34m[1mshellcheck version[0m
	SHELLOPTS
PAM service name to use for login shells
[0;34m/home/travis/build/arky/iiab/roles/1-prep/tasks/detected_network.yml[0m:[0;36m1[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/1-prep/tasks/detected_network.yml[0m:[0;36m12[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/1-prep/tasks/main.yml[0m:[0;36m32[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/centos.yml[0m:[0;36m22[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/fedora.yml[0m:[0;36m13[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/fedora.yml[0m:[0;36m17[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/main.yml[0m:[0;36m31[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/net_mods.yml[0m:[0;36m6[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/net_mods.yml[0m:[0;36m18[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/net_mods.yml[0m:[0;36m25[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/packages.yml[0m:[0;36m1[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/packages.yml[0m:[0;36m5[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/packages.yml[0m:[0;36m9[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/packages.yml[0m:[0;36m18[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/packages.yml[0m:[0;36m130[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/udev.yml[0m:[0;36m22[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/udev.yml[0m:[0;36m26[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/udev.yml[0m:[0;36m33[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/2-common/tasks/xo.yml[0m:[0;36m37[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/awstats/tasks/install.yml[0m:[0;36m91[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/awstats/tasks/install.yml[0m:[0;36m95[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/cups/tasks/main.yml[0m:[0;36m40[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/ejabberd/tasks/main.yml[0m:[0;36m63[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/ejabberd/tasks/main.yml[0m:[0;36m67[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/ejabberd_xs/tasks/main.yml[0m:[0;36m11[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/ejabberd_xs/tasks/main.yml[0m:[0;36m20[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/ejabberd_xs/tasks/main.yml[0m:[0;36m79[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/ejabberd_xs/tasks/main.yml[0m:[0;36m83[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/elgg/tasks/main.yml[0m:[0;36m4[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/elgg/tasks/main.yml[0m:[0;36m17[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/iiab-admin/tasks/admin-user.yml[0m:[0;36m35[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/iiab-admin/tasks/admin-user.yml[0m:[0;36m54[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/kalite/tasks/main.yml[0m:[0;36m25[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/moodle/tasks/main.yml[0m:[0;36m127[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/create_ifcfg.yml[0m:[0;36m1[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/edit_ifcfg.yml[0m:[0;36m42[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/enable_services.yml[0m:[0;36m113[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/hostapd.yml[0m:[0;36m22[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/ifcfg_mods.yml[0m:[0;36m7[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/ifcfg_mods.yml[0m:[0;36m15[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/ifcfg_mods.yml[0m:[0;36m21[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/ifcfg_mods.yml[0m:[0;36m27[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/ifcfg_mods.yml[0m:[0;36m71[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/ifcfg_mods.yml[0m:[0;36m80[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[0;34m/home/travis/build/arky/iiab/roles/network/tasks/ifcfg_mods.yml[0m:[0;36m84[0m: [[1;31mEANSIBLE0013[0m] [0;31mUse shell only when shell functionality is required[0m
[... cut ...]
[0;33mWARN: Best practice "Shell should only be used when essential" not met:[0m
[0;33mroles/ejabberd_xs/tasks/main.yml:83: [ANSIBLE0013] Use shell only when shell functionality is required[0m
[0;33mWARN: Best practice "Shell should only be used when essential" not met:[0m
[0;33mroles/elgg/tasks/main.yml:4: [ANSIBLE0013] Use shell only when shell functionality is required[0m
[0;33mWARN: Best practice "Shell should only be used when essential" not met:[0m
[0;33mroles/elgg/tasks/main.yml:17: [ANSIBLE0013] Use shell only when shell functionality is required[0m
```

jvonau, 2017-10-24:
> Refer: http://willthames.github.io/2016/09/21/using-command-and-shell-in-ansible.html
... So if you’re not stringing two commands together (using pipes or even just && or ;), you don’t really need the shell module....

I'm always open to better ways, from memory the majority of the shell code **is** combining 2 or more commands, because the equivalent functionality did not exist (and still may not) in ansible.

holta, 2017-11-21:
We've made a lot of progress here.

But am deferring to IIAB 6.6 in case further is poss?

closed 2018-06-19 by holta