iiab/iiab #457: scripts/ansible does not upgrade or issue any message

tim-moody, 2017-10-24:
deb 9 vm with ansible 2.2.0.0
latest master - scripts/ansible
terminates quickly with no message
is this expected?

holta, 2017-10-24:
Can you use Ansible 2.4.0 per Step 5 of our [Do Everything from Scratch](https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch) installation instructions?  (That Josh Dennis helped us refine for Debian 9 over the past month.)

tim-moody, 2017-10-24:
root@debian9base:/opt/iiab/iiab# ansible --version
ansible 2.2.0.0
  config file = /opt/iiab/iiab/ansible.cfg
  configured module search path = ['/usr/share/ansible']
root@debian9base:/opt/iiab/iiab# apt purge ansible
Reading package lists... Done
Building dependency tree
Reading state information... Done
**Package 'ansible' is not installed, so not removed**
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

holta, 2017-10-24:
Is it possible 2.2.0 got installed onto 9.1 (or whatever you have) thru pip or some other means than the OS's own "apt install ansible" etc?

(Any idea how 2.2.0 got there?)

tim-moody, 2017-10-24:
I think it got there from the previous version of scripts/ansible

holta, 2017-10-24:
Makes sense.  We have not focused on the installed base, as this has not been a concern to date.

But I do agree with your suggestion that scripts/ansible should clarify matters for "Experts" who become attached to obscure/buggy/older versions of Ansible for their own reasons...and as such we need a PSA right there at the command line...to make clear to them how to extricate...as we all know "overly experienced" sysadmins are people too :-) who need to be able to recover from their own secret sauce per https://github.com/iiab/iiab/wiki/IIAB-Installation#expert-mode (!)

tim-moody, 2017-10-24:
it's not a matter of attachment. there is no way to upgrade

holta, 2017-10-24:
There must be a way to remove Ansible 2.2.0

Unfortunately I'm not the expert )-:

Maybe @georgejhunt & @jvonau & @arky know?

tim-moody, 2017-10-24:
and, btw, this was the official means of installing ansible. it's just that iiab changed afterwards and the changed script is unable to undo what its predecessor did.

tim-moody, 2017-10-24:
see above

holta, 2017-10-24:
When exactly was Ansible 2.2.0 installed (on Debian 9.1 I presume) to help others understand the lineage of what you're working on?  Since 6.4 almost a month ago, we've asked that installers avoid "apt install ansible" and install Ansible 2.4.0 directly using scripts/ansible &mdash; are we talking about a system that predates that?

holta, 2017-10-24:
Yesterday I installed all 5 version of Ansible {2.0, 2.1, 2.2, 2.3, 2.4} on Ubuntu 16.04 LTS to experiment with each (on the very same system) and that is extremely fast+easy using PPA's from launchpad.net

I apologize I'm far less familiar with Debian!

But likely/possibly "apt -a list ansible" can helpful on Debian too, in managing similar quick/pitstop changes between different versions of Ansible on Debian likewise?

tim-moody, 2017-10-24:
pip uninstall ansible got rid of it (I guess it wasn't a distro package)

now scripts/ansible runs

jvonau, 2017-10-24:
> terminates quickly with no message

Thanks for taking this step, fresh installs were the priority, now to backtrack and cover the corner cases in upgrades.  The scripts looks if ansible-playbook is installed and if installed looks at what version is installed, finding version 2.2.0.0 should echo the current version and force a pip upgrade to version 2.4.0. Think I can reproduce the pip installed ansible-2.2.0.0 layout and see if I can reproduce this. Given that 2.2.0.0 was pip installed originally could you try a manual upgrade of ansible with
pip install --upgrade ansible==2.4.0 --disable-pip-version-check

jvonau, 2017-10-24:
> pip uninstall ansible got rid of it (I guess it wasn't a distro package)
now scripts/ansible runs

The script should of forced the upgrade via pip but I see that part needs work. I'll work at that part should time come up, although forcing the use of PPA does have an advantage. I'm thinking if the path to ansible is /usr/loca/bin/ then that is a pip install and proceed to pip uninstall before proceeding? Better as upgrade notes or scripted upgrades?

holta, 2017-10-24:
Thanks @tim-moody & @jvonau this clarifies a lot:

As such I've added `pip uninstall ansible` to STEP 5 of the official ["Do Everything from Scratch"](https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch) install instructions until the dust settles here.

Please everyone help articulate what's best: whether [scripts/ansible](https://github.com/iiab/iiab/blob/master/scripts/ansible) should include a PSA to help installers orient themselves when some weirdo new Ansible version/upgrading corner case hits them and/or whether they never will RTFM.  Regardless it seems wise (to me!) that scripts/runansible itself include a strong recommendation that folks at least be **aware** of the following URL before they proceed :-)

https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch

holta, 2017-10-24:
Ref #375

jvonau, 2017-11-07:
better feedback in #490

jvonau, 2017-11-12:
For the record the behaviour matched it's predecessor when ansible was already installed. 
[root@box iiab]# scripts/ansible-2.2.0 
+ '[' x = xDEBUG -o x = xDEBUG ']'
++ which ansible-playbook
+ '[' '!' /bin/ansible-playbook ']'

holta, 2017-11-12:
Currently ./iiab-install warns if Ansible is missing or too old, per #509 built on #498.

So I'm closing this ticket.

closed 2017-11-12 by holta