iiab/iiab #464: [TravisCI] Variable error 'dict object' has no attribute 'local_facts'

arky, 2017-10-26:
Here is another error surfaced by the Travis CI variable tests.  I am not sure if  ` ansible_local["local_facts"]["iiab_branch"] ` or ` ansible_local.local_facts.iiab_branch ` is right way to retrieve something from the stored facts. 

https://github.com/iiab/iiab/blob/8970bb8fcb4297b7c39f50522d224e36e43a4e0c/roles/1-prep/tasks/iiab_ini.yml#L26-L34

```
TASK [../roles/1-prep : add version section] ***********************************
task path: /home/travis/build/arky/iiab/roles/1-prep/tasks/iiab_ini.yml:17
fatal: [localhost]: FAILED! => {
    "failed": true, 
    "msg": "'dict object' has no attribute 'local_facts'"
}
	to retry, use: --limit @/home/travis/build/arky/iiab/tests/test.retry
PLAY RECAP *********************************************************************
localhost                  : ok=10   changed=6    unreachable=0    failed=1   
```

jvonau, 2017-10-29:
Either way work but I'm cleaning that up in #474 to go with ansible_local.local_facts.iiab_branch for readability. 

>     "msg": "'dict object' has no attribute 'local_facts'"

We are placing local_facts into play with this bit of code from runansible
 if [ ! -f /etc/ansible/facts.d/local_facts.fact ]; then
    mkdir -p /etc/ansible/facts.d
 fi
cp ./scripts/local_facts.fact /etc/ansible/facts.d/local_facts.fact

Needs to be in place before the play is called, just like ansible_local.local_facts.os_ver

holta, 2017-11-07:
Thanks @jvonau & @arky !

Let's resolve during Thursday's community/team call (http://minutes.iiab.io) if at all poss?

In any case, Thursday (Nov 9) being our code freeze target day, let's do our best to start closing out residual issues for [IIAB 6.5](https://github.com/iiab/iiab/wiki/IIAB-6.5-Release-Notes)'s release :)

holta, 2017-11-09:
Thanks all, closed per call discussion (http://minutes.iiab.io)

Plz re-open as nec.

closed 2017-11-09 by holta