iiab/iiab #468: resolvconf issue in roles/network/tasks/debian.yml

holta, 2017-10-27:
Is this warning acting on, on this Ubuntu 16.04 LTS NUC installed 3-4 weeks ago?

Today's upgrade from IIAB 6.4 to the latest iiab/iiab master worked beautifully ("git pull; ./runansible") but FYI this warning arose:

> TASK [network : In upgrade from earlier IIAB 6.2, delete the resolvconf] *******
FAILED! => {"changed": false, "failed": true, "msg": "Unsupported parameters for (apt) module: enabled Supported parameters include: allow_unauthenticated,autoclean,autoremove,cache_valid_time,deb,default_release,dpkg_options,force,force_apt_get,install_recommends,only_upgrade,package,purge,state,update_cache,upgrade"}
...ignoring

The error/warning is arising from [Line 13 of roles/network/tasks/debian.yml](https://github.com/iiab/iiab/blob/master/roles/network/tasks/debian.yml#L13-L17) if that helps?

jvonau, 2017-10-27:
refs: #350 #470

jvonau, 2017-11-07:
resolved in #490