iiab/iiab #481: Admin Console's 3 Network Parameters checkboxes are empty on 1st boot

holta, 2017-11-02:
All 3 checkboxes are empty at http://box/admin -> Configure tab -> Network Parameters when testing against https://github.com/jvonau/iiab/tree/merge-order4 [*] :

( ) Gateway
( ) Appliance
( ) LAN Controller

[*] e.g. when installing on Raspbian 2017-09-07 using e.g. using http://download.iiab.io/6.5/rpi/load-pr477-big-vpn.txt

holta, 2017-11-05:
Side Observation:

Admin Console install times dramatically improved (about 3X faster) in recent weeks, if you use Ansible 2.4.1 on RPi3 / Raspbian Lite fully updated from 2017-09-07, with @tim-moody's latest branch clones as follows:
```
git clone https://github.com/tim-moody/iiab-admin-console -b fixes-oct-24
```
Currently this takes about ~6MIN.  A few weeks ago the equivalent/earlier installation of Admin Console in took ~17MIN on Ansible 2.4.0.  I don't know why such a huge difference.

FWIW a *re-install* of Admin Console just took 3min37s, when doing "rm -rf /opt/iiab/iiab-admin-console" followed by "git clone https://github.com/tim-moody/iiab-admin-console -b fixes-oct-24" then the usual install routine ("cd /opt/iiab/iiab-admin-console" then "./install" with Ansible 2.4.1).

tim-moody, 2017-11-05:
due to change of  iiab_network_mode to  iiab_network_mode_applied in iiab.ini[network]

holta, 2017-11-06:
FYI Network Parameter’s choice among 3 radio buttons is set here in @jvonau's current #486: (branch iiab-order)

https://github.com/jvonau/iiab/blob/iiab-order/roles/network/defaults/main.yml#L35

e.g. by default Line 35 shows:
```
gui_desired_network_role: Gateway
```
(From @jvonau during today's community/team call @ http://minutes.iiab.io)

holta, 2017-11-06:
Apparently fixed by https://github.com/iiab/iiab-admin-console/pull/41

closed 2017-11-06 by holta