iiab/iiab #482: Raspbian kernel update breaks bridging

jvonau, 2017-11-03:
### Expected behavior and actual behavior.
### Steps to reproduce the problem.

BROKEN
root@box:/opt/iiab/iiab# apt list -a raspberrypi-kernel
Listing... Done
raspberrypi-kernel/stable,now 1.20171029-1 armhf [installed,automatic]

root@box:/opt/iiab/iiab# uname -a
Linux box.lan 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux

root@box:/opt/iiab/iiab# brctl addbr br1
add bridge failed: Package not installed

WORKING
root@box:/home/iiab-admin# apt list -a raspberrypi-kernel
Listing... Done
raspberrypi-kernel/testing 1.20171029-1 armhf [upgradable from: 1.20170811-1]
raspberrypi-kernel/now 1.20170811-1 armhf [installed,upgradable to: 1.20171029-1]

root@box:/home/iiab-admin# uname -a
Linux box.lan 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux

root@box:/home/iiab-admin# brctl addbr br1
root@box:/home/iiab-admin# ip a
snip
8: br1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN group default qlen 1000

jvonau, 2017-11-03:
secondary issue of why is 'testing' repo enabled by default as noted in #392 also.

jvonau, 2017-11-03:
Upstream issue https://github.com/raspberrypi/linux/issues/2256

holta, 2017-11-03:
Thanks much @jvonau for escalating this critical problem as we've received no response from irc channel #raspberrypi-kernel on irc.freenode.net over the past 72 hours...

Nor on Raspberry Pi's Forums just yet: https://www.raspberrypi.org/forums/viewtopic.php?t=196623

holta, 2017-11-03:
Clarif on how very serious this is:
* Internet-in-a-Box WiFi hotspots no longer work (Access Points inside RPi3 or RPI Zero W)
* OpenVPN no longer works

jvonau, 2017-11-05:
TEBKAC - needs a reboot to use the new kernel. Please rethink install instructions, curl-me and scripts/ansible's apt-get -y update to handle updating kernel and where the needed reboot should be.

holta, 2017-11-05:
OpenVPN confirmed to work on a clean install with "apt update" then "apt dist-upgrade" then "reboot" then "iiab-install".

The reboot can also be done at the end of the install presumably, possibly with a quick "./runtags OpenVPN" or similar.

Finally, none of this will be necessary when the new https://www.raspberrypi.org/downloads/raspbian/ is released in very few days, to replace the 2017-09-07 editions.

(Of course, please re-open as necessary if this or similar re-occurs!)

closed 2017-11-05 by holta