iiab/iiab #484: Kiwix full-text searches break (& cached results are wiped) "15min into every lesson"

holta, 2017-11-04:
Is there any reason not to experiment with commenting out the cron job @ https://github.com/iiab/iiab/blob/master/roles/kiwix/tasks/kiwix_install.yml#L77-L88 as a result of the much more reliable Kiwix 0.3.0 from 2017-10-25?  i.e. is commenting this out dangerous / premature for other known reasons?

Context: Kiwix's full-text searches work beautifully @ http:// iiab . me : 3000 if you're willing to wait 2-to-7 minutes typically, which is fine at the very beginning of a school's lesson (if it works especially ;)  And thanks to [Kiwix 0.3.0](https://github.com/kiwix/kiwix-tools/commits/master) from [2017-10-25](http://download.kiwix.org/nightly/2017-10-25/) (e.g. https://github.com/iiab/iiab/pull/463) followup search results take about 30 seconds each typically (not fun, but tolerable!) when clicking on these buttons at the bottom of the page, for the 1st 250 search results: (i.e. 25 search results per page)

`<- 1 2 3 4 5 6 7 8 9 10 ->`

HOWEVER 15 minutes after every hour, you can no longer click on these 12 buttons, at the bottom of every search page, without triggering another 2-to-7 minutes search, as a result of the cron job installed in the above kiwix_install.yml :/

holta, 2017-11-04:
Somewhat Related:

https://github.com/kiwix/kiwix-tools/issues/92 "Why are Kiwix searches blocked when asking for more than 67 results?"

closed 2017-11-05 by holta

mgautierfr, 2017-11-06:
It would be better to not especially kill and restart the kiwix-serve processus.
Systemd can be configured to automatically restart a service if the process crash. And (it seems) it is what you want.

See https://jonarcher.info/2015/08/ensure-systemd-services-restart-on-failure/ for an example.

tim-moody, 2017-11-06:
@mgautierfr the problem was in the past that kiwix-serve went to 100% cpu and became unresponsive so people decided to restart it every 15 minutes. I think we should stop restarting it at all and continue to monitor it.

mgautierfr, 2017-11-06:
Ho, I've heard of kiwix-server crashing but never of kiwix-serve taking 100% cpu.
Indeed, in this case, just restarting in case of crash is not the solution.

I hope the rewrite of the low level part of libzim has fixed the problem but I have no specific clue about this :/

commit 7b929140 (2017-11-05): Merge pull request #485 from holta/kiwix-cron-nightly-not-hourly

restart kiwix-serve 4AM nightly (not 15min after every hour) & restrict chown to /opt/iiab/kiwix

Smoke-tested.  Please continue to test locally, so we can refine (or eliminate if nec?) this cron job's frequency as nec!

Closes #484