iiab/iiab #488: runansible fails on current master

tim-moody, 2017-11-05:
### Expected behavior and actual behavior.
runansible completes
### Steps to reproduce the problem.
clone master, scripts/ansible, runansible
### Specifications like the version of the IIAB, operating system version, or hardware details.
deb 9 vm, current master

TASK [vnstat : Create database for LAN to collect vnStat data if not appliance config] *******************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"failed": true, "msg": "The conditional check 'not iiab_lan_iface == \"\"' failed. The error was: error while evaluating conditional (not iiab_lan_iface == \"\"): 'iiab_lan_iface' is undefined\n\nThe error appears to have been in '/opt/iiab/iiab/roles/vnstat/tasks/main.yml': line 20, column 3, but may\nbe elsewhere in the file depending on the exact syntax problem.\n\nThe offending line appears to be:\n\n\n- name: Create database for LAN to collect vnStat data if not appliance config\n  ^ here\n"}

Please note **iiab_lan_iface' is undefined**

holta, 2017-11-05:
Is this VM Debian 9.1 or Debian 9.2?

tim-moody, 2017-11-05:
9.1 . why does it matter?

holta, 2017-11-05:
Point releases have been caused testers many weeks of painful testing recently (worse yet package releases WITHIN OS point-releases!) so far better to be precise, given everyone's heavy investments in testing making this all possible...

Thanks all for your understanding here!

holta, 2017-11-06:
I'm pretty sure @jvonau can solve this overnight, if he has time tonight hopefully, with testing help on Debian 9.2.1 or similar.

Not sure how this arose but when we're being hammered with demands from so many different/evolving OS's, we need to keep a shrewd eye on how many we can legitimately support~

_(Anish's professional opinion last year...was that it's suicide to support more than 1 OS long term ;-)_

holta, 2017-11-06:
@jvonau mentions PostgreSQL appears to have critical packages missing on Debian &mdash; unlike on Ubuntu 16.04 and Raspbian Stretch.

Though I'm not sure this would be related to vnStat?  (I'd thought PostgreSQL only affected Moodle & Pathagar?)

closed 2017-11-06 by holta