iiab/iiab #494: "Failed to start A distributed, fault-tolerant Jabber/XMPP server."

holta, 2017-11-08:
Is it time to document that ejabberd does not work, in our [IIAB 6.5 Release Notes'](https://github.com/iiab/iiab/wiki/IIAB-6.5-Release-Notes) Known Issues ?

As the following message appears in red during every boot, on all installs to date with the [BIG-sized local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars_big.yml) on (apt updated) Raspbian Lite of 2017-09-07:
```
Failed to start A distributed, fault-tolerant Jabber/XMPP server.
```
The following 2 lines in /opt/iiab/iiab/vars/local_vars.yml are setting this in motion:
```
ejabberd_install: True
ejabberd_enabled: False
```
Note this is entirely separate from the earlier ejabberd used in the OLPC/XO/XS era, further up in this same file:
```
# ejabberd_xs_install: False
# ejabberd_xs_enabled: False
```
Context: an effort to improve ejabberd was merged yesterday &mdash; with PR [#143](https://github.com/iiab/iiab/pull/143#issuecomment-339996903) &mdash; just in case somebody wants to take this on for [IIAB 6.6](https://github.com/iiab/iiab/milestone/3) possibly.

jvonau, 2017-11-08:
jenkins big_vars run over vpn:
> root@box:/home/iiab-admin# systemctl status ejjaberd
Unit ejjaberd.service could not be found.
root@box:/home/iiab-admin# systemctl status ejabberd
● ejabberd.service - A distributed, fault-tolerant Jabber/XMPP server
   Loaded: loaded (/lib/systemd/system/ejabberd.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: https://www.process-one.net/en/ejabberd/docs/
Nov 08 05:08:47 box.lan systemd[1]: Starting A distributed, fault-tolerant Jabber/XMPP server...
Nov 08 05:08:52 box.lan systemd[1]: Started A distributed, fault-tolerant Jabber/XMPP server.
Nov 08 05:08:57 box.lan systemd[1]: Stopping A distributed, fault-tolerant Jabber/XMPP server...
Nov 08 05:09:04 box.lan systemd[1]: Stopped A distributed, fault-tolerant Jabber/XMPP server.
root@box:/home/iiab-admin# systemctl start ejabberd
root@box:/home/iiab-admin# systemctl status ejabberd
● ejabberd.service - A distributed, fault-tolerant Jabber/XMPP server
   Loaded: loaded (/lib/systemd/system/ejabberd.service; disabled; vendor preset: enabled)
   Active: active (running) since Wed 2017-11-08 06:19:25 GMT; 4s ago
     Docs: https://www.process-one.net/en/ejabberd/docs/
  Process: 32522 ExecStart=/bin/sh -c /usr/sbin/ejabberdctl start && /usr/sbin/ejabberdctl started (cod
 Main PID: 32547 (beam.smp)
    Tasks: 17 (limit: 4915)
   CGroup: /system.slice/ejabberd.service
           ├─32547 /usr/lib/erlang/erts-8.2.1/bin/beam.smp -K true -P 250000 -- -root /usr/lib/erlang -
           └─32565 erl_child_setup 32768
Nov 08 06:19:19 box.lan systemd[1]: Starting A distributed, fault-tolerant Jabber/XMPP server...
Nov 08 06:19:25 box.lan systemd[1]: Started A distributed, fault-tolerant Jabber/XMPP server.

Lets have a further look, this one looks to be installed ok

jvonau, 2017-11-08:
related #337 #347

holta, 2017-11-09:
@jvonau will take a look in coming days -- we'll defer to IIAB 6.6 if it's not solvable.

holta, 2017-11-09:
Error can also be seen in /var/log/boot.log

holta, 2018-06-11:
No longer relevant due to @jvonau's upgrade to stock ejabberd: https://github.com/iiab/iiab/pull/782