iiab/iiab #501: Admin Console fails to install with MIN-sized local_vars.yml

holta, 2017-11-08:
Possibly this affect all installs of the default [MEDIUM-size local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml) as well?

@tim-moody the fix is to uncomment the 1st two lines of roles/cmdsrv/tasks/main.yml (reversing part of https://github.com/iiab/iiab-admin-console/commit/d885adbc5d850c00d3bcddbf07c105857bbc3feb).  Can you make this or similar happen today ideally?  As this is affecting many.

Admin Console installs properly with the above fix, with these 1 or both of these 2 additional packages apparently being crucial:
```
TASK [cmdsrv : Install packages for cmdsrv] ************************************
ok: [127.0.0.1] => (item=apache2)
ok: [127.0.0.1] => (item=php7.0)
ok: [127.0.0.1] => (item=php7.0-mysql)
changed: [127.0.0.1] => (item=php7.0-dev)
ok: [127.0.0.1] => (item=build-essential)
ok: [127.0.0.1] => (item=php-pear)
ok: [127.0.0.1] => (item=python-dev)
ok: [127.0.0.1] => (item=python-pip)
changed: [127.0.0.1] => (item=libapache2-mod-wsgi)
```
This occurs on Raspbian and possibly several other OS's.

Details of the failure:

> TASK [console : Restart web server] ********************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "Unable to restart service apache2: Job for apache2.service failed because the control process exited with error code.\nSee \"systemctl status apache2.service\" and \"journalctl -xe\" for details.\n"}

Attempting "systemctl restart apache2" did not work, yielding:

>root@raspberrypi:~# systemctl restart apache2
Job for apache2.service failed because the control process exited with error code.
See "systemctl status apache2.service" and "journalctl -xe" for details.

Re-running ./install (within /opt/iiab/iiab-admin-console) yielded this error, even earlier on:

> TASK [console : Enable ssl] ****************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "Error executing /usr/sbin/apache2ctl: [Wed Nov 08 12:39:03.685537 2017] [alias:warn] [pid 29241] AH00671: The Alias directive in /etc/apache2/sites-enabled/010-iiab.conf at line 271 will probably never match because it overlaps an earlier Alias.\n[Wed Nov 08 12:39:03.685710 2017] [alias:warn] [pid 29241] AH00671: The ScriptAlias directive in /etc/apache2/sites-enabled/010-iiab.conf at line 295 will probably never match because it overlaps an earlier ScriptAlias.\n[Wed Nov 08 12:39:03.686243 2017] [alias:warn] [pid 29241] AH00671: The ScriptAlias directive in /etc/apache2/sites-enabled/awstats.conf at line 42 will probably never match because it overlaps an earlier ScriptAlias.\nAH00526: Syntax error on line 1 of /etc/apache2/sites-enabled/cmd-service.conf:\nInvalid command 'WSGIScriptAlias', perhaps misspelled or defined by a module not included in the server configuration\n"}

tim-moody, 2017-11-08:
libapache2-mod-wsgi is installed by pathagar and osm, so if either was already installed this problem will not occur. my view is that it should be installed with apache. don't know about php7.0-dev. might have been used by the old php zmq wrapper.

tim-moody, 2017-11-08:
I'll put it in the installed list for cmdsrv for now

holta, 2017-11-08:
Presumed solved by https://github.com/iiab/iiab-admin-console/pull/42

(Plz re-open if issues arise on Raspbian etc!)

closed 2017-11-08 by holta