iiab/iiab #502: Ubuntu 17.10 (Artful Aardvark) Stretch Goal for IIAB 6.5?

holta, 2017-11-08:
Ubuntu 17.10 is NOT mainline (Ubuntu 16.04 LTS is our most popular OS today, after Raspbian!)  But still if this helps us all recruit new developers & implementers in preparing for April 2018's Ubuntu 18.04 LTS, that could be a very powerful thing.

As such the 3 people (@floydianslips, @deldesir, @jvonau) who are already making bits of progress of installing IIAB 6.5 pre-releases onto Ubuntu 17.10 should (please!) keep notes together here, for the sake of all.  So far this is all I know about, from the Oct 26 notes @ http://minutes.iiab.io:
* Create /opt/iiab/iiab/vars/ubuntu-17.yml by copying [vars/ubuntu-16.yml](https://github.com/iiab/iiab/blob/master/vars/ubuntu-16.yml)
* Insert the following line into [/opt/iiab/iiab/scripts/local_facts.fact](https://github.com/iiab/iiab/blob/master/scripts/local_facts.fact) :
`"ubuntu-17" | \`
* Comment out Lines 49-52 from roles/postgresql/tasks/main.yml (hard-coded PostgresSQL version, followed by “when: is_ubuntu”)
* ?

_Thanks everyone for adding below as progress accelerates/drags/etc &mdash; regardless whether this slips to [IIAB 6.6](https://github.com/iiab/iiab/milestone/3) or [IIAB 7.0](https://github.com/iiab/iiab/milestone/4) or possibly comes together faster than planned!_

floydianslips, 2017-11-09:
Started a new install on a 17.10 VM and now I can't even git clone the repo without my github credentials. This is the first time I have had to put them in when just doing a clone. Any ideas?

holta, 2017-11-10:
@floydianslips:

@jvonau has a theory that you accidentally typed "git push" instead of "git pull" or some such.

If that doesn't solve it, can you paste in the exact input+output?

deldesir, 2017-11-13:
Useful change when ./iiab-install failed at task 3 due to php7.0 issue, just change php7.0 to the actual version of php installed on your distro (should be php7.1 in ubuntu 17.10) in the ubuntu-17.yml file in vars.

deldesir, 2017-11-13:
I had to create the directory /etc/network/interfaces.d because Network Task failed when it doesn't find it. Now it stucks with the following error: "Could not find the requested service networking: host")

holta, 2017-11-13:
Thx @deldesir!

(1) feel free to submit a PR (Pull Request) if you know where this needs to be fixed?

Possibly places like:
* [roles/0-init/tasks/main.yml#L26](https://github.com/iiab/iiab/blob/master/roles/0-init/tasks/main.yml)
* [roles/0-init/tasks/main.yml#L167-L168](https://github.com/iiab/iiab/blob/master/roles/0-init/tasks/main.yml#L167-L168)

(2) @jvonau is away today until late evening, but can likely provide you a fix to /etc/network/interfaces.d particularly if he's able to log in remotely this machine over OpenVPN possible?  Consider `./runtags openvpn` or `./install-support` if so.  Your machine's name on the OpenVPN would be /etc/iiab/handle (create this file, and put any string in there as your machine name!)

deldesir, 2017-11-13:
Installing **ifupdown** solves the error: _Could not find the requested service networking: host._

holta, 2017-11-14:
I completed ./iiab-install and Admin Console's ./install and iiab/iiab-menu's ./cp-menus (and ALL steps from the usual http://download.iiab.io/6.5/rpi/load.txt) on @deldesir's Ubuntu 17.10 machine in Haiti by temporarily changing /opt/iiab/iiab/vars/local_vars.yml from:
```
calibre_install: True
calibre_enabled: True
```
to:
```
calibre_install: True
calibre_enabled: False
```
2 failures we need to look into:

(1) Calibre installed but wouldn't enable: (its flags need to be overhauled as Calibre moves from v2.55 to v3.7 as detailed below)

> 2017-11-13 15:40:12,979 p=14852 u=root |  TASK [calibre : Enable Calibre server] ***********************************************************************************************************************************
> 2017-11-13 15:40:13,844 p=14852 u=root |  fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "Unable to start service calibre-serve: Job for calibre-serve.service failed because the control process exited with error code.\nSee \"systemctl  status calibre-serve.service\" and \"journalctl  -xe\" for
 details.\n"}

Prior to setting "calibre_enabled: False" I tried "./runtags calibre" which also failed, with the same error above.

I was able to get `calibre_enabled: True` to work by:

A) changing [roles/calibre/templates/calibre-serve.service.j2 Line 9](https://github.com/iiab/iiab/blob/master/roles/calibre/templates/calibre-serve.service.j2#L9) from:

> ExecStart=/usr/bin/calibre-server --daemonize --pidfile=/var/run/calibre.pid --port={{ calibre_port }} **--with-library=**{{ calibre_dbpath }}

to:

> ExecStart=/usr/bin/calibre-server --daemonize  **--log=/var/log/calibre.log** --pidfile=/var/run/calibre.pid --port={{ calibre_port }} {{ calibre_dbpath }}

B) Creating a (mandatory) stub library by emulating https://www.digitalocean.com/community/tutorials/how-to-create-a-calibre-ebook-server-on-ubuntu-14-04#step-3-%E2%80%94-creating-the-library as follows
```
mkdir /library/calibre
mkdir /library/calibre/toadd
cd /library/calibre/toadd
wget http://www.gutenberg.org/ebooks/1342.kindle.noimages -O pride.mobi
wget http://www.gutenberg.org/ebooks/46.kindle.noimages -O christmascarol.mobi
calibredb add /library/calibre/toadd/* --with-library /library/calibre/
```
C) systemctl daemon-reload

D) ./runtags calibre

FYI `calibre-server --version` and `calibredb --version` both display VERSION 3.7 on Ubuntu 17.10, whereas on Ubunty 16.04 LTS they both display VERSION 2.55 which explains almost everything above!

(2) Visiting http://localhost/nextcloud puts you in a never-ending loop claiming either of the following:
```
Using http://localhost/nextcloud/index.php/login
```
or
```
http://localhost/nextcloud/index.php
```
Visiting http://box/nextcloud via VPN claims:
```
Forbidden
You don't have permission to access /nextcloud on this server.
```

holta, 2017-11-14:
Aside: should we merge PR #517 "roles/calibre/templates/calibre.conf legacy change from 8010 to 8080 ?" patching up Calibre's port...during proxying??

jvonau, 2017-11-14:
@holta  
> just change php7.0 to the actual version of php installed on your distro (php7.1 in my ubuntu 17.10) in the ubuntu-17.yml file in vars.

@deldesir  Thanks for the ifupdown hint, I'll add that to the package install list. I'll look at the machine when I get a chance later tonight, sounds like there was a shift in networking, moving more toward systemd or networkmanager. 

@holta  calibre_enabled: False" Think we are supplying a systemd unit file that is now wrong. Ill look at that in a bit.

holta, 2017-11-14:
> FYI `calibre-server --version` and `calibredb --version` both display VERSION 3.7 on Ubuntu 17.10, whereas on Ubunty 16.04 LTS they both display VERSION 2.55 which explains the syntax changes required by both above!

Also FYI we (IIAB) currently support calibre-server 2.75.1 and calibredb 2.75.1 on Raspbian...not through our own choosing...but that's the latest that's installed by https://raw.githubusercontent.com/kovidgoyal/calibre/master/setup/linux-installer.py within that we install from [roles/calibre/defaults/main.yml](https://github.com/iiab/iiab/blob/master/roles/calibre/defaults/main.yml)

jvonau, 2017-11-14:
looking at systemd-networkd on U17... looks 'fun'

> root@box:/opt/iiab/iiab# cat /etc/network/interfaces
`# /etc/network/interfaces -- configuration file for ifup(8), ifdown(8)
 #Generated by debian-installer.
 #The loopback interface
auto lo
iface lo inet loopback

source /etc/network/interfaces.d/*     <- missing from above...

jvonau, 2017-11-14:
2 wifi devices make this a bit more involved...
> root@box:/opt/iiab/iiab# ip link set up wlp2s0b1
RTNETLINK answers: Operation not possible due to RF-kill
brctl addbr br0   works   I think its the missing line so networking is lost without it... 

I need a live backup to work it out but from a quick look a round I thing that line might just be it as far as why br0 didn't come up. 

@deldesir  The 2 wifi need some feedback as to rfkill settings and intended use of the devices.

jvonau, 2017-11-14:
Um, no linux-installer.py  is for CentOS only, name: Install Calibre (OS's other than CentOS) uses the OS provided "packages"  while calibre-serve.service.j2 that we install to /etc/systemd/system/ as you say needs updating.

holta, 2017-11-14:
> [linux-installer.py](https://raw.githubusercontent.com/kovidgoyal/calibre/master/setup/linux-installer.py) is for CentOS only

OK, understood!

Whereas distros in Debuntu-family use the OS itself (Raspbian, Ubuntu, Debian) to recommend which version of calibredb & calibre-server to install, with Ansible installing the `calibre` package here: [roles/calibre/tasks/main.yml#L15-L21](https://github.com/iiab/iiab/blob/master/roles/calibre/tasks/main.yml#L15-L21)

_Would it be OK to change `state=present` to `state=latest` on Line 18 there to take advantage of Calibre's very latest improvements, as we do with WordPress and Nextcloud etc?_

holta, 2017-11-14:
@deldesir: can you help test networking across reboots today, keeping an eye on /etc/network/interfaces and/or /etc/network/interfaces.d/* ?  In consideration of @jvonau's networking questions for you above, Thanks!

jvonau, 2017-11-14:
At install time the newest packages would be installed anyway, there is no advantage to using "latest" here.

deldesir, 2017-11-14:
New bits of progress have been made for Ubuntu 17.10 thanks to @jvonau and @holta who provide support along the way. Now IIAB 6.5 is installed and is working.

deldesir, 2017-11-16:
New bits of progress have been made for Ubuntu 17.10 thanks to @jvonau and @holta who provide support along the way. Now IIAB 6.5 is up and running. Access Point is not stable. To temporary fix it, stop NetworkManager if it's running, then `ifconfig [wifi-adapter] up` and finally run `iiab-install`

jvonau, 2017-11-16:
I didn't think NetworkManager was involved, I can come up with a better
solution if given more time and further access

holta, 2017-11-16:
fyi @deldesir's Calibre 3.7.0 (installed by Ubuntu 17.10 with libGL.so.1) is further discussed @ https://github.com/iiab/iiab/pull/530#issuecomment-345032277 to try to get the Calibre playbook solid across many/most of the common IIAB Linux distros.

holta, 2017-11-20:
@deldesir & @floydianslips can you possibly test our much-enhanced IIAB master using http://download.iiab.io/6.5/rpi/load.txt or similar on Ubuntu 17.10, as we quickly approach RC1 / release now?

_Thanks if possible for testing the much-improved Calibre 3.12+ as part of your mix (http://box:8080), so we can close this ticket prior to US Thanksgiving (Thursday) if possibly, and this week for sure!_

holta, 2017-11-21:
@deldesir had success here with testing upgrading his Ubuntu 17.10 to the latest IIAB master...which all boils down to:
```
cd /opt/iiab/iiab/
git pull
./iiab-install --debug    (runs Stage 0, then Stages 3 thru 9/10)
./iiab-install --reinstall    (also worked; runs Stages 0 thru 9/10)

holta, 2017-11-21:
Huge Thanks @floydianslips if you hopefully get a chance to run thru an install of http://download.iiab.io/6.5/rpi/ onto Ubuntu 17.10 prior to this coming weekend!

MIN-sized, MEDIUM-sized, BIG-sized or ANY-sized :-)

holta, 2017-11-22:
FYI daily builds of Ubuntu 18.04 LTS "Bionic Beaver" are here:

http://cdimage.ubuntu.com/daily-live/current/ (to be released April 26, 2018) 

For those interested in helping to prepare for what follows *after* Ubuntu 17.10 (whose support/updates end July 2018, whereas Ubuntu 18.04 LTS support/updates end April 2023!)

jvonau, 2017-11-23:
https://wiki.ubuntu.com/ArtfulAardvark/ReleaseNotes#Network_configuration
https://wiki.ubuntu.com/Netplan

holta, 2017-11-23:
@deldesir & @floydianslips : is Nextcloud 12.0.3 (http://box/nextcloud) now working on Ubuntu 17.10 as part of IIAB 6.5, unlike [10 days ago](https://github.com/iiab/iiab/issues/502#issuecomment-344104061) ?

deldesir, 2017-11-24:
I cannot get past the login page.

holta, 2017-11-24:
> I cannot get past Nextcloud login page.

Are you using username: Admin and password: changeme ?

deldesir, 2017-11-24:
No I am not, sorry. Now I can confirm Nextcloud 12.0.3 works great on Ubuntu 17.10 with IIAB 6.5

holta, 2017-12-07:
IIAB 6.5 supports Ubuntu 17.10 very experimentally!

closed 2017-12-07 by holta