iiab/iiab #504: on deb 9.2 runansible stops at network restart

tim-moody, 2017-11-08:
******************************************
2017-11-08 16:57:50,321 p=13477 u=root |  fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "Unable to restart service networking: Job for networking.service failed because the control process exited with error code.\nSee \"systemctl status networking.service\" and \"journalctl -xe\" for details.\n"}
2017-11-08 16:57:50,326 p=13477 u=root |        to retry, use: --limit @/opt/iiab/iiab/iiab.retry

2017-11-08 16:57:50,326 p=13477 u=root |  PLAY RECAP ***********************************************************************************************************************************************************************************
2017-11-08 16:57:50,326 p=13477 u=root |  127.0.0.1                  : ok=221  changed=133  unreachable=0    failed=1
root@box:/opt/iiab/iiab# systemctl status networking.service
● networking.service - Raise network interfaces
   Loaded: loaded (/lib/systemd/system/networking.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-11-08 17:00:40 EST; 12min ago
     Docs: man:interfaces(5)
 Main PID: 388 (code=exited, status=1/FAILURE)

Nov 08 17:00:40 box.lan ifup[388]: Loading netfilter rules...run-parts: executing /usr/share/netfilter-persistent/plugins.d/15-ip4tables start
Nov 08 17:00:40 box.lan ifup[388]: run-parts: executing /usr/share/netfilter-persistent/plugins.d/25-ip6tables start
Nov 08 17:00:40 box.lan ifup[388]: done.
Nov 08 17:00:40 box.lan ifup[388]: ifup: missing required variable: address
Nov 08 17:00:40 box.lan ifup[388]: ifup: missing required configuration variables for interface br0/inet
Nov 08 17:00:40 box.lan ifup[388]: ifup: failed to bring up br0
Nov 08 17:00:40 box.lan systemd[1]: networking.service: Main process exited, code=exited, status=1/FAILURE
Nov 08 17:00:40 box.lan systemd[1]: Failed to start Raise network interfaces.
Nov 08 17:00:40 box.lan systemd[1]: networking.service: Unit entered failed state.
Nov 08 17:00:40 box.lan systemd[1]: networking.service: Failed with result 'exit-code'.

tim-moody, 2017-11-08:
had to reboot and then got a connection to the vm again

./runansible just stops with no message (guessing this is first_run)

jvonau, 2017-11-08:
Can you tell me a little more about how you have the VM configured? single interface/appliance? 2 interfaces bridged to host interfaces? The files in question would be /etc/network/interfaces and /etc/network/interfaces.d/iiab during networking restart. The detected_network and computed_network sections of the /etc/iiab/iiab.ini file would be helpful also.

tim-moody, 2017-11-08:
single interface/appliance

root@box:~# cat /etc/network/interfaces
# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

source /etc/network/interfaces.d/*

# The loopback network interface
auto lo
iface lo inet loopback

# The primary network interface
allow-hotplug enp0s3
iface enp0s3 inet dhcp

root@box:~# cat /etc/network/interfaces.d/iiab
# iiab_network_mode is Gateway
# gui_desired_network_role is Gateway

#################   LANCONTROLLER   ###################
auto br0
iface br0 inet static
    bridge_ports
    bridge_maxwait 0
    address 172.18.96.1
    netmask 255.255.224.0
    dns-nameservers 127.0.0.1
    dns-search lan

#################   GATEWAY or APPLIANCE #########################
# start dhcp clients
# client or user defined
# end dhcp clients
# static wan

tim-moody, 2017-11-08:
```
[detected_network]
has_ifcfg_gw = none
prior_gateway_(device_gw2) =
dhcpcd_result =
wan_in_interfaces = True
wireless_list_1(wifi1) = not found-1
wireless_list_2(wifi2) = not found-2
num_wifi_interfaces = 0
discovered_wireless_iface = none
discovered_wired_iface = none
num_lan_interfaces = 0
gui_static_wan = False
iiab_lan_iface = br0
iiab_wan_iface = enp0s3
[computed_network]
iiab_wan_enabled = True
user_wan_iface = enp0s3
iiab_wan_iface = enp0s3
iiab_lan_enabled = True
user_lan_iface = auto
iiab_lan_iface = br0
iiab_network_mode = Gateway
hostapd_enabled = True
host_ssid = Internet in a Box
host_wifi_mode = g
host_channel = 6
[network]
iiab_network_mode_applied = Gateway
dhcp_service2 = dhcpd
dnsmasq_enabled = False
no_net_restart = False

```

jvonau, 2017-11-09:
Something is weird:
discovered_wireless_iface = none
discovered_wired_iface = none
num_lan_interfaces = 0
iiab_lan_iface = br0     <-- I would expect to see none here.

jvonau, 2017-11-09:
Quick workaround would be to add to vars/local_vars.yml:  is_VM: True until I can look at your install.log to see which rule was matched in computed_network.yml to fully resolve this.

jvonau, 2017-11-09:
I should add /etc/network/interfaces.d/iiab would need to be removed.

jvonau, 2017-11-09:
Unclear which rule had set br0 in network/tasks/detected_network.yml without the log as the number of interfaces restriction should of been met that is the main issue. During debian.yml br0 would of been matched resulting in /etc/network/interfaces.d/iiab being written with bridge_ports being correctly blank. I've used is_VM in jenkins to fake a single interface because 4 are detected, resulting in the desired appliance mode with no /etc/network/interfaces.d/iiab present. See PR above for proposed fix as that shifts to actually setting the fact to none although a warning vnstat might not like none.

tim-moody, 2017-11-09:
https://pastebin.com/6KJZ01Si

tim-moody, 2017-11-09:
any machine with only one interface could have this problem, not just a vm.

jvonau, 2017-11-09:
Yes I know, lack of a test subject hampered further debugging earlier. I've adjusted the # below for readability
> 2017-11-08 16:57:30,822 p=13477 u=root |  TASK [network : Blacklisted WiFi adapter] 
2017-11-08 16:57:30,850 p=13477 u=root |  skipping: [127.0.0.1]
2017-11-08 16:57:30,860 p=13477 u=root |  TASK [network : Create list of LAN (non WAN) ifaces] *
2017-11-08 16:57:31,045 p=13477 u=root |  fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": "ls /sys/class/net | grep -v -e wwlan -e ppp -e lo -e br0 -e tun -e enp0s3 -e none", "delta": "0:00:00.004453", "end": "2017-11-08 16:57:31.024828", "failed": true, "msg": "non-zero return code", "rc": 1, "start": "2017-11-08 16:57:31.020375", "stderr": "", "stderr_lines": [], "stdout": "", "stdout_lines": []}
2017-11-08 16:57:31,045 p=13477 u=root |  ...ignoring
2017-11-08 16:57:31,059 p=13477 u=root |  TASK [network : Count LAN ifaces] 
2017-11-08 16:57:31,239 p=13477 u=root |  ok: [127.0.0.1]
2017-11-08 16:57:31,248 p=13477 u=root |  TASK [network : Calulate number of LAN interfaces including WiFi] 
2017-11-08 16:57:31,287 p=13477 u=root |  ok: [127.0.0.1]
2017-11-08 16:57:31,297 p=13477 u=root |  TASK [network : Set discovered_wired_iface fact if present] 
2017-11-08 16:57:31,336 p=13477 u=root |  TASK [network : Set discovered_wireless_lan_iface fact if present] 
2017-11-08 16:57:31,362 p=13477 u=root |  skipping: [127.0.0.1]
2017-11-08 16:57:31,377 p=13477 u=root |  TASK [network : Set variable discovered_wired_lan_iface if present] 
2017-11-08 16:57:31,407 p=13477 u=root |  skipping: [127.0.0.1]
2017-11-08 16:57:31,417 p=13477 u=root |  TASK [network : 2 or more devices on the LAN - use bridging] 
2017-11-08 16:57:31,459 p=13477 u=root |  ok: [127.0.0.1]
2017-11-08 16:57:31,469 p=13477 u=root |  TASK [network : For Debian, always use bridging - except RPi] 
2017-11-08 16:57:31,511 p=13477 u=root |  ok: [127.0.0.1]
2017-11-08 16:57:31,523 p=13477 u=root |  TASK [network : WiFi is on the LAN - use bridging - except RPi] 
2017-11-08 16:57:31,550 p=13477 u=root |  skipping: [127.0.0.1]
2017-11-08 16:57:31,562 p=13477 u=root |  TASK [network : Setting wired LAN as only interface - RPi] 
2017-11-08 16:57:31,591 p=13477 u=root |  skipping: [127.0.0.1]
2017-11-08 16:57:31,604 p=13477 u=root |  TASK [network : Setting wireless LAN as only interface - RPi] 
2017-11-08 16:57:31,634 p=13477 u=root |  skipping: [127.0.0.1]
2017-11-08 16:57:31,647 p=13477 u=root |  TASK [network : in VM disable LAN - needs local_vars entry to activate] 
2017-11-08 16:57:31,674 p=13477 u=root |  skipping: [127.0.0.1]
2017-11-08 16:57:31,684 p=13477 u=root |  TASK [network : Has old gateway and no discovered gateway setting WAN] 
2017-11-08 16:57:31,715 p=13477 u=root |  skipping: [127.0.0.1]
2017-11-08 16:57:31,725 p=13477 u=root |  TASK [network : Add location section to config file] 

These two are firing: 
> 2017-11-08 16:57:31,417 p=13477 u=root | TASK [network : 2 or more devices on the LAN - use bridging]
2017-11-08 16:57:31,459 p=13477 u=root | ok: [127.0.0.1]
2017-11-08 16:57:31,469 p=13477 u=root | TASK [network : For Debian, always use bridging - except RPi]
2017-11-08 16:57:31,511 p=13477 u=root | ok: [127.0.0.1]

Now that means I've broken the number comparison, something changed, or it didn't work quite right to start with as all RPi's always has 2 or more interfaces and never really got tested well. In the short term try the is_VM workaround to enable the install to complete with further debugging available post install via the vpn for final resolution.

jvonau, 2017-11-09:
Going with the broken numbers theme, think we have to declare an int to be an int every time and not just when recorded at num_lan_interfaces so as a possible alternative have a look at
https://github.com/jvonau/iiab/commit/7fb7eb9c0f790d38c444999fc6a8cab9caa1276e

jvonau, 2017-11-09:
fixed with #507

closed 2017-11-09 by holta

jvonau, 2017-11-10:
Without have a single interface test subject I've never seen this one:
> 2017-11-09 12:10:32,106 p=846 u=root |  TASK [network : Blacklisted WiFi adapter] 
2017-11-09 12:10:32,130 p=846 u=root |  skipping: [127.0.0.1]
2017-11-09 12:10:32,139 p=846 u=root |  TASK [network : Create list of LAN (non WAN) ifaces] 
2017-11-09 12:10:32,299 p=846 u=root |  fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": "ls /sys/class/net | grep -v -e wwlan -e ppp -e lo -e br0 -e tun -e enp0s3 -e none", "delta": "0:00:00.004255", "end": "2017-11-09 12:10:32.281816", "failed": true, "msg": "non-zero return code", "rc": 1, "start": "2017-11-09 12:10:32.277561", "stderr": "", "stderr_lines": [], "stdout": "", "stdout_lines": []}
2017-11-09 12:10:32,299 p=846 u=root |  ...ignoring
2017-11-09 12:10:32,309 p=846 u=root |  TASK [network : Count LAN ifaces] 
2017-11-09 12:10:32,470 p=846 u=root |  ok: [127.0.0.1]

If we change the order here and create a conditional using 'Count LAN ifaces' return value as a means to suppress blindly looking for interface names we know are not present this  ...ignoring workaround can be removed. Included in #508 and is jenkins tested