iiab/iiab #512: DanGuardian blocks BIG-sized many/most IIAB installs?

holta, 2017-11-11:
### Expected behavior and actual behavior.

TASK [network : Enable DansGuardian] *******************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "failed": true, "msg": "Could not find the requested service dansguardian: host"}

### Steps to reproduce the problem.

Run "./iiab-install --reinstall" after upgrading a standard install from [MEDIUM-sized local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml) to [BIG-sized local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars_big.yml).

This same problem occurred when upgrading a [MIN-sized local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars_min.yml) to BIG-sized.

### Specifications like the version of the IIAB, operating system version, or hardware details.

IIAB 6.5 master on Raspbian Lite 2017-09-07 on RPi3.

holta, 2017-11-11:
> some people sincerely believe that Squid and DansGuardian should both be deprecated.  Which may well be a thoughtful + constructive way forward to focus on more modern grassroots use cases.  In any case, we'll discuss different perspectives on Bit Rot versus Wise Metrics For Amputation (etc) during our Monday community/team call @ http://minutes.iiab.io -- so we're not shooting ourselves in the foot with nearly perpetual maintenance of rarely use packages.

(Excerpt from http://lists.laptop.org/pipermail/server-devel/2017-November/008221.html)

jvonau, 2017-11-12:
Upgrading post install has always been problematic that is why in the past it was the kitchen sink approach with no option not to install to install some services while default_vars just lied about being able to do so, just leaving the services disabled. Now with the slimmed down installs in place moving to big should really be a install time option by starting with "big" or "from scratch". There is no issue if you start with the services to be installed from the start, and with a bit of jiggling I can restore the selective post install option, I'm debating if it's worth the effort.

jvonau, 2017-11-12:
The fix is in #508 f4c0e3b6914e9122025a010ba2d0819196abe0db posted long before this issue was recorded.

holta, 2017-11-13:
Apparently fixed.  We can re-open if nec.

closed 2017-11-13 by holta