iiab/iiab #518: Should "Install packages (debuntu)" & "Update common packages (debuntu)" be merged in roles/2-common/tasks/packages.yml ?

holta, 2017-11-14:
Or are these 2 sections really distinct for a good reason in [master/roles/2-common/tasks/packages.yml](https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/packages.yml) ?

[Line 26-35](https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/packages.yml#L26-L35) use:
```
state=present
```
[Lines 95-104](https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/packages.yml#L95-L104) use:
```
state=latest
```
_Or should both sections be merged into 1 single section, installing all Debuntu-family packages?_

jvonau, 2017-11-14:
Use 'present' only would be my vote as the docs say to start with a fully updated and rebooted machine.

holta, 2017-11-20:
PR #542 merged after carefuly review with @georgejhunt, @tim-moody, @jvonau, @holta during community/team call @ http://minutes.iiab.io

If future exFAT packages might be moved down from "Install common packages" for CentOS/Fedora etc possibly, instead of up top in "Install packages (debuntu)".

closed 2017-11-20 by holta