iiab/iiab #529: Calibre 2.x images don't appears at http://box/books (works with Calibre 3.x on Ubuntu 17.10)

holta, 2017-11-14:
In Calibre 3.x, image appears properly in the nested directory, e.g. book title images appear beautifully from image URL's like:
http://box/books/get/thumb/1/calibre?sz=300x400
So there is no need to use http://box:8080 anymore going forward!  Or so I thought!  But...

In Calibre 2.x this is not working: while the main page http://box/books loads, all of its images are trying to load from the wrong place, e.g.
http://192.168.0.123/static/calibre.png

The image itself is where it should be!  e.g.
http://192.168.0.123/books/static/calibre.png

_Is our own [calibre.conf logic creating the http://box/books mnemonic for Apache](https://github.com/iiab/iiab/blob/master/roles/calibre/tasks/main.yml#L23-L39) in any way to blame &mdash; or is this a Calibre 2.x bug possibly?_

PS "calibre-server --version" and "calibredb --version" show 3.7 on Ubuntu 17.10 &mdash; but show 2.55 on Ubuntu 16.04 LTS, and shows 2.75.1 on the latest/updated Raspbian Stretch.  Ref #528

holta, 2017-11-14:
@darkenvy you can prob solve this in your sleep...so sleep well in the coming week and hopefully you can solve this by installing any [IIAB 6.5 master 1-line installer](http://download.iiab.io/6.5/rpi/) or "git pull" on a recent one :)

jvonau, 2017-11-15:
Think we might need to state a rule for each of the sub-directories  ProxyPassReverse /calibre/static http://localhost:{{}}/calibre/static going by kiwix's iiab-make-apache-config.py

jvonau, 2017-11-15:
Quick test of /books on debian-9.1 VM  calibre-server (calibre 2.75.1) has a css issue while :8080 renders cleanly.

holta, 2017-11-15:
It turns out Calibre 3.x has issues on http://box/books too.

There are no problems on http://box:8080 thankfully!

More evolving at PR #530

holta, 2017-11-19:
FYI Calibre 3.x (3.12) @ http://box/books fails to load completely in Internet Explorer 11

FYI Calibre 3.x (3.12) @ http://box/books seems to appears in Firefox 57.0 but e-books essentially all fail to display as follows:

> InvalidStateError: responseText is only available if responseType is '', 'document', or 'moz-chucked-text'.  Error at books:4830:  Show details

holta, 2017-11-20:
FYI Calibre 3.2 (3.12) @ http://box/books also seems to appear in Firefox 47 (similar to 57 above) but e-books fails as follows:

> InvalidStateError: An attempt was made to use an object that is not, or is no longer, usable
Error at books:4830:
Traceback (most recent call last):
>
> ajax/</ρσ_anonfunc@http://10.8.0.74/books:4905:21
complete_callback@http://10.8.0.74/books:4874:21
set_error@http://10.8.0.74/books:4830:21
"ajax/</ρσ_anonfunc@http://10.8.0.74/books:4905:21
complete_callback@http://10.8.0.74/books:4874:21
set_error@http://10.8.0.74/books:4830:21"

jvonau, 2017-11-20:
> Error: Unhandled error
InvalidStateError: An attempt was made to use an object that is not, or is no longer, usable
Error at books:4830:
Traceback (most recent call last):
ajax/</ρσ_anonfunc@http://10.8.0.74/books:4905:21
complete_callback@http://10.8.0.74/books:4874:21
set_error@http://10.8.0.74/books:4830:21

holta, 2017-11-21:
Small issue!  But if the above errors within http://box/books aren't fixable for IIAB 6.5's final release in the coming days &mdash; let's revert->revise these changes from 6 days ago:

- [local_vars.yml](https://github.com/iiab/iiab/commit/a7872757f046d2e8470379ffd39fe60fd65a436b#diff-e80e2d6e1381eafe7da36ba77df65701) (also @ http://wiki.laptop.org/go/IIAB/local_vars.yml)
- [default_vars.yml](https://github.com/iiab/iiab/commit/c2e047fd57355a19bd65d84fdbe31fcb207d96dc#diff-0cb5210e140dd4dc63a39c47bf909cda)

jvonau, 2017-11-22:
VM error from #530, requirement to have a desktop installed first [is back](http://10.8.0.22:8080/job/Mem-change-test/113/console). In the past this dependency was met when using the stock .deb packaging as a desktop got dragged in. Need to install calibre or something in its place to have this python installer work cleanly on a headless install so debian-9 is broken also. Using calibre 3.12 debs using the same mod as the rpi uses is an option for debian-9 but with the same pitfalls I discussed.

holta, 2017-11-22:
> VM error from #530, requirement to have a desktop installed first [is back](http://10.8.0.22:8080/job/Mem-change-test/113/console).

Pasting in above error: (as these Jenkins.io build logs can sometimes disappear!)

> 16:13:04 ==> ubuntu: TASK [calibre : Incorporate sample book into Calibre DB (mandatory since Calibre 3.x)] ***
16:13:04 ==> ubuntu: fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "calibredb add /library/downloads/Metamorphosis-jackson.epub --with-library /library/calibre", "delta": "0:00:00.177246", "end": "2017-11-22 00:13:04.942329", "failed": true, "msg": "non-zero return code", "rc": 1, "start": "2017-11-22 00:13:04.765083", "stderr": "Traceback (most recent call last):\n  File \"site.py\", line 75, in main\n  File \"site-packages/calibre/db/cli/main.py\", line 17, in <module>\n  File \"site-packages/calibre/db/legacy.py\", line 18, in <module>\n  File \"site-packages/calibre/db/backend.py\", line 33, in <module>\n  File \"site-packages/calibre/utils/img.py\", line 18, in <module>\nImportError: libGL.so.1: cannot open shared object file: No such file or directory", "stderr_lines": ["Traceback (most recent call last):", "  File \"site.py\", line 75, in main", "  File \"site-packages/calibre/db/cli/main.py\", line 17, in <module>", "  File \"site-packages/calibre/db/legacy.py\", line 18, in <module>", "  File \"site-packages/calibre/db/backend.py\", line 33, in <module>", "  File \"site-packages/calibre/utils/img.py\", line 18, in <module>", "ImportError: libGL.so.1: cannot open shared object file: No such file or directory"], "stdout": "", "stdout_lines": []}

holta, 2017-11-23:
Above is NOT resolved but a partial workaround & documentation (PR #555) has been merged during our live community/team call @ http://minutes.iiab.io

holta, 2018-07-18:
Attempting http://box/books with Calibre 3.27.1 on IIAB 6.6/master on Ubuntu 18.04 (or Raspbian) the error now reads:
```
Failed to communicate with "/books/interface-data/init?1531946630827", with status: [400] Bad Request

The uri in the Request Line and the Authorization header do not match
```

holta, 2018-07-19:
/var/log/apache2/error.log shows nothing but am pasting in the results of 2 log files if it helps:
```
root@box:~# tail /var/log/calibre.log
[...]
calibre server listening on 0.0.0.0:8080
OPDS feeds advertised via BonJour at: 192.168.0.121 port: 8080
Authorization URI mismatch: (u'interface-data', u'init') != (u'books', u'interface-data', u'init') from client: 127.0.0.1
```
```
root@box:~# tail -20 /var/log/apache2/access.log
[...]
127.0.0.1 - - [19/Jul/2018:10:50:05 -0400] "GET /server-status?auto HTTP/1.1" 200 1392 "-" "munin/2.0.37-1ubuntu0.1 (libwww-perl/6.31)"
127.0.0.1 - - [19/Jul/2018:10:50:06 -0400] "GET / HTTP/1.1" 302 525 "-" "Munin - http_loadtime"
127.0.0.1 - - [19/Jul/2018:10:50:06 -0400] "GET /home HTTP/1.1" 301 562 "-" "Munin - http_loadtime"
127.0.0.1 - - [19/Jul/2018:10:50:06 -0400] "GET /home/ HTTP/1.1" 200 7246 "-" "Munin - http_loadtime"
127.0.0.1 - - [19/Jul/2018:10:50:07 -0400] "GET /server-status?auto HTTP/1.1" 200 1391 "-" "munin/2.0.37-1ubuntu0.1 (libwww-perl/6.31)"
127.0.0.1 - - [19/Jul/2018:10:50:08 -0400] "GET /server-status?auto HTTP/1.1" 200 1392 "-" "munin/2.0.37-1ubuntu0.1 (libwww-perl/6.31)"
192.168.0.111 - - [19/Jul/2018:10:50:47 -0400] "GET /books HTTP/1.1" 200 276958 "-" "Mozilla/5.0 (Windows NT 6.3; Win64; x64; rv:61.0) Gecko/20100101 Firefox/61.0"
192.168.0.111 - - [19/Jul/2018:10:50:48 -0400] "GET /books/ajax-setup?1532011841330 HTTP/1.1" 200 420 "http://192.168.0.121/books" "Mozilla/5.0 (Windows NT 6.3; Win64; x64; rv:61.0) Gecko/20100101 Firefox/61.0"
192.168.0.111 - - [19/Jul/2018:10:50:48 -0400] "GET /calibre.appcache HTTP/1.1" 404 508 "http://192.168.0.121/books" "Mozilla/5.0 (Windows NT 6.3; Win64; x64; rv:61.0) Gecko/20100101 Firefox/61.0"
192.168.0.111 - - [19/Jul/2018:10:50:48 -0400] "GET /books/interface-data/init?1532011841334 HTTP/1.1" 401 400 "http://192.168.0.121/books" "Mozilla/5.0 (Windows NT 6.3; Win64; x64; rv:61.0) Gecko/20100101 Firefox/61.0"
192.168.0.111 - - [19/Jul/2018:10:50:48 -0400] "GET /favicon.png HTTP/1.1" 404 503 "-" "Mozilla/5.0 (Windows NT 6.3; Win64; x64; rv:61.0) Gecko/20100101 Firefox/61.0"
192.168.0.111 - - [19/Jul/2018:10:50:48 -0400] "GET /favicon.png HTTP/1.1" 404 504 "-" "Mozilla/5.0 (Windows NT 6.3; Win64; x64; rv:61.0) Gecko/20100101 Firefox/61.0"
192.168.0.111 - - [19/Jul/2018:10:50:48 -0400] "GET /favicon.png HTTP/1.1" 404 503 "-" "Mozilla/5.0 (Windows NT 6.3; Win64; x64; rv:61.0) Gecko/20100101 Firefox/61.0"
192.168.0.111 - - [19/Jul/2018:10:50:54 -0400] "GET /books/interface-data/init?1532011841334 HTTP/1.1" 400 268 "http://192.168.0.121/books" "Mozilla/5.0 (Windows NT 6.3; Win64; x64; rv:61.0) Gecko/20100101 Firefox/61.0"
```

holta, 2018-08-15:
@arky is it possible we can bypass the need for #529 here in the end &mdash; using http://box/calibre-web (abbreviated to http://box/books etc) as described in #816 ?

holta, 2018-08-23:
Let's strongly consider http://box/books as a universal teacher's shortcut to calibre-web if PR #999 [New calibre-web playbook (Fixes #816)] works out?!

jvonau, 2020-04-29:
works with nginx

holta, 2020-04-30:
> works with nginx

Can you recommend a `/etc/nginx/conf.d/calibre-nginx.conf` that would make http://box/calibre work?

Related: PR #2378 "Cleaner Calibre role (its conversion tool can supplement Calibre-Web)"

holta, 2020-04-30:
> Can you recommend a `/etc/nginx/conf.d/calibre-nginx.conf` that would make http://box/calibre work?

That would be best, if anyone can suggest what NGINX file to put in place above!

( In the past https://github.com/iiab/iiab/blob/master/roles/calibre/templates/calibre.conf never worked, even when it is in fact put in place at /etc/apache2/sites-available/calibre.conf -- as is currently put in place by https://github.com/iiab/iiab/blob/master/roles/calibre/tasks/install.yml#L46 )

jvonau, 2021-08-29:
#2739

jvonau, 2025-01-27:
status?

holta, 2025-01-27:
Calibre-Web is now much more important than Calibre, for IIAB schools and communities.

closed 2025-01-27 by holta