iiab/iiab #547: Rename "network" to "10-network" & adjust logic in ./iiab-install ?

holta, 2017-11-20:
If the new approach offered by #537 is looking solid across many OS's & many scenarios in coming days?

holta, 2017-11-20:
Somewhat/tangentially related:
#534 "document "reserved_wifi: wlan0" + iiab-hotspot-on / off + similar"
#531 (comment) "revision2: iiab_base_ver, iiab_revision & reserved_wifi"

tim-moody, 2017-11-20:
I would make it 10-finalize or something more generic. It becomes the place where things that must be done last are found.

holta, 2017-11-20:
> I would make it 10-finalize or something more generic. It becomes the place where things that must be done last are found.

That was kinda the purpose of 9-local despite its name.

Is it too crazy to go with something like 9-network and 10-finalize ?

Or...9-local, 10-network and 11-finalize?

@jvonau & @georgejhunt will have ideas as to what's tolerable / wise.

tim-moody, 2017-11-20:
As the one who named 9-addons I can attest that finalize was not its purpose. It was meant to be a home for non-canonical additions. Perhaps we should just put a number on all roles.

holta, 2017-11-25:
Another possibility might to have 2 different network sections&mdash;before and after server apps are installed&mdash;something like:

- 5-network
- 6-xo-services
- 7-diy-apps
- 8-edu-apps
- 9-mgmt-tools
- 10-local-addons
- 11-network

Idea: Implementers can greatly benefit if networking is (largely!) already in place when server apps are being installed, for incremental sanity checks during installation, and even more so when there is a glitch (inevitably!) that interrupts ./iiab-install.

Then "Networking QA/Validation" can happen in Stage 11?

If any of this is a good idea: "./iiab-network" and "./runtags network" could execute both networking sections (and more?) based on Ansible tags.

And 11-network could of course be renamed to 11-finalize, or to any other better name...

jvonau, 2017-11-25:
Your "idea" is already in practice, with the installation of named/dhcpd/squid/DG(dependencies) during 4-server-options you effectively have 5-network already, the actual configuration of the physical layout of the network held until the last step as above. Can you expand on what the are 'benefits' you seek?

holta, 2017-11-25:
I could be wrong, but I thought one of the big ideas here was to move networking affairs into their own Stage(s), splitting networking out from Stage 4, so that Ansible's tags work better, e.g. with "./iiab-network" and ".runtags network".

If that's true, why not put networking into its own sections around 5-network (to do the bulk of the work, allowing complete testing of server apps if Ansible fails between Stages 5-and-11) and then also 11-network (to validate all networking, ensuring server apps haven't screwed anything up etc...)

Significant Bonus: new contributors will be ecstatic with more rapid Ansible/workflow understanding and onboarding if all 10-to-12 Stages are tightly/descriptively named.

holta, 2017-12-15:
FYI we tried hard to reach consensus here during 2 separate calls over the past 2 weeks, and have made very real progress achieving joint understanding several of the many underlying challenges — but still consensus is still lacking in a couple areas, with strong opinions expressed — so further discussion is clearly needed before a meaningful consensus is achievable here!

holta, 2018-02-15:
Am closing this ticket as we need to move towards IIAB 6.5 release.

It can be re-opened later if truly nec!

closed 2018-02-15 by holta