iiab/iiab #551: how to mitigate campuswide failures for Nextcloud, WordPress, Moodle, Elgg etc

holta, 2017-11-21:
The challenge is to get the most urgently needed campuswide LMS-like apps working **well enough** for campus-like environments (rather than perfectly in all situations, not possible!)

Typically these server apps fail when accessed over the WAN side of an Internet-in-a-Box, but work well over the LAN side or Internet WiFi of an Internet-in-a-Box.

_Many tricks exist to overcome these common failures, e.g.:_
* WordPress must be configured from the 1st moment using the correct network &mdash; as it locks in all future URL's to that FQDN or IP address, such as http://hogwarts/wordpress or whatever you-as-administrator use during your 1st visit to the site &mdash; whereupon http://box/wordpress and others will not work from that point onwards :-(
* Nextcloud also blocks campuswide access in its own way, but this can sometimes (?) be overcome using:
  * /etc/apache2/sites-available/nextcloud.conf (commenting out Line 9 (that begins with "Require" was not enough!)
  * /opt/nextcloud/config/config.php
  * /opt/nextcloud/config/.htaccess
* ETC!

_The challenge is to inventory -> document at http://FAQ.IIAB.IO all these annoying tricks so implementers can work with developers to truly deliver!_

holta, 2017-11-21:
To help us all keep these tricks growing in 1 place, findable by all &mdash; allowing implementers to work with us building a sane implementation roadmap &mdash; this seed is placed in http://FAQ.IIAB.IO as Article 27:

> "Is campuswide access possible for all server apps?"
http://wiki.laptop.org/go/IIAB/FAQ#Is_campuswide_access_possible_for_all_server_apps.3F

holta, 2017-11-21:
Ref #111 "nextcloud not accessible from vpn"

holta, 2017-12-15:
While critically important for larger schools etc, I'm deferring this to IIAB 6.6 so that IIAB 6.5 can be finalized very shortly now.

holta, 2023-01-08:
Please see the latest at [FAQ.IIAB.IO](https://wiki.iiab.io/go/FAQ) and similar places:

- [Matomo install requirements with IIAB](https://github.com/iiab/iiab/tree/master/roles/matomo#install-it)
- [Is campuswide access possible for all server apps?](https://wiki.iiab.io/go/FAQ#Is_campuswide_access_possible_for_all_server_apps%3F)
- [WordPress & Moodle Administration: What tips & tricks exist?](https://wiki.iiab.io/go/FAQ#WordPress_.26_Moodle_Administration:_What_tips_.26_tricks_exist%3F)

_Finally, this problem should no longer affect Nextcloud installed with IIAB._ :heavy_check_mark:

jvonau, 2025-01-27:
status?

holta, 2025-01-27:
Generally good enough all around, as is.

closed 2025-01-27 by holta