iiab/iiab #552: Remove networking pre-req "ifupdown"

holta, 2017-11-21:
This seems to be in [roles/2-common/tasks/packages.yml](https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/packages.yml#L35) and possibly other places?

Suggested by @jvonau.

jvonau, 2017-11-22:
Only place, #502 is where it was noted that should ifupdown not be installed networking failed to setup correctly as this package was absent. I suspect systemd-networkd or networkmanger is running the network configuration, lots of NM code is peppered about the code in the redhat side. Looked at what it might take to port over to systemd-networkd with a [quick layout](https://github.com/iiab/iiab/compare/master...jvonau:systemd-net?expand=1)

jvonau, 2017-11-23:
https://wiki.ubuntu.com/ArtfulAardvark/ReleaseNotes#Network_configuration
https://wiki.ubuntu.com/Netplan

holta, 2017-12-07:
Done, as part of #554 etc!

closed 2017-12-07 by holta