iiab/iiab #558: WiFi-to-Internet fails regularly & doesn't recover on Raspbian Lite

holta, 2017-11-24:
Raspbian Desktop 2017-09-07 (updated with "apt update" & "apt dist-upgrade") does not seem to have this problem, reconnecting itself to WiFi/Internet automatically/regularly as necessary, all on its own.

Raspbian Lite 2017-09-07 (also updated with apt) on the other hand, does not seem to hold onto it WiFi connections to the internet :-(

@jvonau suspects this is related to dhcpcd having "7200 sec" (2 hour leases) on the WiFi router in the building where this was tested.

Sounds very plausible: but do we know why Raspbian Desktop seems to be able to renew its DHCP lease & WiFi connection every 2 hours, whereas Raspbian Lite completely loses its Internet/WiFi connection?

holta, 2017-11-24:
In comparison, a new/untouched Raspbian Lite seems to contain the following stub in /etc/wpa_supplicant/wpa_supplicant.conf :
```
country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
```

holta, 2017-12-04:
GOOD NEWS: The new raspi-config from 20171127 allows headless configuration of SSID/psk (private shared key) WiFi, under "2 Network Options" -> "N2 Wi-fi" / "Enter SSID and passphrase" !

BAD NEWS: This just missed the cut for Raspbian 2017-11-29, so an "apt update" then "apt dist-upgrade" are needed a for now (until the net release of Raspbian in coming weeks/months) so this is not currently feasible for folks who aren't fluent enough to use to hand-configure a minimal wpa_supplicant.conf as follows: (e.g. using using "nano /etc/wpa_supplicant/wpa_supplicant.conf")
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=GB

network={
        ssid="ACTUAL_SSID"
        psk="ACTUAL_PASSWORD"
```

holta, 2017-12-05:
> BAD NEWS: This just missed the cut for Raspbian 2017-11-29, so an "apt update" then "apt dist-upgrade" are needed a for now (until the next release of Raspbian in coming weeks/months) so... this is not YET currently feasible for folks who aren't fluent enough to use to hand-configure a minimal wpa_supplicant.conf as follows: (e.g. using using "nano /etc/wpa_supplicant/wpa_supplicant.conf")

CORRECTION / GREAT NEWS: raspi-config 20171109 with SSID/passphrase picker is part of Raspbian Desktop 2017-11-29 (and hopefully Raspbian Lite 2017-11-29) !

ASIDE: Raspbian Lite _seems_ substantially more prone to WiFi failures as compared to Raspbian Desktop.  This pattern has occurred repeatedly in 2 separate homes, with several very different RPi3 units.  I'll continue to monitor/confirm/deny if indeed this is the case &mdash; so we can improve the reliability of Raspbian Lite as a WiFi client machine regardless &mdash; during IIAB installations especially!

holta, 2017-12-07:
CONFIRMED GOOD: raspi-config works to set SSID/passphrase even on Raspbian Lite 2017-11-29!  As I've documented @ https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch

Aside: the raspi-config pkf itself has since been updated again, to "stable" version 20171201, if you run 'apt update' then 'apt dist-upgrade'.

holta, 2017-12-07:
Rebooting the RPi3 always fixes the WiFi (in both homes where Raspbian Lite's WiFi regularly drops its connection, not so may minutes after it was working perfectly!)

It definitely feels like a protocol failure (or certainly protocol weakness) is involved, as a reboot always restores the WiFi connection to the Internet.

[FYI all RPi3's contained the very latest master / pre-release of IIAB 6.5 during these ongoing tests.]

If it helps, the output of "ip a" prior to reboot is typically something like:
```
root@box:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group defaul
t qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen
 1000
    link/ether b8:27:eb:73:7c:ab brd ff:ff:ff:ff:ff:ff
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DO
WN group default qlen 1000
    link/ether b8:27:eb:26:29:fe brd ff:ff:ff:ff:ff:ff
```
After reboot, the output is typically something like:
```
root@box:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether b8:27:eb:73:7c:ab brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:26:29:fe brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.102/24 brd 192.168.0.255 scope global wlan0
       valid_lft forever preferred_lft forever
4: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 100
    link/none
    inet 10.8.0.70 peer 10.8.0.69/32 scope global tun0
       valid_lft forever preferred_lft forever
```
I will try a more expensive WiFi router as necessary?  Towards verifying whether there might possibly be firmware/incompatility issues with RPi3's WiFi module, or whatever the root cause(s) may be uncoverable!

Also I'd like strong, more scientific confirmation that "Raspbian With Desktop" is better at holding onto its WiFi connection to the Internet, which so far seems to be the case... (whether or not it's a DHCP lease renewal issue, after the 120min leases run out, as assigned by the WiFi router here!)

holta, 2017-12-08:
Observation: RPi3's WiFi connection [to the home's router, and onto the Internet] is much more stable if it's actually used, e.g. during a long run of ./iiab-install

(Whereas the RPi3's WiFi connection fails often when it's left unused, often within 10min or so of inactivity...at which point "ping 8.8.8.8" and "ping mit.edu" etc generally all fail.)

holta, 2017-12-08:
This workaround brings "ping 8.8.8.8" and "ping mit.edu" back to life:
```
systemctl restart dhcpcd
```
Thanks @jvonau for pointing the way.

_Do you know if this extreme flakiness (without any attempt to restore the WiFi connection) really the expected behavior of dhcpcd on a consumer OS like Raspbian?_

georgejhunt, 2017-12-11:
Adam -- since you have a failing situation, try commenting line 41 of /etc/dhcpcd.conf which tells dhcpcd to ignore wlan0. 
It seems to me that "hotspot-on" and "hotspot-off" would need to diddle with line 41 referenced.

holta, 2017-12-11:
> Adam -- since you have a failing situation, try commenting line 41 of /etc/dhcpcd.conf which tells dhcpcd to ignore wlan0.

That worked to cause eth0 to get an IP address!!

(Is PR #617 no longer necessary, or still a good idea?)

FWIW "ip r" now replies as follows:
```
root@box:~# ip r
default via 192.168.0.1 dev eth0 src 192.168.0.133 metric 202
default via 192.168.0.1 dev wlan0 src 192.168.0.100 metric 303
10.8.0.0/18 via 10.8.0.69 dev tun0
10.8.0.1 via 10.8.0.69 dev tun0
10.8.0.69 dev tun0 proto kernel scope link src 10.8.0.70
192.168.0.0/24 dev eth0 proto kernel scope link src 192.168.0.133 metric 202
192.168.0.0/24 dev wlan0 proto kernel scope link src 192.168.0.100 metric 303
```
> It seems to me that "hotspot-on" and "hotspot-off" would need to diddle with line 41 referenced.

Thanks!  Hopefully @jvonau can reply later?

holta, 2018-01-12:
#642 ("WiFi-installed RPi3's add erroneous br0 route when WiFi drops") is possibly related?

But thankfully Raspbian's several WiFi problems appear much more rare this week thanks to PRs #635 #649 #651 #653, allowing WiFi-based installs of Internet-in-a-Box to move forward meaningfully, after months of work here.

Notwithstanding the seriousness of #638 & https://github.com/raspberrypi/linux/issues/1342 that we hope will be solved at the source with new firmware etc.

Recap: Raspbian's ongoing WiFi issues are increasingly understood &mdash; if not quite mitigated yet &mdash; _but these are certainly not blockers for IIAB 6.5's release!_

holta, 2018-02-15:
Seems under control 3 months later?

But am keeping this open to monitor as part of IIAB 6.6...

holta, 2018-06-12:
Keep the focus on lead tkt #638

closed 2018-06-12 by holta