iiab/iiab #560: Runansible on rpi smoketest testing fails immediately with no error message

georgejhunt, 2017-11-25:
### Expected behavior and actual behavior.
Should at least output a message
### Steps to reproduce the problem.
On a machine loaded with an earlier version of IIAB, download current master, and ./runansible.
Was able to fix by removing the '-e' on the first line of runansible

### Specifications like the version of the IIAB, operating system version, or hardware details.
rpi3 with current master. There must be some obvious thing I'm overlooking. Others would have hit this.

holta, 2017-11-25:
Going forward `./runansible` should mention the recommended new alternative `./iiab-install` even when it does not fail.

_But we should also support the older approach `./runansible` with rudimentary error messaging if we can hopefully find out what the most common problem(s) are...is that clear in this situation?_

jvonau, 2017-11-25:
I can reproduce George's runansible error by altering IIAB_DIR= within /etc/iiab/iiab.env to some other value than "/opt/iiab/iiab" then the cd $IIB_DIR kicks in and returns to the prompt.

georgejhunt, 2017-11-25:
changing -e to -ex shows that runansible was erroring out at current line
34:(after the PR adding a message to use install-iiab

OLD=`grep XSCE /etc/iiab/iiab.env` This error can be eliminated by
bracketing the expected error with "+e" and "-e"

holta, 2017-11-25:
PR #565 looks to solve the root cause, whereas #563 papers it over?

Or do we want both possibly?

holta, 2017-11-25:
I'm concerned `grep XSCE /etc/iiab/iiab.env` will fail regardless in #565, just like it has to date.

However I still don't understand why it is failing, right after the file /etc/iiab/iiab.env has confirmed to exist: does anybody understand this?

jvonau, 2017-11-25:
Think the -e to reacting to grep returning no value,  I changed to having the test always return a value with the use of wc -l and test for greater than zero.

holta, 2017-11-25:
> Think the -e to reacting to grep returning no value, I changed to having the test always return a value with the use of wc -l and test for greater than zero.

Confirmed grep indeed forces an error with "set -e"

(Which is counter-intuitive to ppl who run grep from the prompt, and do not see an error where the keyword/regexp is missing from the file...)

Thanks for explaining!

holta, 2017-11-25:
I'm merged PR #565 after testing it in isolation &mdash; if there's also a need for PR #563, let's discuss that Monday during our community/team call (http://minutes.iiab.io).

holta, 2017-11-25:
Another/shorter solution to problem can be found in [./iiab-install Line 86](https://github.com/iiab/iiab/blob/master/iiab-install#L86) here:
```
if [[ `grep XSCE /etc/iiab/iiab.env` ]]; then
```
Which does the same thing as [./runansible Lines 34-36](https://github.com/iiab/iiab/blob/master/runansible#L34-L36) here:
```
OLD=`grep XSCE /etc/iiab/iiab.env | wc -l`
if [ "$OLD" -gt 0 ]
then
```
(For future reference &mdash; both avoid the earlier error &mdash; without the need for "set +e")

holta, 2017-11-27:
Issue was solved by PR #565.  This should not stop us from:
1) considering PR #563 (toggle "set +e") within code paths where we need to ignore errors?
2) using the abbreviated ```if [[ `grep XSCE /etc/iiab/iiab.env` ]]; then``` in future?

closed 2017-11-27 by holta