iiab/iiab #566: Upgrade kiwix-tools from 2017-11-12

holta, 2017-11-25:
If blocker-level bug https://github.com/kiwix/kiwix-tools/issues/109 is fixed, Internet-in-a-Box should upgrade kiwix-tools immediately &mdash; hopefully this has a shot at happening in the coming week or so, prior to IIAB 6.5's release!

tim-moody, 2017-11-26:
wait for kiwix/kiwix-tools#96

holta, 2017-11-27:
Great News:

(3) https://github.com/kiwix/kiwix-tools/issues/114 (ZIM's that worked but failed to appear on main pag http://box:3000) is confirmed fixed in http://download.kiwix.org/nightly/2017-11-26/kiwix-tools_linux64_2017-11-26.tar.gz

(4) https://github.com/kiwix/kiwix-build/issues/94 (for i686 build of kiwix-tools) is looking very plausible...

In any case, our Minneapolis Content Hackathon starting Thursday means we should upgrade to the best available around that time &mdash; and can further upgrade after that as nec~

holta, 2017-11-29:
Great news kiwix-tools is converging on several fronts, but fyi http://download.kiwix.org/nightly/2017-11-29/kiwix_tools_linux64_2017-11-29.tar.gz has not yet solved kiwix/kiwix-tools#109 ("25 per page" search results at bottom) so we'll try again with build 2017-11-30 tomorrow!

holta, 2017-12-01:
Closing as initial testing of #583 is looking good &mdash; @tim-moody please confirm kiwix/kiwix-tools#96 is working too?

closed 2017-12-01 by holta