iiab/iiab #577: scripts/ansible should rerun "apt update; apt install ansible" on Raspbian & Debian too

holta, 2017-11-29:
@jvonau I noticed "scripts/ansible" does not upgrade from Ansible 2.4.1 to 2.4.2 on Raspbian & Debian&mdash;however this does work on Ubuntu&mdash;can you make Lines 85-86 below run across all Debian-class OS's?

https://github.com/iiab/iiab/blob/741b973d69323f2863113cfd0ad129fa7b89fc2e/scripts/ansible#L82-L91

jvonau, 2017-11-29:
Scripts scripts/ansible is intended to install the PPA for debian then ansible and should exit with "Ansible 2.4.1.0 installed exiting" if run a second time.(might have to stop looking at the version and just look at if ansible is installed.) The script should never try to add the repos sources twice, once he PPA is in place and available use 'apt update ansible' to upgrade ansible.

holta, 2017-11-29:
So long as people have crystal clear instructions how to upgrade Ansible to the latest 2.4.x (or possibly even 2.5.x before we know it) we'll work it out.

As long as I know what to write.  While I'd thought people were encouraged to run "scripts/ansible" to upgrade Ansible after-the-fact, any other succinct approach is fine too.

FWIW these 3 sections are (nearly!) identical & might be tightened later to avoid dups when we get to it:

- [scripts/ansible#L61-L70](https://github.com/iiab/iiab/blob/master/scripts/ansible#L61-L70)
- [scripts/ansible#L72-L81](https://github.com/iiab/iiab/blob/master/scripts/ansible#L72-L81)
- [scripts/ansible#L82-L91](https://github.com/iiab/iiab/blob/master/scripts/ansible#L82-L91)

jvonau, 2017-11-29:
Clean installs should use scripts/ansible only once to install the PPA and the latest ansible at that point in time. Anything else should use the standard "apt upgrade" approach.

holta, 2017-11-30:
> FWIW these 3 sections are (nearly!) identical & might be tightened later to avoid dups when we get to it:
>
> - [...]

Clarification: I ran Lines 61-70 on Ubuntu 16.04 LTS and that was sufficient to install Ansible 2.4.2 (after Ansible 2.2.0 was removed using "pip uninstall ansible").

Preliminary Conclusion: these Lines 61-70 appear sufficient for all Debian-family OS's &mdash; and if so Lines 72-81 and Lines 82-91 can both likely be removed, if the conditional on Line 60 is readjusted for all Debian-family OS's as such?

jvonau, 2017-11-30:
Re-running L61-L70 is not the correct solution as /etc/apt/sources.list will now have multiple entries for the same repo and results with the warning below.
> apt update
Get:1 http://security.debian.org/debian-security stretch/updates InRelease [63.0 kB]
Hit:2 https://deb.nodesource.com/node_6.x stretch InRelease                    
Get:4 http://ppa.launchpad.net/ansible/ansible/ubuntu xenial InRelease [18.0 kB]
Ign:3 http://cdn-fastly.deb.debian.org/debian stretch InRelease                
Hit:5 http://cdn-fastly.deb.debian.org/debian stretch Release                  
Get:6 http://security.debian.org/debian-security stretch/updates/main Sources [87.7 kB]
Get:8 http://ppa.launchpad.net/ansible/ansible/ubuntu xenial/main amd64 Packages [540 B]
Get:9 http://security.debian.org/debian-security stretch/updates/main amd64 Packages [232 kB]
Fetched 401 kB in 1s (290 kB/s)   
Reading package lists... Done
Building dependency tree       
Reading state information... Done
67 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: Target Packages (main/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list:12 and /etc/apt/sources.list:13
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:12 and /etc/apt/sources.list:13
W: Target Packages (main/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list:12 and /etc/apt/sources.list:13
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:12 and /etc/apt/sources.list:13

holta, 2017-11-30:
> Re-running L61-L70 is not the correct solution

That was never the idea.

These things (Lines 61-70) had never been run before on this IIAB on Ubuntu 16.04 LTS machine from August.

The point (if it's valid, per this initial test anyway) is that Lines 61-70 are sufficient — as such the 2 identical/similar stanzas below (Lines 72-81 and Lines 82-91) would appear to be unnecessary.

jvonau, 2017-11-30:
Well sort of, Debian and Raspbian can be treated the same(same code dup'd) while Ubuntu is slightly different as the keys are pre-installed. Have a look at #582 lightly tested in older VM with ansible 2.4.1 installed.

closed 2017-12-07 by holta