iiab/iiab #580: Calibre upgrade requires "systemctl daemon-reload" on Ubuntu 16.04 LTS

holta, 2017-11-30:
Following  the `systemctl daemon-reload` instructions in the error message indeed does work:
```
TASK [calibre : Enable Calibre service -- runs calibre-server by Kovid Goyal] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service calibre-serve: Warning: calibre-serve.service changed on disk. Run 'systemctl daemon-reload' to reload units.\nJob for calibre-serve.service failed because the control process exited with error code. See \"systemctl status calibre-serve.service\" and \"journalctl -xe\" for details.\n"}
```
Context: this was an upgrade of an IIAB installed in August 2017 onto Ubuntu 16.04 LTS.

holta, 2017-12-04:
@jvonau agrees with me this is not a blocked for IIAB 6.5's release, and it an be deferred until [IIAB 6.6](https://github.com/iiab/iiab/milestone/3) or closed as it's largely self-documented already &mdash; even a reboot will fix for those who refuse to read :)

holta, 2017-12-06:
FYI @jvonau this same "bug" occurred when I (otherwise successfully!) upgraded a June/July IIAB on Ubuntu 16.04 LTS, using "pip uninstall ansible" then "scripts/ansible" then "./iiab-install"

Is this manual "systemctl daemon-reload" something likely also nec during future upgrades?

(If so I'll try to help fix this!)

jvonau, 2017-12-06:
> Warning: calibre-serve.service changed on disk. Run 'systemctl daemon-reload' to reload units.

Any time a systemd unit file's content changes the daemon needs a reload as the unit files are read once by systemd during boot and then cached in memory.

> Job for calibre-serve.service failed because the control process exited with error code. See "systemctl status calibre-serve.service" and "journalctl -xe" for details.

That I'll assume to be caused by the content of the unit file changing, the one in memory has stale information.

jvonau, 2017-12-06:
#600

closed 2017-12-07 by holta