iiab/iiab #581: DNS failure during upgrade from Aug 2017 IIAB to 6.5/master on Ubuntu 16.04 LTS

holta, 2017-11-30:
Error2:

> TASK [network : Restart the networking service] ********************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to restart service networking: Job for networking.service failed because the control process exited with error code. See \"systemctl status networking.service\" and \"journalctl -xe\" for details.\n"}

> root@box:/opt/iiab/iiab# ping mit.edu
ping: unknown host mit.edu
root@box:/opt/iiab/iiab# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=57 time=16.6 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=57 time=16.2 ms

Unfortunately "systemctl daemon-reload" was not sufficient:

> root@box:/opt/iiab/iiab# systemctl restart networking.service
Job for networking.service failed because the control process exited with error code. See "systemctl status networking.service" and "journalctl -xe" for details.

> root@box:/opt/iiab/iiab# systemctl status networking.service
● networking.service - Raise network interfaces
   Loaded: loaded (/lib/systemd/system/networking.service; enabled; vendor preset: enabled)
  Drop-In: /run/systemd/generator/networking.service.d
           └─50-insserv.conf-$network.conf
   Active: failed (Result: exit-code) since Thu 2017-11-30 04:41:47 EST; 2min 24s ago
     Docs: man:interfaces(5)
  Process: 15019 ExecStart=/sbin/ifup -a --read-environment (code=exited, status=1/FAILURE)
  Process: 15014 ExecStartPre=/bin/sh -c [ "$CONFIGURE_INTERFACES" != "no" ] && [ -n "$(ifquery --read-environment --list --exclude=lo)" ] && udevadm settle (code=exite
 Main PID: 15019 (code=exited, status=1/FAILURE)
>
> Nov 30 04:41:47 box.lan ifup[15019]:  * Loading netfilter rules...
Nov 30 04:41:47 box.lan ifup[15019]: run-parts: executing /usr/share/netfilter-persistent/plugins.d/15-ip4tables start
Nov 30 04:41:47 box.lan ifup[15019]: run-parts: executing /usr/share/netfilter-persistent/plugins.d/25-ip6tables start
Nov 30 04:41:47 box.lan ifup[15019]:    ...done.
Nov 30 04:41:47 box.lan ifup[15019]: RTNETLINK answers: File exists
Nov 30 04:41:47 box.lan ifup[15019]: Failed to bring up br0.
Nov 30 04:41:47 box.lan systemd[1]: networking.service: Main process exited, code=exited, status=1/FAILURE
Nov 30 04:41:47 box.lan systemd[1]: Failed to start Raise network interfaces.
Nov 30 04:41:47 box.lan systemd[1]: networking.service: Unit entered failed state.
Nov 30 04:41:47 box.lan systemd[1]: networking.service: Failed with result 'exit-code'.

holta, 2017-11-30:
A reboot seems to have worked &mdash; here is the result of "./iiab-network" after rebooting:
```
PLAY RECAP *********************************************************************
127.0.0.1                  : ok=92   changed=23   unreachable=0    failed=0

ls: cannot access '/etc/sys*/net*/ifcfg*': No such file or directory
default via 192.168.0.1 dev eno1  src 192.168.0.122  metric 202
10.8.0.0/18 via 10.8.0.29 dev tun0
10.8.0.1 via 10.8.0.29 dev tun0
10.8.0.29 dev tun0  proto kernel  scope link  src 10.8.0.30
169.254.0.0/16 dev tun0  scope link  metric 1000
172.18.96.0/19 dev br0  proto kernel  scope link  src 172.18.96.1 linkdown
192.168.0.0/24 dev eno1  proto kernel  scope link  src 192.168.0.122  metric 202
bridge name     bridge id               STP enabled     interfaces
br0             8000.00c2c6ba7714       no              wlp1s0
run start:Thu Nov 30 04:49:49 EST 2017
run end: Thu Nov 30 04:50:31 EST 2017
```

jvonau, 2017-12-01:
https://github.com/iiab/iiab/pull/584 was written last week but I forgot to summit the PR till now.

commit 0b3a5eab (2017-12-04): Merge pull request #584 from jvonau/upgrades

guard against losing dns lookups when the machine has alreay been con…

closes #581
closed 2017-12-04 by holta