iiab/iiab #586: regression: iiab-admin shell access is somewhat hostile to new users of CLI

holta, 2017-12-03:
### Expected behavior and actual behavior.

@darkenvy notices that iiab-admin shell access (CLI = command-line interface) is painful to use compared to xsce-admin in the past.  Today he can no longer use up-arrow-to-retriece-prior-commands and the prompt appears as a single character "$" rather than a modern cwd or similar.  FWIW the shell appears as "/bin/sh" instead of "bin/bash" as seen here:
```
$ echo $SHELL
/bin/sh
```

Indeed it's more than a bit hypocritical to ask new implementers/operators to use iiab-admin/g0adm1n at the command line using "sudo" for each command, if we bypass that ourselves with a "do as I say, not as I do" attitude :-)

### Steps to reproduce the problem.

Install http://download.iiab.io/6.5/rpi/load.txt onto a RPi3 and log in (ssh or attached monitor/screen) as user iiab-admin with password g0adm1n

### Specifications like the version of the IIAB, operating system version, or hardware details.

RPi3 running Raspbian Lite 2017-11-29.

tim-moody, 2017-12-03:
why is this important if we require that users sudo su -?

holta, 2017-12-03:
> why is this important if we require that users sudo su -?

Security-conscious folk strongly disagree with running as root for all things, hence the mention of "curl download.iiab.io/6.5/rpi/load.txt | sudo bash" in our documentation which I do test occasionally to make sure it indeed works.

holta, 2017-12-04:
@tim-moody suggests trying to force iiab-admin's login shell to /bin/bash as follow...

http://docs.ansible.com/ansible/latest/user_module.html

For example:
```
- user:
    name: james
    shell: /bin/bash
    groups: admins,developers
    append: yes
```

holta, 2017-12-07:
@darkenvy: Resolved, for you especially!

closed 2017-12-07 by holta