iiab/iiab #587: Client Devices cannot connect to RPi3's internal WiFi on fresh install (DHCP issue?)

holta, 2017-12-03:
FYI client devices get stuck in a neverending loop showing things like:

* "Obtaining IP address..."
* "Saved Connection"
* Etc

Thanks to @darkenvy for showcasing/uncovering this obvious showstopper bug, on a fresh install of http://download.iiab.io/6.5/rpi/load-min.txt

@jvonau is investigating.

holta, 2017-12-03:
@jvonau has found a race condition (between dhcpd and having the LAN address assigned?) and prototyped solution we hope can be rolled into master within 24h.

holta, 2017-12-03:
PR https://github.com/iiab/iiab/pull/588 might fix this?

holta, 2017-12-03:
Strangely similar situation occurred on Ubuntu 16.04 LTS (coincidentally?) the moment after PR #588 was applied onto IIAB 6.5/master there (as upgraded November 30th, from August 2017's original install of IIAB).

@jvonau fixed this as follows:

> Your upgrade had stale networking info from the prior older install  in /etc/network/interfaces... cleaned that up (see the revised file) then "systemctl disable dhcpcd" "./iiab-network"  to re-write interfaces.d/iiab
rebooted for good measure.

FWIW Ubuntu 16.04 LTS /etc/network/interfaces now appears as follows:
```
# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

source /etc/network/interfaces.d/*

# The loopback network interface
auto lo
iface lo inet loopback

# IIAB no longer writes to this file
# iiab_network_mode is Gateway
auto eno1
iface eno1 inet dhcp
```
A priori, its /etc/network/interfaces likely appeared as follows:
```
# This file describes the network interfaces available on your system
# and how to activate them. For more information, see interfaces(5).

source /etc/network/interfaces.d/*

# The loopback network interface
auto lo
iface lo inet loopback

# iiab_network_mode is Gateway
# we always want the wireless to be configured (and under bridge) if it exists
auto wlp1s0
iface wlp1s0 inet manual
   pre-up   ifconfig $IFACE up
   pre-down ifconfig $IFACE down

#################   GATEWAY   #########################
auto br0
iface br0 inet static
    bridge_ports  wlp1s0
    bridge_maxwait 0
    address 172.18.96.1
    netmask 255.255.224.0
    dns-nameservers 172.18.96.1
iface eno1 inet manual
```

jvonau, 2017-12-03:
> Yes WiFi hotspot on 10.8.0.34 = Reno's RPi3 continues to work since 20min ago.

That is #588 

> On NUC/Ubuntu = 10.8.0.30 I only did: git checkout -b jvonau-dhcpd-race master git pull https://github.com/jvonau/iiab.git dhcpd-race reboot And after that WiFi (so far) no longer works, for whatever reason.

Nothing from new branch was applied as there was no ansible action preformed. The initial install occurred prior to https://github.com/iiab/iiab/commit/ed2b5d78f20099e1a2ce2cc66bc3bdd408ddb9df that is prior to the 6.4 release point as  /etc/network/interfaces was populated.

tim-moody, 2017-12-03:
This is quite similar to what I experienced on ub 16.04. My take is that 6.5 networking is not backwards compatible; manual changes need to be made to /etc/network/interfaces and dhcpd may need to be disabled. This could be documented.

holta, 2017-12-04:
I'm closing this ticket as PR #588 seems to solve it.

> This is quite similar to what I experienced on ub 16.04. My take is that 6.5 networking is not backwards compatible; manual changes need to be made to /etc/network/interfaces and dhcpd may need to be disabled. This could be documented.

Please if poss open a new ticket suggesting where this should be documented if it's important.

closed 2017-12-04 by holta