iiab/iiab #594: "get_url:" needs "timeout: 1800" (or such) for WiFi installs / low-bandwidth

holta, 2017-12-05:
This worked, allowing WiFi-based install of IIAB 6.5/master onto an RPi3.

Insert "timeout: 1800" (or similar) after "get_url:" in these places:

* [calibre/tasks/create-db.yml](https://github.com/iiab/iiab/blob/master/roles/calibre/tasks/create-db.yml#L13) downloads 513KB Metamorphosis-jackson.epub
* [kiwix/tasks/main.yml](https://github.com/iiab/iiab/blob/master/roles/kiwix/tasks/main.yml#L27) downloads 98MB kiwix-tools_armhf_2017-12-01.tar.gz
* [nextcloud/tasks/main.yml](https://github.com/iiab/iiab/blob/master/roles/nextcloud/tasks/main.yml#L17) downloads 42MB nextcloud_latest-12.tar.bz2
* [phpmyadmin/tasks/main.yml](https://github.com/iiab/iiab/blob/master/roles/phpmyadmin/tasks/main.yml#L2) downloads 11MB phpMyAdmin-4.7.5-all-languages.zip
* [sugarizer/tasks/main.yml](https://github.com/iiab/iiab/blob/master/roles/sugarizer/tasks/main.yml#L2) downloads 217MB sugarizer-0.9.tar.gz
* [wordpress/tasks/install.yml](https://github.com/iiab/iiab/blob/master/roles/wordpress/tasks/install.yml#L15) downloads 9.2MB wordpress-4.9.1.tar.gz

_Caveat: KA Lite's installation can still fail during its own internal downloader, see [#589](https://github.com/iiab/iiab/issues/589)._

jvonau, 2017-12-05:
I don't like the one size fits all, big stick approach. If you are going to add a timeout do something like 
>  get_url:
    url: "{{ some_path }}"
    dest: "{{ downloads_dir }}/"
    timeout: "{{ max_timeout }}"

Next define max_timeout in roles/0-init/defaults/main.yml, keep this set to a sane/safe value for wired devices. This allows for an override that could be adjusted in local_vars.yml to suit the end-user's environment without waiting for hours for a failure to show up when a wired device is used.

holta, 2017-12-06:
> I don't like the one size fits all, big stick approach.

Nobody does.  But the fact is Ansible's "get_url:" is profoundly broken for developing world implementers, as if these people really don't matter at all.

If `shell: wget <URL>` with no "timeout:" at all is fundamentally a better solution for all comers, then should we agree to go with that instead?

jvonau, 2017-12-06:
The rub is get_url supports proxies transparently once defined in the environment(from working with Anish) while wget takes a cmdline option for proxy support to work. Think an optional pre-download script could be used where internet performance is questionable, wget or something more robust like rsync might be employed.

holta, 2017-12-06:
> Think an optional pre-download script could be used where internet performance is questionable, wget or something more robust like rsync might be employed.

Great.  In any case let's not treat 3rd Worlders as 3rd Class Citizens, and get the very basics working here in IIAB 6.5.  Even if this has to evolve later in IIAB 6.6+ that's A-Ok.

holta, 2017-12-07:
Progress from our community/team call today (http://minutes.iiab.io) even if Ansible doesn't cleanly handle developing world implementers.  Who should NOT be punished with arbitrary timeouts cutting off their expensive downloads after 30min (or whatever the number is!)  So many folk really do still have ~1mbit/sec-or-slower Internet connections, and should not be socked with artificial limits :/

Another Issue: @jvonau is concerned about proxy settings baked into certain OS's in certain schools/corporations/govts, and so favors get_url: over wget/rsync etc.  Be that as it may, this is an extremely rare use case (no users or implementers have ever asked about this to date, and I prefer we implement based on what our community is actually asking for, and deal with statistically rare corner cases later).

Still, let's look at sticking with get_url: for now, instead of baking our own wget or rsync solution &mdash; and that likely means using get_url: parameters such as the following:
```
async: 300   (similar to "timeout: 300" so let's stick with the "timeout:" approach for now?)
poll: 60   (if this puts a visual on screen every 60s, then this is a great idea, especially if incremental progress is shown like with wget !)
```
SEE http://docs.ansible.com/ansible/latest/playbooks_async.html

In the short term, we'll likely implement this with the following variable place into default_vars.yml and local_vars.yml:
```
max_download_time: 1800
```
The 6 aforementioned playbooks would then do their best to implement this after each get_url: with something like the following:
```
timeout: "{{ max_download_time }}"
```
(Adding in arithmetic to scale this 30min down for smaller downloads possibly??)

holta, 2017-12-08:
While its poorly documented, there's strong evidence "timeout: 10" (default) applies to the URL initial request call, not to the entire download.

As further corroborated by the fact that few of these downloads complete within 10 seconds!

As such I believe it's safe (and wise) to experiment with "timeout: 100" across the board, everywhere we're using get_url:  &mdash; and then re-evaluate if these 100 seconds (1min 40 sec) are sufficient to overcome most all WiFi glitches and 3G Internet glitches in the developing world etc.

closed 2017-12-08 by holta

holta, 2017-12-08:
Great news, 2 separate [BIG-sized](http://wiki.laptop.org/go/IIAB/local_vars_big.yml) installs have completed on Raspbian Lite over WiFi, which I don't believe has ever happened before!

So PR #613's `timeout: 100` across the board appears (at the bottom of every `get_url:` stanza) to be helping indeed :)