iiab/iiab #608: Captive Portal experimental support summary in FAQ.IIAB.IO ?

holta, 2017-12-07:
I'm in favor, even if just a couple sentences that we work on together @ http://FAQ.IIAB.IO on how to try out @tim-moody's nodogsplash PR #597 prototype for IIAB 6.5's release.

holta, 2017-12-30:
Documented in http://minutes.iiab.io under Dec 28, 2017 Item 2. h.

Hoping to migrate this to http://FAQ.IIAB.IO with @deldesir's help in January...or in coming days!

holta, 2018-05-17:
@tim-moody has also suggested @georgejhunt research an iptables approach.

holta, 2018-05-20:
As nodogsplash Captive Portal support might be changing to an iptables approach instead, I won't push these DRAFT configuration materials into FAQ.IIAB.IO until the outcome is more clear:

Captive Portal Experimentation (#597) Summary:

1. Set these both to True in your /opt/iiab/iiab/vars/local_vars.yml :
nodogsplash_install: True
nodogsplash_enabled: True
2. Arrange for Ansible to run the nodogsplash playbook AFTER ./iiab-install (incl network playbook!) have completed, similar to the 5-line “CALIBRE” section in  roles/9-local-addons/tasks/main.yml but remember the “network” playbook runs AFTER Stage 9. 
   1. _Explanation: IIAB’s networking playbook will revert iptables.  So you need the nodogsplash playbook To Run At The Very End!_
3. Works very well on Windows/Mac...but Android sign-in page never quite goes away?  Also a time-out issue might hit Android folk etc.  So encourage them to bookmark it after they get thru to http://box or whatever.  Android is the worst, old or new Android :-(  (Possibly Firefox on Android isn’t so bad?)
   1. Tim pasted in his testing across 4 platforms, so others can confirm problem clients: github.com/iiab/iiab/pull/597#issuecomment-354317071 
4. Modify its configuration until you get your favorite browsers/clients working better:
   1. /etc/nodogsplash/nodogsplash.conf ← modify this one especially!
   2. /etc/nodogsplash/htdocs/splash.html
   3. Of course run “systemctl restart nodogsplash” each time to test!
   4. Possibly change the timeout (which uses client devices’ MAC addresses) reducing it to 10min or so? (George's suggestion)
   5. Later do the same with Ansible?
https://github.com/iiab/iiab/blob/master/roles/nodogsplash/templates/nodogsplash.conf.j2
https://github.com/iiab/iiab/blob/master/roles/nodogsplash/templates/splash.html.j2

m-anish, 2018-05-24:
Since we are looking at "advanced networking", I wanted to know if anyone recently tried radius + coova?

I am going to try installing this in a VM, and hopefully, if it works, try to get it upstream.

tim-moody, 2018-05-24:
I currently prefer nodogsplash to coova for reasons given elsewhere.
There is a experimental version at https://archive.org/details/iiab-6.4-180321-experimental-only-g3ea4256.img

holta, 2018-07-12:
Possibly this will be replaced by PR #870 ?

holta, 2018-07-23:
While this TK-inspired documentation is insufficient, it's finally getting a bit closer to what http://FAQ.IIAB.IO needs to become genuinely useful globally:

https://github.com/iiab/iiab/pull/870#issuecomment-406869713

holta, 2018-08-11:
Update &mdash; we have 2 (or 3!) competing approaches to Captive Portal evolving here:

#826 "Nodogsplash Captive Portal requires internet connection [try dnsmasq approach?]"
#870 "Tweaking iptables for Captive Portal (Anish's Python approach, thx to Nikos Fotiou)"

So: we likely don't have what it takes to write up a coherent recommendation for IIAB 6.6 at this time &mdash; deferring to [IIAB 6.7](https://github.com/iiab/iiab/milestone/4) makes more sense at this point &mdash; so that IIAB 6.6 can be released now in coming weeks, without confusing the hell out of everyday implementers and users.

holta, 2018-08-16:
@arky & @m-anish:

@georgejhunt is working hard exploring Captive Portal to get a basic understanding of Android (especially) and Windows/iOS (secondarily) and will try to keep basic notes on his progress here, if it helps others: https://github.com/georgejhunt/iiab/wiki/Captive-Portal-Notes

(For a quick summary, see today's http://minutes.iiab.io notes under item 1. b. ii.)

georgejhunt, 2018-08-20:
> georgejhunt/iiab/wiki/Captive-Portal-Notes

Progress report:

   1. At time of wifi AP association, I can reliably capture the http
   connectivity test from ansible,mac, and windows, using dnsmasq (this avoids
   the man in the middle warning)
   2. Getting out of the "walled garden" in android is difficult -- but a
   https request works -- If our splash page can successfully educate user
   that the "man in the middle warning" is ok this time.
   3. A splash page works on all three operating systems, with internet
   connection or not.
   4. Tried installing jquery, and bootstrap, but still cannot get the
   splash page to be sized properly on android.
   5. The Admin Console works in non ssl mode. But I expect problems when
   admin console is using ssl.
   6. So there's still a lot of development and testing to do.

tim-moody, 2018-08-20:
2. is https required or does http work?

4. page sizing - what do you mean by the 'splash' page; is this /home/index.html using iiab-menu or something else?

georgejhunt, 2018-08-20:
>
>    1.
>
>    is https required or does http work?
>
> https is required (or at least it is the only way I discovered) to escape
the "walled garden". Apparently the crippled webview instance cannot handle
ssl.

>
>    1.
>
>    page sizing - what do you mean by the 'splash' page; is this
>    /home/index.html using iiab-menu or something else?
>
> I created a jinja2 splash page, to which I redirect, when the OS in
question does its connectivity check. The splash page I create may need to
be tailored to the OS. On this splash there is a link to http://box.lan --
which works directly for mac and windows. For android, I currently have two
links; one which is https, and brings up chrome, the other which loads
http://box.lan/home.

georgejhunt, 2018-08-23:
Rocks in the road:

   1. Android 5.0 (lollipop) asks dnsmasq for ipv6 address via ipv4. We do
   not have the ipv6 stack enabled, so we cannot configure dnsmasq to shunt
   the GET requests to us.
   2. If internet is available to dnsmasq, it proxies the request and
   returns the ipv6 address
   3. Failing to connect via ipv6, (the kernel's ipv6 stack and routing is
   turned off), android concludes internet is not available, and disables wifi
   -- ref:
   https://www.androidpolice.com/2014/10/18/lollipop-feature-spotlight-android-now-defaults-to-mobile-data-when-wi-fi-has-no-internet-access-signal-icon-adds-a-for-no-connection/
   4. There is no option to disable dnsmasq's responding to AAAA (ipv6)
   requests -- we might be able to recompile to accomplish this.
   5. If internet is not available, dnsmasq cannot query upstream for ipv6
   addresses, and so android asks for an ipv4 address, which gets shunted to
   our server.
   6. But it will take more time to figure out why the redirect does not
   work in android 5, but does in android 6.

There's so much complexity to the interactions, it will take a some time to
figure out.

How many other OS versions, will require special treatment? I have not yet
figured out how to distinguish a MAC request from and IOS request!

I do know that what works for the MAC, does not work for IOS version 5.

So I don't think captive portal is possible for 6.6, just as Adam has
already decided.

Maybe my approach is doomed.

georgejhunt, 2018-09-14:
Some testing results:
1. Start with a Aug 27 image
2. load current (6.6) master, iiab-install --reinstall,
3. needed to remove outdated ansible, and install with script/ansible
4. change local_vars.yml dnsmasq_enabled, py_captive_portal_enabled
5. merge in the test branch http://github.com/georgejhunt/iiab/ --branch cpsmall
6. run iiab-network and reboot
7. do it again, because local_vars moved from /opt/iiab/iiab/vars/local_vars.yml to /etc/iiab/
8. iiab-network is not enough -- need full iiab-install --reinstall because of changes to iiab.env
### With ethernet cable (iiab_gateway_enabled = false)
* Android 5.1 lollipop - Splash page with "go to home page", click takes user to home, but header still says "sign in to network" at the top. Initial timeout attempts to return 204, to remove sign-in, not succcessfull.
* Android 6.0.1 -- Splash page shows 2 buttons, First says please ignore security warning when you click this button that makes a https request. Splash page returns. click second button takes to home page (no "sign in to network" at top).The trick here is to escape the dumbed down browser with a https request. But the sign-in request is gone.
* Windows 10.1 -- Splash shows "go to IIAB home page" which opens -- but the splash page remains as tab that is hidden by home page tab. 
* Mac 10.13.4 -- Splash displayed, click on "go to IIAB home page" opens default browser (but the splash page is also visible in another tab.
### With no Ethernet Cable (and after reboot)
* no change from the above results
### TROUBLE TIME  -- Ethernet connected (iiab_gateway_enabled = true)
* Changed /etc/local_vars.yml (iiab_gateway_enabled) and ran ./iiab-network. But forgot to plug in the ethernet cable. And of course, our self adjusting networking code adjusted to that, and did not enable masquerade. Our poor customers. There are just too many ways to screw up!
* But with client ethernet access, android 6, and the macintosh failed.  So we will need to block additional URLS. With all the clients out there, this is not a very efficient approach.
### Evaluation
* Lingering splash page, and annoying "sign-in to network" headings, are perhaps blockers.
* This dnsmasq approach still needs more work -- particularly since its function depends on blocking certain URLs, especially in the online situation.
* But again it's probably better than relying on internet access to deal with https, since our use case is primarily offline. (I don't know how we can deal with https long term, unless we do everything we can to stay in the http insecure realm). We have to hope that initial captive portal exchanges do not move to https.

holta, 2018-09-14:
Great writeup, thanks @georgejhunt !

> 2. load current (6.6) master, iiab-install --reinstall

TINY CLARIF: current master is now beginning to evolve beyond IIAB 6.6 to [IIAB 6.7](https://github.com/iiab/iiab/milestone/4) :)

holta, 2018-12-04:
It's time to put pen to paper and make this happen for the new (experimental for now) [captive-portal role](https://github.com/iiab/iiab/tree/master/roles/captive-portal) that should be part of IIAB 6.7 in coming weeks.

( pre-releases of IIAB 6.7/master are already installable today from http://download.iiab.io/6.7 )

holta, 2018-12-10:
FYI

1) “systemctl status captive-portal” is no longer relevant since this PR #1327 was merged, enacting Captive Portal as part of Apache.

2) NEW WAY TO SEE IF captive-portal IS ENGAGED:
```
ls /etc/apache2/sites-enabled    # you would see 001-captive-portal.conf
```

holta, 2018-12-11:
@georgejhunt what is the recommended way to turn off Captive Portal?

This needs to be documented in http://FAQ.IIAB.IO in language that everybody can understand.

I'll write that up once I understand your recommendations, Thanks-

georgejhunt, 2018-12-11:
I just tried putting "captive-portal: False" in /etc/iiab/local_vars.yml,
and ran './runrole captive-portal', and achieved the desired result

holta, 2018-12-11:
> I just tried putting "captive-portal: False" in /etc/iiab/local_vars.yml, and ran './runrole captive-portal', and achieved the desired result

Less-technical operators greatly benefit from 1-line commands like:
https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L123
```
# In a pinch, disable it by running: systemctl disable captive-portal
```
e.g. if a similar such command also becomes possible, that'll be a huge win.

As I clean up all 3 local_vars.yml and default_vars.yml...

tim-moody, 2018-12-11:
can probably use a2dissite <name of the conf file minus .conf>  then systemctl restart apache2

georgejhunt, 2018-12-11:
I can confirm that running "a2dissite 001-captive-portal" yields the
following
root@box:/opt/iiab/iiab# a2dissite 001-captive-portal
Site 001-captive-portal disabled.
To activate the new configuration, you need to run:
  systemctl reload apache2

jvonau, 2018-12-12:
Please remember that /etc/dnsmasq.d/capture needs to be dealt with by removing the file along with the apache config symlink.

georgejhunt, 2018-12-12:
Thanks Jerry. I missed that one

holta, 2018-12-12:
> I just tried putting "captive-portal: False" in /etc/iiab/local_vars.yml, and ran './runrole captive-portal', and achieved the desired result

@georgejhunt I assume you mean "captive_portal_enabled: False" ?

holta, 2018-12-12:
> @georgejhunt I assume you mean "captive_portal_enabled: False" ?

Can everyone please review http://FAQ.IIAB.IO Item 32 ("Captive Portal Administration: What tips & tricks exist?")

http://wiki.laptop.org/go/IIAB/FAQ#Captive_Portal_Administration:_What_tips_.26_tricks_exist.3F

Thanks!

jvonau, 2018-12-14:
1 Older installs of captive-portal should remove /etc/systemd/system/captive-portal.service after disabling the unit file. 
2 After removing /etc/dnsmasq.d/capture, dnsmasq needs to be restarted or reboot.
3 Older installs of captive-portal should ensure that the network role is run once via iiab-intall --reinstall or iiab-network to get rid of the frivolous iptables rules that have been removed from iiab-gen-iptables
4 Shouldn't the state of 'service' be recorded in /etc/iiab/iiab/iiab.ini like all the other services that are installed by the installer?

holta, 2018-12-14:
> 4 Shouldn't the state of 'service' be recorded in /etc/iiab/iiab/iiab.ini like all the other services that are installed by the installer?

Good Idea: for now [roles/captive-portal/defaults/main.yml](https://github.com/iiab/iiab/blob/master/roles/captive-portal/defaults/main.yml) contains only this one:
```
captive_portal_port: 9090
```

What other variables are important to add?

> 2 After removing /etc/dnsmasq.d/capture, dnsmasq needs to be restarted or reboot.

Thanks.  A new command-line recipe is now added to http://wiki.laptop.org/go/IIAB/FAQ#Captive_Portal_Administration:_What_tips_.26_tricks_exist.3F

holta, 2019-01-03:
Seems sufficiently complete for the release of IIAB 6.7 in coming weeks/days, Thanks All!

closed 2019-01-03 by holta

holta, 2020-01-23:
Captive Portal FAQ Overhauled...please Let Me Know what more should be added here!

http://FAQ.IIAB.IO -> "38 Captive Portal Administration: What tips & tricks exist?"