iiab/iiab #616: WiFi-installed IIAB does not connect using an Ethernet cable

holta, 2017-12-08:
Both lights on the RPi3 light up (green and orange, the green occasionally blinks) when the Ethernet cable is inserted, after http://download.iiab.io/6.5/rpi/load-big.txt was successfully installed on Raspbian Lite 2017-11-29 here.

As the WiFi is extremely flaky, I inserted a live Ethernet cable, which (used to!) work a few weeks ago, on a nearly identical RPi3 machine.  But no IP address appears under eth0 when I run "ip a".

Occasionally the "ip a" still shows its residual IP address under wlan0 and tun0, as mentioned @ https://github.com/iiab/iiab/issues/558#issuecomment-350115597

Is a reboot truly expected/required, before eth0 will get its IP address (and eventually allow Internet access...?)

Plausibly/Ideally this can be quick-fixed with @jvonau & I hacking at this a bit over the weekend, so that (ideally) RPi folk can insert a live Ethernet cable for Internet access in almost all circumstances?  (For updates/provisioning and many other reasons, etc...)

holta, 2017-12-09:
If luck goes our way, this is one of the last remaining issues to close the loop on umbrella ticket:

#197 "Permit installation of IIAB over Wi-Fi instead of Ethernet"

holta, 2017-12-09:
We'd faced a similar issue Oct 20: (solved by/around Nov 22)

#438 "no Internet (Ethernet or WiFi) after WiFi installs"

At the time the interim workaround...to restore Ethernet/Internet...was:
```
rm /etc/network/interfaces.d/iiab
reboot
```

jvonau, 2017-12-09:
The network was configured per the defaults provided by https://github.com/iiab/iiab/blob/master/roles/network/defaults/main.yml#L37

holta, 2017-12-09:
> The network was configured per the defaults provided by https://github.com/iiab/iiab/blob/master/roles/network/defaults/main.yml#L37

I don't know if such defaults are good or bad, but I have to imagine quasi-simultaneous gateways (eth0 & wlan0) unfortunately aren't easy to implement :/

ASIDE: A bit like Android/iPhone networking is always trying to guess which Internet connection to use (3G/4G versus a flurry of quasi-open SSID's coming & going within range...)

Anyway that's not our problem, but it was interesting talking to an Android networking engineer last night, who mentioned Google engineers are in a constant cat+mouse game trying to outwit hostile Captive Portals, while opportunistically looking for free/open Access Points...as Android users walk around auto-connecting to diff SSID's...and commercial Captive Portals in turn try to outwit each successive Android :|

jvonau, 2017-12-09:
#617

holta, 2017-12-11:
Huge thanks to @georgejhunt who found the root cause of Ethernet failing here, on line 41 of /etc/dhcpcd.conf: https://github.com/iiab/iiab/issues/558#issuecomment-350842090

holta, 2017-12-18:
@jvonau outlined an idea to make Ethernet-as-gateway work alongside WiFi-as-gateway _by default_ during our community/team call this morning (http://minutes.iiab.io).

QUESTION: Should this happen on RPi only, or on all installs?

_Idea in a Nutshell: those implementers who also want an *external* AP (i.e. a physical WiFi router for nearby IIAB users, even if this is a rare case and increasingly less common) need clean documentation on how to proceed...and can/should perhaps use "LAN Controller" mode &mdash; with their physical/external AP added to bridge manually perhaps??_

holta, 2018-01-12:
Resolved by PR #635

closed 2018-01-12 by holta