iiab/iiab #619: Test IIAB 6.5/master against Ubuntu 18.04 LTS daily builds

holta, 2017-12-12:
Thanks to @floydianslips & @deldesir whose works is ongoing here:

https://cdimage.ubuntu.com/daily-live/current/ <- Josh Dennis will work on this for Jerry, to confirm that IIAB 6.5/master networking now works on Ubuntu 18.04 NON-server edition?

https://cdimage.ubuntu.com/ubuntu-server/daily/current/ <- Blondel prefers the Server Edition of Ubuntu 18.04 dailies

Related: #612 (Debian 9.3 testing)

floydianslips, 2017-12-13:
iiab/vars/ubuntu-18.yml
postgresql_version: 9.6 ==> postgresql_version: 10

deldesir, 2017-12-13:
Nodjs won't install automatically. Will give a try using manual install for Ubuntu 17.10 (Artful Aardvark).

`Your distribution, identified as "Ubuntu Bionic Beaver (development branch)", is a pre-release version of Ubuntu. NodeSource does not maintain official support for Ubuntu versions until they are formally released. You can try using the manual installation instructions available at https://github.com/nodesource/distributions and use the latest supported Ubuntu version name as the distribution identifier, although this is not guaranteed to work.`

holta, 2017-12-13:
FYI @deldesir's 18.04 server "daily" in question here is actually the two-week-old 2017-11-30 at http://cdimage.ubuntu.com/ubuntu-server/daily/current/bionic-server-amd64.iso (which is still the most recent posted amd64 vrrsfor now, so he then upgraded with "apt update").

When running "apt -a list nodejs" he got:

```nodejs/bionic 6.12.0~dfsg-2ubuntu1 amd64```

He also tried https://community.parrotsec.org/viewtopic.php?id=489

Any other tips he should try?

deldesir, 2017-12-14:
Editing script from https://deb.nodesource.com/setup_6.x by commenting out exit 1 in pre-release condition and adding "Ubuntu" "bionic" "Ubuntu" "xenial" to install it manually like If it was Ubuntu Xenial Xerus. It works.

deldesir, 2017-12-14:
Had to install libgl1-mesa-dev to make calibre to install. Is it #530 related?

deldesir, 2017-12-14:
ifupdown is still mandatory at Network stage.

deldesir, 2017-12-14:
Looks like dhcpd service is messy. Clients won't connect to AP for me. I've already seen a similar issue that was fixed by #588

floydianslips, 2017-12-14:
Postgres 10 requires encrypted passwords:
roles/moodle/tasks/main.yml: after line 110 add,

encrypted: yes

deldesir, 2017-12-18:
Do **systemctl restart iiab-cmdsrv.service** when having the following error with admin-console (thanks @holta for the hint):
`GET-CONF FAILED and reported CMDSERV is not runnig
GET-ANS FAILED and reported CMDSRV is not running.
GET-IIAB-INI FAILED and reported CMDSRV is not running
GET-ZIM-STAT FAILED and reported CMDSRV is not running.
GET-WHLIST FAILED and reported CMDSRV is not running.
GET-VARS FAILED and reported CMDSRV is not running.
GET-SPACE-AVAIL FAILED and reported CMDSRV is not running.`

holta, 2017-12-18:
> Do systemctl restart iiab-cmdsrv.service when having the following error with admin-console (thanks @holta for the hint):
GET-CONF FAILED and reported CMDSERV is not runnig GET-ANS FAILED and reported CMDSRV is not running. GET-IIAB-INI FAILED and reported CMDSRV is not running GET-ZIM-STAT FAILED and reported CMDSRV is not running. GET-WHLIST FAILED and reported CMDSRV is not running. GET-VARS FAILED and reported CMDSRV is not running. GET-SPACE-AVAIL FAILED and reported CMDSRV is not running.

Clarif: Admin Console (http://box/admin) was not starting for @deldesir's after every boot of his Ubuntu 18.04 Server.

Sounds like a timing issue that will have to be worked out before 18.04 is released &mdash; let us know @deldesir if you discover the root cause!

holta, 2017-12-18:
@deldesir & @floydianslips : do you have time to create any PR's (pull requests) for the above Ubuntu 18.04 LTS issues you've resolved?

(This can also wait for IIAB 6.6 if you prefer &mdash; but if there are any later-breaking merges needed, that we should slip into IIAB 6.5 this week &mdash; let us know very soon in coming days!!)

holta, 2017-12-18:
> Do systemctl restart iiab-cmdsrv.service when having the following error with admin-console (thanks @holta for the hint):
GET-CONF FAILED and reported CMDSERV is not runnig GET-ANS FAILED and reported CMDSRV is not running. GET-IIAB-INI FAILED and reported CMDSRV is not running GET-ZIM-STAT FAILED and reported CMDSRV is not running. GET-WHLIST FAILED and reported CMDSRV is not running. GET-VARS FAILED and reported CMDSRV is not running. GET-SPACE-AVAIL FAILED and reported CMDSRV is not running.

@deldesir is there any possibility this error would go away all by itself, e.g. if you Wait A Full 5min after booting before attempting http://box/admin ?  (Or have you already tried that?!)

holta, 2017-12-23:
@floydianslips & @deldesir should I move this to[ IIAB 6.6](https://github.com/iiab/iiab/milestone/3)?

(Or will you be submitting the 18.04 quick fixes you discovered as PR's in the coming days?)

holta, 2018-02-05:
IIAB 6.5 was supposed to have been released in November but now that we've redefined its scope into a larger 7.0-like product (*) we might need to reconsider embracing April's Ubuntu 18.04 LTS release as something to take more seriously:

https://wiki.ubuntu.com/BionicBeaver/ReleaseSchedule

(*) given IIAB 6.5's stated reason (installing IIAB over WiFi) proved Far More Complicated than anybody anticipated &mdash; this core feature remains incomplete 4+ months later &mdash; due to [RPi/Raspbian WiFi bug](https://github.com/iiab/iiab/issues/638) etc, e.g. the latest being #655 where work is ongoing ("iiab-hotspot-on fails if IIAB installed via WiFi (/etc/sysconfig/dhcpd missing, so dhcpd won't start)").

holta, 2018-02-15:
Update: With [18.04's final feature freeze](https://wiki.ubuntu.com/BionicBeaver/ReleaseSchedule) rapidly approaching at the end of February, it's time now to consolidate our fixes for all "big iron" users like @m-anish, building off #502 "U17-2: refine network for Ubuntu 18.04 LTS daily builds".

holta, 2018-06-10:
IIAB 6.5 and 6.6/master now work well on Ubuntu 18.04 LTS.

closed 2018-06-10 by holta