iiab/iiab #625: Client devices can't connect to Ubuntu 18.04 (Development) | IIAB Testing

deldesir, 2017-12-14:
DHCP doesn't work. So client devices can't get their IP addresses from WIFI Access Point.
Similar to #587

deldesir, 2017-12-14:
This issue is fixed by moving contents of /etc/network/interfaces.d/iiab to /etc/network/interfaces.
Now We have a nice running IIAB AP on Ubuntu Bionic Beaver (development).

holta, 2017-12-14:
> This issue is fixed by moving contents of /etc/network/interfaces.d/iiab to /etc/network/interfaces.
Now We have a nice running IIAB AP on Ubuntu Bionic Beaver (development).

FYI @jvonau is the above really the recommended / best solution for Ubuntu 18.04 LTS ?  Or can we automate something like this?

deldesir, 2017-12-14:
We'll wait for @jvonau as he will fix this better and permanently. I think this should remain open until @jvonau intervention.

deldesir, 2018-02-19:
@holta Same problem encountered on 17.10 now. Same workaround works.

jvonau, 2018-02-19:
Need to confirm if there is a desktop installed and if NetworkManager is active or not. The later could be deduced from /etc/iiab/iiab.ini

holta, 2018-02-20:
Thanks much @deldesir who documented this current workaround at the bottom of https://github.com/iiab/iiab/wiki/IIAB-Networking

Let's also keep http://FAQ.IIAB.IO on the ball
 — question #16 especially ("Any other networking tips?")

holta, 2018-02-26:
> Need to confirm if there is a desktop installed and if NetworkManager is active or not. The latter could be deduced from /etc/iiab/iiab.ini

@jvonau's question above is crucial to &mdash; please run `scripts/local_facts.fact` to determine if your particular OS is running "NetworkManager" or not _(on Ubuntu 18.04 this depends whether you're running a GUI like "X Windows")_.  On Ubuntu 16.04 LTS this command shows:
```
root@box:/opt/iiab/iiab# scripts/local_facts.fact
dhcpcd.service is not a native service, redirecting to systemd-sysv-install
Executing /lib/systemd/systemd-sysv-install is-enabled dhcpcd
{"phplib_dir"             : "/usr/lib/php",
"stage"                   : "9",
"dhcpcd"                  : "enabled",
"network_manager"         : "enabled",
"systemd_networkd"        : "disabled",
"iiab_branch"             : "holta-kiwix-2018-02-16",
"iiab_commit"             : "9e669f39b385371d1ad7eca0cff08617af8530e5",
"xo_model"                : "none",
"ansible_version"         : "2.4.2.0",
"os"                      : "ubuntu",
"os_ver"                  : "ubuntu-16"}
```
On Raspbian 2017-11-29 this shows:
```
root@box:/opt/iiab/iiab# scripts/local_facts.fact
{"phplib_dir"             : "/usr/lib/php",
"stage"                   : "9",
"dhcpcd"                  : "enabled",
"network_manager"         : "",
"systemd_networkd"        : "disabled",
"iiab_branch"             : "master",
"iiab_commit"             : "b08fa5f7c3c9993dd502e1b33836dd91d0e82d79",
"xo_model"                : "none",
"ansible_version"         : "2.4.2.0",
"os"                      : "raspbian",
"os_ver"                  : "raspbian-9"}
```

holta, 2018-06-10:
Appears resolved.  @deldesir & @jvonau please re-open as nec!

closed 2018-06-10 by holta