iiab/iiab #627: Moodle 3.5 LTS - ETA May 8th 2018

holta, 2017-12-14:
While unlikely for [IIAB 6.6](https://github.com/iiab/iiab/milestone/3), let's keep an eye on Moodle 3.5 LTS pre-releases so we're ready for its 3-year LTS support window beginning "May 8th 2018" :

https://docs.moodle.org/dev/Releases#Version_support

As we upgrade from [Moodle 3.1.x LTS](https://docs.moodle.org/31/en/New_features) originally released May 2014, per Moodle's calendar-based release schedule:

https://docs.moodle.org/dev/Releases#General_release_calendar

holta, 2018-02-22:
http://FAQ.IIAB.IO Question 22 (["Can I upgrade or reinstall server apps?"](http://wiki.laptop.org/go/IIAB/FAQ#Can_I_upgrade_or_reinstall_server_apps.3F)) may also need to be updated &mdash; e.g. if Moodle 3.5.x's maintenance upgrade process[*] changes from 3.1.x LTS possibly?

[*] Second Monday of July, September, November, January, March and May

holta, 2018-03-22:
It'd be wonderful if [MoodleBox](https://moodlebox.net/)’s Nicolas Martignoni (@martignoni) in Switzerland had time to work with us on this (for Internet-in-a-Box 6.6) ...let's try to get in touch in coming weeks if possible!

martignoni, 2018-03-22:
How do you install Moodle? Please point me to the role where you do this.

holta, 2018-03-22:
> How do you install Moodle in IIAB? Please point me to the role where you do this.

Here's how we do it today:
https://github.com/iiab/iiab/tree/master/roles/moodle

But Moodle 3.1 LTS is getting extremely long in the tooth; we could definitely use your insights Nicolas on all fronts!!

(Also feel free to join one of our Internet-in-a-Box community/team calls @ http://minutes.iiab.io anytime~)

holta, 2018-03-22:
> https://github.com/iiab/iiab/tree/master/roles/moodle

http://FAQ.IIAB.IO might help a bit if the above is too much to swallow &mdash; or ask away if any questions :)

holta, 2018-04-26:
@georgejhunt special request that you help me modernize Moodle / playbook in coming weeks when it's released (Moodle 3.5 LTS appears far more modern/usable by the general public!)

https://www.moodleworld.com/whats-coming-in-moodle-3-5-the-next-lts-version-of-moodle/
https://www.moodlenews.com/2018/privacy-better-quizzes-faster-and-modern-the-latest-scoop-on-moodle-3-5/

martignoni, 2018-04-26:
@holta Just had a look on your tasks for installing Moodle. Everything regarding Moodle installation seems OK to me.

Some things you have to check before updating to Moodle 3.5:
- Server requirements fulfilled (PHP, db)? (see https://tracker.moodle.org/browse/MDL-59159)
- Get a shallow git clone of Moodle, to save HD space (see [this ansible task](https://github.com/moodlebox/moodlebox/blob/5cfff2e75ea76696e1357cf0849dcbce20052c27/roles/moodle/tasks/coreinstall.yml#L3-L8))

HTH

holta, 2018-04-30:
Volunteers are being sought to help here, modernizing Internet-in-a-Box to Moodle 3.5 as Ubuntu 18.04 and others increasingly require:

Please see https://github.com/iiab/iiab/issues/759 "Moodle 3.1.11 LTS impossible on Ubuntu 18.04? (due to php7.2, even w/ encrypted password)" and/or get in touch during our community/team calls @ http://minutes.iiab.io, Thanks All!

holta, 2018-05-01:
PR #768 (integrating Moodle 3.5 Beta) looks ready to merge in coming hours/days, after it's reviewed/refined by others.

closed 2018-05-01 by holta