iiab/iiab #628: IIAB 6.5/master failed to install Calibre on RPi3

holta, 2017-12-16:
1) This occurred while running http://download.iiab.io/6.5/rpi/load.txt to install onto the latest Raspbian Lite 2017-11-29.

2) Error below is also pasted into https://pastebin.mozilla.org/9074763 in case it's easier to read.

3) Error did not recur when running "./runtags calibre" &mdash; others seem to have faced similar in the past week &mdash; might there be a workaround?

> 2017-12-16 18:25:36,201 p=11494 u=root |  fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["scripts/calibre-install-latest.sh"], "delta": "0:08:18.285629", "end": "2017-12-16 18:25:31.811194", "msg": "non-zero return code", "rc": 100, "start": "2017-12-16 18:17:13.525565", "stderr": "\nWARNING: apt does not have a stable CLI interface. Use with caution in scripts.\n\nWarning: apt-key output should not be parsed (stdout is not a terminal)\ngpg: key 7638D0442B90D010: public key \"Debian Archive Automatic Signing Key (8/jessie) <ftpmaster@debian.org>\" imported\ngpg: Total number processed: 1\ngpg:               imported: 1\n\nWARNING: apt does not have a stable CLI interface. Use with caution in scripts.\n\n\nWARNING: apt does not have a stable CLI interface. Use with caution in scripts.\n\n\rExtracting templates from packages: 60%\rExtracting templates from packages: 100%\n\nWARNING: apt does not have a stable CLI interface. Use with caution in scripts.\n\n\nWARNING: apt does not have a stable CLI interface. Use with caution in scripts.\n\n\nWARNING: apt does not have a stable CLI interface. Use with caution in scripts.\n\nE: The repository 'http://archive.raspberrypi.org/debian stretch Release' does no longer have a Release file.", "stderr_lines": ["", "WARNING: apt does not have a stable CLI interface. Use with caution in scripts.", "", "Warning: apt-key output should not be parsed (stdout is not a terminal)", "gpg: key 7638D0442B90D010: public key \"Debian Archive Automatic Signing Key (8/jessie) <ftpmaster@debian.org>\" imported", "gpg: Total number processed: 1", "gpg:               imported: 1", "", "WARNING: apt does not have a stable CLI interface. Use with caution in scripts.", "", "", "WARNING: apt does not have a stable CLI interface. Use with caution in scripts.", "", "", "Extracting templates from packages: 60%", "Extracting templates from packages: 100%", "", "WARNING: apt does not have a stable CLI interface. Use with caution in scripts.", "", "", "WARNING: apt does not have a stable CLI interface. Use with caution in scripts.", "", "", "WARNING: apt does not have a stable CLI interface. Use with caution in scripts.", "", "E: The repository 'http://archive.raspberrypi.org/debian stretch Release' does no longer have a Release file."], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\ncalibre-bin is already the newest version (2.75.1+dfsg-1).\ncalibre-bin set to manually installed.\ndirmngr is already the newest version (2.1.18-8~deb9u1).\nThe following packages were automatically installed and are no longer required:\n  libtomcrypt0 libtommath1\nUse 'apt autoremove' to remove them.\n0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.\nExecuting: /tmp/apt-key-gpghome.jidlf64aUj/gpg.1.sh --recv-key --keyserver keyserver.ubuntu.com 7638D0442B90D010\nIgn:1 http://linux.teamviewer.com/deb stable InRelease\nIgn:2 http://linux.teamviewer.com/deb preview InRelease\nHit:3 http://linux.teamviewer.com/deb stable Release\nHit:4 http://mirrordirector.raspbian.org/raspbian stretch InRelease\nHit:5 http://ppa.launchpad.net/ansible/ansible/ubuntu xenial InRelease\nHit:6 http://linux.teamviewer.com/deb preview Release\nGet:7 http://deb.debian.org/debian testing InRelease [142 kB]\nHit:8 http://archive.raspberrypi.org/debian stretch InRelease\nHit:10 https://deb.nodesource.com/node_6.x stretch InRelease\nGet:12 http://deb.debian.org/debian testing/main armhf Packages [7,099 kB]\nGet:13 http://deb.debian.org/debian testing/main Translation-en [5,558 kB]\nFetched 12.8 MB in 51s (249 kB/s)\nReading package lists...\nBuilding dependency tree...\nReading state information...\n713 packages can be upgraded. Run 'apt list --upgradable' to see them.\nReading package lists...\nBuilding dependency tree...\nReading state information...\nThe following packages were automatically installed and are no longer required:\n  libegl1-mesa libpcre16-3 libpodofo0.9.4 libqt5clucene5 libqt5opengl5\n  libtomcrypt0 libtommath1\nUse 'apt autoremove' to remove them.\nThe following additional packages will be installed:\n  calibre-bin fontconfig-config libdrm-common libdrm2 libegl-mesa0 libegl1\n  libegl1-mesa libfontconfig1 libgbm1 libglvnd0 libhyphen0 libjs-coffeescript\n  libpcre2-16-0 libpodofo0.9.5 libqt5dbus5 libqt5designer5 libqt5gui5\n  libqt5help5 libqt5network5 libqt5opengl5 libqt5positioning5\n  libqt5printsupport5 libqt5qml5 libqt5quick5 libqt5sensors5 libqt5sql5\n  libqt5sql5-sqlite libqt5svg5 libqt5webchannel5 libqt5webkit5 libqt5widgets5\n  libqt5xml5 python-html5-parser python-msgpack python-pyqt5\n  python-pyqt5.qtsvg python-pyqt5.qtwebkit python-regex python-sip\n  qml-module-qtgraphicaleffects qml-module-qtquick-controls\n  qml-module-qtquick-dialogs qml-module-qtquick-layouts\n  qml-module-qtquick-privatewidgets qml-module-qtquick-window2\n  qml-module-qtquick2 qt5-gtk-platformtheme\nSuggested packages:\n  coffeescript qt5-image-formats-plugins qtwayland5 qt5-qmltooling-plugins\n  python-lxml-dbg python-lxml-doc python-pyqt5-dbg\nRecommended packages:\n  geoclue-2.0\nThe following NEW packages will be installed:\n  libdrm-common libegl-mesa0 libegl1 libglvnd0 libhyphen0 libjs-coffeescript\n  libpcre2-16-0 libpodofo0.9.5 libqt5positioning5 libqt5sensors5\n  libqt5webchannel5 python-html5-parser python-msgpack python-regex\nThe following packages will be upgraded:\n  calibre calibre-bin fontconfig-config libdrm2 libegl1-mesa libfontconfig1\n  libgbm1 libqt5core5a libqt5dbus5 libqt5designer5 libqt5gui5 libqt5help5\n  libqt5network5 libqt5opengl5 libqt5printsupport5 libqt5qml5 libqt5quick5\n  libqt5sql5 libqt5sql5-sqlite libqt5svg5 libqt5webkit5 libqt5widgets5\n  libqt5xml5 python-lxml python-pyqt5 python-pyqt5.qtsvg python-pyqt5.qtwebkit\n  python-sip qml-module-qtgraphicaleffects qml-module-qtquick-controls\n  qml-module-qtquick-dialogs qml-module-qtquick-layouts\n  qml-module-qtquick-privatewidgets qml-module-qtquick-window2\n  qml-module-qtquick2 qt5-gtk-platformtheme\n36 upgraded, 14 newly installed, 0 to remove and 677 not upgraded.\nNeed to get 52.2 MB of archives.\nAfter this operation, 5,185 kB disk space will be freed.\nGet:1 http://deb.debian.org/debian testing/main armhf libdrm-common all 2.4.88-1 [13.2 kB]\nGet:2 http://deb.debian.org/debian testing/main armhf libdrm2 armhf 2.4.88-1 [35.0 kB]\nGet:3 http://raspbian-us.ngc292.space/raspbian stretch/main armhf libhyphen0 armhf 2.8.8-5 [29.7 kB]\nGet:5 http://deb.debian.org/debian testing/main armhf fontconfig-config all 2.12.6-0.1 [304 kB]\nGet:4 http://raspbian-us.ngc292.space/raspbian stretch/main armhf libjs-coffeescript all 1.10.0~dfsg-1 [88.0 kB]\nGet:6 http://deb.debian.org/debian testing/main armhf libfontconfig1 armhf 2.12.6-0.1 [351 kB]\nGet:7 http://deb.debian.org/debian testing/main armhf libpodofo0.9.5 armhf 0.9.5-8 [448 kB]\nGet:8 http://deb.debian.org/debian testing/main armhf libqt5dbus5 armhf 5.9.2+dfsg-6 [199 kB]\nGet:9 http://deb.debian.org/debian testing/main armhf libqt5widgets5 armhf 5.9.2+dfsg-6 [1,979 kB]\nGet:10 http://deb.debian.org/debian testing/main armhf libqt5network5 armhf 5.9.2+dfsg-6 [580 kB]\nGet:11 http://deb.debian.org/debian testing/main armhf libqt5positioning5 armhf 5.9.2+dfsg-3 [114 kB]\nGet:12 http://deb.debian.org/debian testing/main armhf libqt5printsupport5 armhf 5.9.2+dfsg-6 [189 kB]\nGet:13 http://deb.debian.org/debian testing/main armhf libqt5quick5 armhf 5.9.2-3 [1,036 kB]\nGet:14 http://deb.debian.org/debian testing/main armhf qml-module-qtquick2 armhf 5.9.2-3 [40.6 kB]\nGet:15 http://deb.debian.org/debian testing/main armhf qml-module-qtquick-window2 armhf 5.9.2-3 [27.4 kB]\nGet:16 http://deb.debian.org/debian testing/main armhf qml-module-qtquick-privatewidgets armhf 5.9.2-2 [38.4 kB]\nGet:17 http://deb.debian.org/debian testing/main armhf qml-module-qtquick-layouts armhf 5.9.2-3 [47.9 kB]\nGet:18 http://deb.debian.org/debian testing/main armhf qml-module-qtquick-dialogs armhf 5.9.2-2 [119 kB]\nGet:19 http://deb.debian.org/debian testing/main armhf qml-module-qtgraphicaleffects armhf 5.9.2-2 [96.2 kB]\nGet [... cut ...] irrordirector.raspbian.org/raspbian stretch InRelease", "Ign:4 http://linux.teamviewer.com/deb preview InRelease", "Hit:5 http://linux.teamviewer.com/deb stable Release", "Hit:6 http://linux.teamviewer.com/deb preview Release", "Ign:7 http://archive.raspberrypi.org/debian stretch InRelease", "Hit:8 https://deb.nodesource.com/node_6.x stretch InRelease", "Err:9 http://archive.raspberrypi.org/debian stretch Release", "  503  Service Unavailable [IP: 93.93.130.214 80]", "Reading package lists..."]}

holta, 2017-12-16:
Is there a chance this is related to Calibre 3.14's release yesterday (https://calibre-ebook.com/whats-new) given all the mentions of `Setting up calibre (3.13.0+dfsg-1)` (and similar) in the error log above?

holta, 2017-12-16:
FYI this (fresh) install of IIAB 6.5/master shows Calibre 3.13 (rather than 3.14) for whatever reason:
```
root@box:~# calibre --version
calibre (calibre 3.13)
root@box:~# calibre-server --version
calibre-server (calibre 3.13)
```

holta, 2017-12-16:
Another data point &mdash; on a completely separate RPi3 fresh install of IIAB 6.5/master (BIG-sized in this case) in a completely different home &mdash; this time Calibre installed on the 1st try, but it continues to install the older/stale version instead of v3.14:
```
root@box:~# calibre --version
calibre (calibre 3.13)
root@box:~# calibre-server --version
calibre-server (calibre 3.13)
```
Any ideas why?

jvonau, 2017-12-16:
> http://archive.raspberrypi.org/debian stretch Release", " 503 Service Unavailable [IP: 93.93.130.214 80] 
  
Looks like the site was down. There is a delay between upstreams, that is the latest deb available http://archive.raspbian.org/raspbian/pool/main/c/calibre/ correct? You want 3.14, open an issue upstream would be my suggestion or flip to using the other installer and not debs.

holta, 2017-12-17:
> flip to using the other installer and not debs

If this work, is this something we should be doing in general on RPi possibly?

holta, 2017-12-17:
>> flip to using the other installer and not debs
>
> If this works, is this something we should be doing in general on RPi possibly?

Ah well, no luck trying this other approach: (/opt/iiab/downloads/calibre-installer.py)

> TASK [calibre : Run calibre-installer.py to install Calibre programs into /usr/bin - MANUALLY REMOVE /usr/bin/calibre-uninstall TO FORCE calibre-installer.py TO REINSTALL/UPGRADE HERE!] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/opt/iiab/downloads/calibre-installer.py >> /dev/null", "delta": "0:00:00.241774", "end": "2017-12-17 00:35:24.461616", "msg": "non-zero return code", "rc": 1, "start": "2017-12-17 00:35:24.219842", "stderr": "You are running on an ARM system. The calibre binaries are only available for x86 systems. You will have to compile from source.", "stderr_lines": ["You are running on an ARM system. The calibre binaries are only available for x86 systems. You will have to compile from source."], "stdout": "", "stdout_lines": []}

holta, 2017-12-18:
> http://archive.raspbian.org/raspbian/pool/main/c/calibre/

@jvonau is optimistic that Calibre 3.14[*] deb package might appear on the above site in coming days.

I remain concerned that it might be many months between supported releases of Calibre on RPi possibly?!

[*] released Friday 2017-12-15 @ https://calibre-ebook.com/whats-new

holta, 2017-12-23:
Resolving as @jvonau correctly predicted Calibre 3.14 .deb (and associated files) would appear at: http://archive.raspbian.org/raspbian/pool/main/c/calibre/

closed 2017-12-23 by holta