iiab/iiab #632: KA Lite 0.17.4-> 0.17.5

holta, 2017-12-23:
If https://github.com/learningequality/ka-lite/milestone/65 is possibly released in time?

benjaoming, 2018-01-24:
Hi @holta 

Just noticed this and wanted to say that we are probably not releasing 0.17.5 but instead going straight to 0.18.

The original intention was to fix problems in the content pack for 0.17.4, but we noticed that the new content pack would require further changes.

With the new content pack, we will allow the same video to show up in multiple topics. Before, videos would only show in one place, and then other occurrences would be hidden.

After building the new content pack and fixing the code base (i.e. the video download manager) to support this behavior, we'll release it as 0.18, and then stick to a new release cycle: New content pack, means new minor release bump.

For your case, I would suggest to upgrade to 0.18 once it's available. Older content packs will still be compatible with it, but I think (with the current content pack issue in mind) that users will benefit from the upgrade.

holta, 2018-01-24:
Profound thanks @benjaoming — am working with your Learning Equality colleague Hermes here in Mexico as we speak!

holta, 2018-07-08:
Update: an interim [KA Lite 0.17.5](https://github.com/learningequality/ka-lite/milestone/65) now seems likely thanks to @benjaoming's hard work.

holta, 2018-07-09:
Assistance is requested testing `pip install --pre ka-lite[-static]` for KA Lite 0.17.5b1 &mdash; those who can help: https://github.com/learningequality/ka-lite/pull/5582#issuecomment-403502988

holta, 2018-07-09:
FYI those IIAB implementers who want to experiment with KA Lite 0.17.5b1 can now do so, by following the guidelines in http://FAQ.IIAB.IO Item 23 ("Can I upgrade or reinstall server apps? ") to upgrade KA Lite:

0. Backup your KA Lite data (if you care)
1. Place `kalite_version: "0.17.5b1"` into [/opt/iiab/iiab/vars/local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml) (or [/opt/iiab/iiab/roles/kalite/defaults/main.yml](https://github.com/iiab/iiab/blob/master/roles/kalite/defaults/main.yml#L4))
2. Move /usr/local/kalite out of the way (FYI this contains 160+ MB "venv")
3. Move /library/ka-lite out of the way
4. cd /opt/iiab/iiab
5. ./runrole kalite
6. Reboot IIAB and clear your browser cache (not sure if both are required) to force 0.17.5b1 to appear at the bottom of each page similar to http://box:8008

Worked for me on the latest BIG-sized IIAB 6.6/master on Raspbian Lite on RPi 3 B+, thanks @benjaoming