iiab/iiab #636: establish default "no_admin: False" (instead of undefined) in default_vars & local_vars

holta, 2017-12-28:
Currently the top of http://wiki.laptop.org/go/IIAB/local_vars.yml & https://github.com/iiab/iiab/blob/master/vars/default_vars.yml are unnecessarily confusing, as these attempt to explain that no_admin: is undefined by default &mdash; and that any value at all is equivalent to True.

Making its "no_admin:" 's actual values True or False can avoid all these painfully long-winded explanations.

closed 2017-12-30 by holta

tim-moody, 2017-12-30:
why not follow the convention and make it admin_install: True in default vars?

holta, 2017-12-30:
> why not follow the convention and make it admin_install: True in default vars?

I can change the name of auto_admin: to admin_install: if that's best &mdash; in both [local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml) and [default_vars.yml](https://github.com/iiab/iiab/blob/master/vars/default_vars.yml)

If so, is its implied spouse variable `admin_enable:` relevant in any way?

tim-moody, 2017-12-30:
I agree that adds a complication. I just don't find auto_admin very descriptive. Perhaps create_admin_user. Whatever the name I'm not sure that anyone who sets it to false will have access to the admin console.

holta, 2017-12-30:
> I agree that adds a complication. I just don't find auto_admin very descriptive. Perhaps create_admin_user.

"admin_install:" seems better as it's doing a lot more than just creating the admin user &mdash; as explained in [local_vars.yml Lines 14-25](https://github.com/iiab/iiab/blob/master/vars/medium.localvars#L14-L25) it's also creating wheel group to monitor slackers' published passwords etc.

Basically it runs the 11 stanzas in [roles/iiab-admin/tasks/admin-user.yml](https://github.com/iiab/iiab/blob/master/roles/iiab-admin/tasks/admin-user.yml)

> Whatever the name I'm not sure that anyone who sets it to false will have access to the admin console.

@jvonau says it will work, if they follow the instructions above to fill in `iiab_admin_user: <my_sudo_user>` so I'm hoping he's correct!

tim-moody, 2017-12-30:
The additional problem I hit was that on a vps sudo can already be installed and configured, so https://github.com/iiab/iiab/blob/e9bbe7169c38879531db2776e4e0a85ebfe1b1e1/roles/2-common/tasks/packages.yml should not install sudo if admin_install: False
(or move the sudo package into admin_user.yml)

holta, 2017-12-30:
> The additional problem I hit was that on a vps sudo can already be installed and configured, so https://github.com/iiab/iiab/blob/e9bbe7169c38879531db2776e4e0a85ebfe1b1e1/roles/2-common/tasks/packages.yml should not install sudo if admin_install: False
(or move the sudo package into admin_user.yml)

Can you generate such a PR in coming days possibly?  No obligation, just if there's time..

commit ba60d771 (2017-12-30): Merge pull request #639 from holta/auto_admin

establish "auto_admin: True" (instead of no_admin: undefined) in default_vars & local_vars

closes #636