iiab/iiab #638: ping works for 10sec, fails for 18-45sec, then repeats (WiFi-installed IIAB/RPi)

holta, 2017-12-30:
@jvonau or @georgejhunt have you seen something like this before?  Finally we're getting closer to understanding the systemic failure of WiFi described in #558 ("WiFi-to-Internet fails regularly & doesn't recover on Raspbian Lite") that's been ongoing for months.

A misconfiguration of WiFi/routing must be causing this clear pattern of packet loss, that recurs every 28-to-55 seconds, when almost exactly 10 packets are returned, every time.  Consequently packet loss is always right around 70% across any multi-minute trial (see the 100+ lines pasted in below, so a representative test is visible from this complete dump over ~400 seconds).

Observations:
- This is IIAB 6.5/master installed using WiFi onto an RPi3 with Raspbian Lite 2017-11-29, currently at 10.8.0.42
- You can plug in a live Ethernet cable during one of the these ping sessions, to watch the problem disappear temporarily.  Then the exact same problem returns, the moment you pull out the Ethernet cable.
- _Rebooting temporarily solves the problem &mdash; for some minutes.  Causality is unclear, as a 3-hour install over WiFi generally works &mdash; but then the WiFi-to-Internet connection somehow "destroys itself" after some minutes of disuse?_
- _"systemctl restart networking" and "systemctl restart dhcpcd" do not solve anything.  What is a reboot doing above and beyond that, to fix the situation?_
```
root@box:~# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=57 time=20.8 ms
[23 missing]
64 bytes from 8.8.8.8: icmp_seq=25 ttl=57 time=886 ms
64 bytes from 8.8.8.8: icmp_seq=26 ttl=57 time=23.8 ms
64 bytes from 8.8.8.8: icmp_seq=27 ttl=57 time=23.4 ms
64 bytes from 8.8.8.8: icmp_seq=28 ttl=57 time=33.3 ms
64 bytes from 8.8.8.8: icmp_seq=29 ttl=57 time=21.9 ms
64 bytes from 8.8.8.8: icmp_seq=30 ttl=57 time=22.5 ms
64 bytes from 8.8.8.8: icmp_seq=31 ttl=57 time=42.1 ms
64 bytes from 8.8.8.8: icmp_seq=32 ttl=57 time=25.1 ms
64 bytes from 8.8.8.8: icmp_seq=33 ttl=57 time=29.3 ms
[9 successful above, then 45 missing]
64 bytes from 8.8.8.8: icmp_seq=79 ttl=57 time=564 ms
64 bytes from 8.8.8.8: icmp_seq=80 ttl=57 time=26.9 ms
64 bytes from 8.8.8.8: icmp_seq=81 ttl=57 time=21.8 ms
64 bytes from 8.8.8.8: icmp_seq=82 ttl=57 time=19.6 ms
64 bytes from 8.8.8.8: icmp_seq=83 ttl=57 time=26.7 ms
64 bytes from 8.8.8.8: icmp_seq=84 ttl=57 time=17.8 ms
64 bytes from 8.8.8.8: icmp_seq=85 ttl=57 time=20.4 ms
64 bytes from 8.8.8.8: icmp_seq=86 ttl=57 time=21.6 ms
64 bytes from 8.8.8.8: icmp_seq=87 ttl=57 time=20.7 ms
[9 successful above, then 24 missing]
64 bytes from 8.8.8.8: icmp_seq=112 ttl=57 time=1926 ms
64 bytes from 8.8.8.8: icmp_seq=113 ttl=57 time=887 ms
64 bytes from 8.8.8.8: icmp_seq=114 ttl=57 time=35.0 ms
64 bytes from 8.8.8.8: icmp_seq=115 ttl=57 time=20.8 ms
64 bytes from 8.8.8.8: icmp_seq=116 ttl=57 time=20.2 ms
64 bytes from 8.8.8.8: icmp_seq=117 ttl=57 time=20.6 ms
64 bytes from 8.8.8.8: icmp_seq=118 ttl=57 time=31.1 ms
64 bytes from 8.8.8.8: icmp_seq=119 ttl=57 time=19.9 ms
64 bytes from 8.8.8.8: icmp_seq=120 ttl=57 time=21.6 ms
64 bytes from 8.8.8.8: icmp_seq=121 ttl=57 time=20.9 ms
64 bytes from 8.8.8.8: icmp_seq=122 ttl=57 time=20.2 ms
[11 successful above, then 18 missing]
64 bytes from 8.8.8.8: icmp_seq=141 ttl=57 time=1924 ms
64 bytes from 8.8.8.8: icmp_seq=142 ttl=57 time=886 ms
64 bytes from 8.8.8.8: icmp_seq=143 ttl=57 time=19.9 ms
64 bytes from 8.8.8.8: icmp_seq=144 ttl=57 time=20.8 ms
64 bytes from 8.8.8.8: icmp_seq=145 ttl=57 time=21.3 ms
64 bytes from 8.8.8.8: icmp_seq=146 ttl=57 time=29.8 ms
64 bytes from 8.8.8.8: icmp_seq=147 ttl=57 time=20.7 ms
64 bytes from 8.8.8.8: icmp_seq=148 ttl=57 time=20.1 ms
64 bytes from 8.8.8.8: icmp_seq=149 ttl=57 time=20.8 ms
64 bytes from 8.8.8.8: icmp_seq=150 ttl=57 time=21.1 ms
[10 successful above, then 24 missing]
64 bytes from 8.8.8.8: icmp_seq=175 ttl=57 time=881 ms
64 bytes from 8.8.8.8: icmp_seq=176 ttl=57 time=24.1 ms
64 bytes from 8.8.8.8: icmp_seq=177 ttl=57 time=20.3 ms
64 bytes from 8.8.8.8: icmp_seq=178 ttl=57 time=23.4 ms
64 bytes from 8.8.8.8: icmp_seq=179 ttl=57 time=24.1 ms
64 bytes from 8.8.8.8: icmp_seq=180 ttl=57 time=27.9 ms
64 bytes from 8.8.8.8: icmp_seq=181 ttl=57 time=22.8 ms
64 bytes from 8.8.8.8: icmp_seq=182 ttl=57 time=22.2 ms
64 bytes from 8.8.8.8: icmp_seq=183 ttl=57 time=31.0 ms
[9 successful above, then 18 missing]
64 bytes from 8.8.8.8: icmp_seq=202 ttl=57 time=2964 ms
64 bytes from 8.8.8.8: icmp_seq=203 ttl=57 time=1925 ms
64 bytes from 8.8.8.8: icmp_seq=204 ttl=57 time=885 ms
64 bytes from 8.8.8.8: icmp_seq=205 ttl=57 time=24.3 ms
64 bytes from 8.8.8.8: icmp_seq=206 ttl=57 time=52.9 ms
64 bytes from 8.8.8.8: icmp_seq=207 ttl=57 time=20.6 ms
64 bytes from 8.8.8.8: icmp_seq=208 ttl=57 time=20.7 ms
64 bytes from 8.8.8.8: icmp_seq=209 ttl=57 time=26.5 ms
64 bytes from 8.8.8.8: icmp_seq=210 ttl=57 time=28.5 ms
64 bytes from 8.8.8.8: icmp_seq=211 ttl=57 time=20.8 ms
64 bytes from 8.8.8.8: icmp_seq=212 ttl=57 time=20.5 ms
[11 successful above, then 24 missing]
64 bytes from 8.8.8.8: icmp_seq=237 ttl=57 time=890 ms
64 bytes from 8.8.8.8: icmp_seq=238 ttl=57 time=33.7 ms
64 bytes from 8.8.8.8: icmp_seq=239 ttl=57 time=21.0 ms
64 bytes from 8.8.8.8: icmp_seq=240 ttl=57 time=21.5 ms
64 bytes from 8.8.8.8: icmp_seq=241 ttl=57 time=23.1 ms
64 bytes from 8.8.8.8: icmp_seq=242 ttl=57 time=20.8 ms
64 bytes from 8.8.8.8: icmp_seq=243 ttl=57 time=28.2 ms
64 bytes from 8.8.8.8: icmp_seq=244 ttl=57 time=21.8 ms
64 bytes from 8.8.8.8: icmp_seq=245 ttl=57 time=22.0 ms
[9 successful above, then 21 missing]
64 bytes from 8.8.8.8: icmp_seq=267 ttl=57 time=884 ms
64 bytes from 8.8.8.8: icmp_seq=268 ttl=57 time=20.6 ms
64 bytes from 8.8.8.8: icmp_seq=269 ttl=57 time=21.1 ms
64 bytes from 8.8.8.8: icmp_seq=270 ttl=57 time=21.9 ms
64 bytes from 8.8.8.8: icmp_seq=271 ttl=57 time=26.0 ms
64 bytes from 8.8.8.8: icmp_seq=272 ttl=57 time=20.9 ms
64 bytes from 8.8.8.8: icmp_seq=273 ttl=57 time=20.7 ms
64 bytes from 8.8.8.8: icmp_seq=274 ttl=57 time=20.5 ms
64 bytes from 8.8.8.8: icmp_seq=275 ttl=57 time=30.8 ms
[9 successful above, then 22 missing]
64 bytes from 8.8.8.8: icmp_seq=298 ttl=57 time=1926 ms
64 bytes from 8.8.8.8: icmp_seq=299 ttl=57 time=887 ms
64 bytes from 8.8.8.8: icmp_seq=300 ttl=57 time=23.9 ms
64 bytes from 8.8.8.8: icmp_seq=301 ttl=57 time=20.3 ms
64 bytes from 8.8.8.8: icmp_seq=302 ttl=57 time=20.7 ms
64 bytes from 8.8.8.8: icmp_seq=303 ttl=57 time=21.8 ms
64 bytes from 8.8.8.8: icmp_seq=304 ttl=57 time=20.7 ms
64 bytes from 8.8.8.8: icmp_seq=305 ttl=57 time=24.4 ms
64 bytes from 8.8.8.8: icmp_seq=306 ttl=57 time=21.7 ms
64 bytes from 8.8.8.8: icmp_seq=307 ttl=57 time=28.8 ms
[10 successful above, then 19 missing]
64 bytes from 8.8.8.8: icmp_seq=327 ttl=57 time=1925 ms
64 bytes from 8.8.8.8: icmp_seq=328 ttl=57 time=885 ms
64 bytes from 8.8.8.8: icmp_seq=329 ttl=57 time=30.1 ms
64 bytes from 8.8.8.8: icmp_seq=330 ttl=57 time=20.5 ms
64 bytes from 8.8.8.8: icmp_seq=331 ttl=57 time=20.0 ms
64 bytes from 8.8.8.8: icmp_seq=332 ttl=57 time=21.5 ms
64 bytes from 8.8.8.8: icmp_seq=333 ttl=57 time=21.3 ms
64 bytes from 8.8.8.8: icmp_seq=334 ttl=57 time=28.7 ms
64 bytes from 8.8.8.8: icmp_seq=335 ttl=57 time=21.4 ms
64 bytes from 8.8.8.8: icmp_seq=336 ttl=57 time=18.4 ms
[10 successful above, then 24 missing]
64 bytes from 8.8.8.8: icmp_seq=361 ttl=57 time=1924 ms
64 bytes from 8.8.8.8: icmp_seq=362 ttl=57 time=885 ms
64 bytes from 8.8.8.8: icmp_seq=363 ttl=57 time=30.2 ms
64 bytes from 8.8.8.8: icmp_seq=364 ttl=57 time=20.5 ms
64 bytes from 8.8.8.8: icmp_seq=365 ttl=57 time=25.9 ms
64 bytes from 8.8.8.8: icmp_seq=366 ttl=57 time=20.4 ms
64 bytes from 8.8.8.8: icmp_seq=367 ttl=57 time=20.2 ms
64 bytes from 8.8.8.8: icmp_seq=368 ttl=57 time=23.1 ms
64 bytes from 8.8.8.8: icmp_seq=369 ttl=57 time=21.3 ms
64 bytes from 8.8.8.8: icmp_seq=370 ttl=57 time=22.4 ms
[11 successful above]

--- 8.8.8.8 ping statistics ---
396 packets transmitted, 108 received, 72% packet loss, time 407148ms
rtt min/avg/max/mdev = 17.899/241.499/2964.385/552.383 ms, pipe 3
```

holta, 2017-12-30:
> - This is IIAB 6.5/master installed using WiFi onto an RPi3 with Raspbian Lite 2017-11-29, currently at 10.8.0.42

Correction: the particular RPi3 above was installed using https://github.com/holta/iiab/tree/ethernet in order to allow the Ethernet testing below.  But the exact same issue has been ongoing for the last couple months, as part of every WiFi-installed 6.5/master.

> - You can plug in a live Ethernet cable during one of the these ping sessions, to watch the problem disappear temporarily. Then the exact same problem returns, the moment you pull out the Ethernet cable.

holta, 2017-12-30:
> - _Rebooting temporarily solves the problem &mdash; for some minutes.  Causality is unclear, as a 3-hour install over WiFi generally works &mdash; but then the WiFi-to-Internet connection somehow [...]

CLARIF: I set the WiFi/DHCP router's lease time to 3 minutes (instead of the usual 120min) just in case this might be related.

This appears not to be related, certainly soon after reboot: "ping 8.8.8.8" continues to work properly while the RPi3 has renewed its WiFi/DHCP a dozen times.

holta, 2017-12-30:
Data Points:
- One the RPi3's exhibiting this glitchy #638 ping pattern (somehow) cured itself, after a few hours &mdash; while running ping &mdash; I was astonished as I'd never seen that happen ever before. Unfortunately I was not piping "ping 8.8.8.8" to a log file in that one case.  So I don't have a ping pattern log of what happened as the machine (10.8.0.30) somehow fixed itself about an hour ago (around 10AM EST, since rebooted).
- FWIW the above RPi3 was blacking out (losing all pings) a bit more often, for about 10-to-20sec each time, with outlier blackouts lasting 4sec, 30sec and 40 sec etc.  As always, ping's when working continuously last almost exactly 10 seconds.  Very occasionally pings work for 17-or-18sec. continuously, before the next ping-blackout...
- Presence or Absence of an OpenVPN connection (e.g. "systemctl stop openvpn@xscenet") seems to have no bearing at all on this #638 bug, neither inducing it nor or preventing it.
- Rebooting the router had no bearing, positively or negatively, as I tried several times.  I've restored the WiFi/DHCP router's lease time from 3min to 120min.
- _Regrettably, I still don't know how to induce #638, as (whatever causes it) typically tends to arise many hours after RPi3 boot :-(_

tim-moody, 2017-12-30:
maybe add ip a after failed ping to see if there is an ip addr
? swap routers

holta, 2017-12-30:
> maybe add ip a after failed ping to see if there is an ip addr

There is an IP address throughout, as issued by the WiFi router.

> ? swap routers

That's an idea, next time I catch one of these #638 failures in the act.

New Data:
- RPi3 10.8.0.30 (would normally be 192.168.0.132 if Ethernet-connected) began exhibiting the #638 glitchy pings, only 30min after boot!
- This RPi3 had lost its WiFi connection to the Internet, which I recovered using "systemctl restart dhcpcd" and then removed the errant route within "ip r" as described in #642 (allowing default route via 192.168.0.1 to Internet to again work, given its "metric 303" is a lower number than br0's erroneous route with "metric 204")
- I logged the output of "ping 8.8.8.8" to a file (excerpt below...)
- About 3min later, the problem cured itself &mdash; likely/apparently as a result of the 20-second WiFi drop/recovery that began ~167 seconds in &mdash; but I have no idea why :/
```
root@box:~# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=5 ttl=57 time=885 ms
64 bytes from 8.8.8.8: icmp_seq=6 ttl=57 time=24.9 ms
64 bytes from 8.8.8.8: icmp_seq=7 ttl=57 time=21.6 ms
64 bytes from 8.8.8.8: icmp_seq=8 ttl=57 time=17.7 ms
64 bytes from 8.8.8.8: icmp_seq=9 ttl=57 time=34.2 ms
64 bytes from 8.8.8.8: icmp_seq=10 ttl=57 time=21.5 ms
64 bytes from 8.8.8.8: icmp_seq=11 ttl=57 time=22.1 ms
64 bytes from 8.8.8.8: icmp_seq=12 ttl=57 time=21.3 ms
64 bytes from 8.8.8.8: icmp_seq=13 ttl=57 time=18.1 ms
[9 successful above, then 12 missing]
64 bytes from 8.8.8.8: icmp_seq=26 ttl=57 time=1926 ms
64 bytes from 8.8.8.8: icmp_seq=27 ttl=57 time=887 ms
64 bytes from 8.8.8.8: icmp_seq=28 ttl=57 time=19.7 ms
64 bytes from 8.8.8.8: icmp_seq=29 ttl=57 time=42.8 ms
64 bytes from 8.8.8.8: icmp_seq=30 ttl=57 time=26.4 ms
64 bytes from 8.8.8.8: icmp_seq=31 ttl=57 time=19.9 ms
64 bytes from 8.8.8.8: icmp_seq=32 ttl=57 time=22.3 ms
64 bytes from 8.8.8.8: icmp_seq=33 ttl=57 time=28.1 ms
64 bytes from 8.8.8.8: icmp_seq=34 ttl=57 time=30.7 ms
64 bytes from 8.8.8.8: icmp_seq=35 ttl=57 time=20.4 ms
[10 successful above, then 18 missing]
64 bytes from 8.8.8.8: icmp_seq=54 ttl=57 time=1931 ms
64 bytes from 8.8.8.8: icmp_seq=55 ttl=57 time=892 ms
64 bytes from 8.8.8.8: icmp_seq=56 ttl=57 time=20.3 ms
64 bytes from 8.8.8.8: icmp_seq=57 ttl=57 time=33.2 ms
64 bytes from 8.8.8.8: icmp_seq=58 ttl=57 time=21.5 ms
64 bytes from 8.8.8.8: icmp_seq=59 ttl=57 time=52.4 ms
64 bytes from 8.8.8.8: icmp_seq=60 ttl=57 time=23.7 ms
64 bytes from 8.8.8.8: icmp_seq=61 ttl=57 time=21.1 ms
64 bytes from 8.8.8.8: icmp_seq=62 ttl=57 time=24.0 ms
64 bytes from 8.8.8.8: icmp_seq=63 ttl=57 time=22.5 ms
[10 successful above, then 12 missing]
64 bytes from 8.8.8.8: icmp_seq=76 ttl=57 time=1924 ms
64 bytes from 8.8.8.8: icmp_seq=77 ttl=57 time=884 ms
64 bytes from 8.8.8.8: icmp_seq=78 ttl=57 time=24.3 ms
64 bytes from 8.8.8.8: icmp_seq=79 ttl=57 time=34.8 ms
64 bytes from 8.8.8.8: icmp_seq=80 ttl=57 time=21.5 ms
64 bytes from 8.8.8.8: icmp_seq=81 ttl=57 time=32.2 ms
64 bytes from 8.8.8.8: icmp_seq=82 ttl=57 time=23.9 ms
64 bytes from 8.8.8.8: icmp_seq=83 ttl=57 time=22.2 ms
64 bytes from 8.8.8.8: icmp_seq=84 ttl=57 time=20.8 ms
64 bytes from 8.8.8.8: icmp_seq=85 ttl=57 time=31.2 ms
[10 successful above, then 12 missing]
64 bytes from 8.8.8.8: icmp_seq=98 ttl=57 time=1364 ms
64 bytes from 8.8.8.8: icmp_seq=99 ttl=57 time=325 ms
64 bytes from 8.8.8.8: icmp_seq=100 ttl=57 time=24.3 ms
64 bytes from 8.8.8.8: icmp_seq=101 ttl=57 time=20.5 ms
64 bytes from 8.8.8.8: icmp_seq=102 ttl=57 time=25.5 ms
64 bytes from 8.8.8.8: icmp_seq=103 ttl=57 time=19.5 ms
64 bytes from 8.8.8.8: icmp_seq=104 ttl=57 time=35.8 ms
64 bytes from 8.8.8.8: icmp_seq=105 ttl=57 time=24.5 ms
64 bytes from 8.8.8.8: icmp_seq=106 ttl=57 time=22.9 ms
64 bytes from 8.8.8.8: icmp_seq=107 ttl=57 time=22.3 ms
[10 successful above, then 15 missing]
[... 48 lines cut ...]
64 bytes from 8.8.8.8: icmp_seq=197 ttl=57 time=19.9 ms
64 bytes from 8.8.8.8: icmp_seq=198 ttl=57 time=19.0 ms
64 bytes from 8.8.8.8: icmp_seq=199 ttl=57 time=20.7 ms
64 bytes from 8.8.8.8: icmp_seq=200 ttl=57 time=22.0 ms
64 bytes from 8.8.8.8: icmp_seq=201 ttl=57 time=19.7 ms
64 bytes from 8.8.8.8: icmp_seq=202 ttl=57 time=27.6 ms
64 bytes from 8.8.8.8: icmp_seq=203 ttl=57 time=21.3 ms
64 bytes from 8.8.8.8: icmp_seq=204 ttl=57 time=22.4 ms
64 bytes from 8.8.8.8: icmp_seq=205 ttl=57 time=28.6 ms
64 bytes from 8.8.8.8: icmp_seq=206 ttl=57 time=30.9 ms
64 bytes from 8.8.8.8: icmp_seq=207 ttl=57 time=22.4 ms
64 bytes from 8.8.8.8: icmp_seq=208 ttl=57 time=21.3 ms
64 bytes from 8.8.8.8: icmp_seq=209 ttl=57 time=34.1 ms
64 bytes from 8.8.8.8: icmp_seq=210 ttl=57 time=19.3 ms
64 bytes from 8.8.8.8: icmp_seq=211 ttl=57 time=22.0 ms
64 bytes from 8.8.8.8: icmp_seq=212 ttl=57 time=21.3 ms
64 bytes from 8.8.8.8: icmp_seq=213 ttl=57 time=20.5 ms
64 bytes from 8.8.8.8: icmp_seq=214 ttl=57 time=23.8 ms
64 bytes from 8.8.8.8: icmp_seq=215 ttl=57 time=28.4 ms
64 bytes from 8.8.8.8: icmp_seq=216 ttl=57 time=20.7 ms
64 bytes from 8.8.8.8: icmp_seq=217 ttl=57 time=22.4 ms
64 bytes from 8.8.8.8: icmp_seq=218 ttl=57 time=30.4 ms
64 bytes from 8.8.8.8: icmp_seq=219 ttl=57 time=20.1 ms
...
```

tim-moody, 2017-12-30:
sorry missed this
From 169.254.2.87 icmp_seq=167 Destination Host Unreachable means no ip from router

georgejhunt, 2017-12-30:
I think I'd look at /etc/dhcpcd line 41, to see if it makes sense.  (it
might help to print out the value of all the variables used in the
denyinterface statement with a debug, pause before the template statement
which writes /etc/dhcpcd).

I do not remember all the definitions of all the variables, and when they
are set, to determine whether the logic will create something that the
dhcpcd daemon will understand.

holta, 2017-12-30:
> I think I'd look at /etc/dhcpcd line 41

FWIW denyinterfaces is unrelated to this current #638 issue on both these 2 RPi3's.

On one RPi3 (10.8.0.30) denyinterfaces was already taken care of by PR #635 :
```
#denyinterfaces  wlan0
```
Inspired by the other RPi3's (10.8.0.38) where this was manually commented out to:
```
#denyinterfaces   eth0
```
_PS this intermittent WiFi/routing bug (#638) comes (and very occasionally goes away) despite the fact Ansible and ./iiab-network have not been run since earlier in the week.  The only way to reproduce it is to WiFi-install IIAB 6.5/master and then wait a few hours for WiFi drops to (presumably) make a mess of the router table (etc) &mdash; but note that #642 (pretty well understood, br0 route has the wrong precedence) is just one of the problems!_

georgejhunt, 2017-12-30:
> make a mess of the router table (etc) — but note that #642

I think both Tim, and I were on the same chain of logic -- ie that
169.254.x.x means that dhcp conversation is faulty.

As an experiment, you might try /etc/dhcpcd.conf, line 42 "denyinterfaces
br0"

Support for this hypothesis is found in /var/log/syslog (search for br0) on
10.8.0.30

holta, 2017-12-30:
> As an experiment, you might try /etc/dhcpcd.conf, line 42 "denyinterfaces br0"

Excellent Idea:

1. Even while "denyinterfaces br0" does not solve this current bug (#638) affecting 10.8.0.38[*] as we speak...

2. I'm greatly in favor of "denyinterfaces br0" as it Dramatically simplifies the output of "ip a" and "ip r" both (removing all references to 169.254.*.*).

3. That is if br0 is even needed at all on RPi3...which someone should explain to me one day...e.g. is br0 nec/wise after "iiab-hotspot-on" etc?

[*] ironically setting up "denyinterfaces br0" and then "systemctl restart networking; systemctl restart dhcpcd" somehow triggered this #638 bug behavior.  Which is a good thing: helping me get a (BIT) closer to reproducing this "ping-works-for-10-sec" pattern, intermixed with "networking-blackouts-roughly-every-15-to-30sec".

tim-moody, 2017-12-30:
If I understand this it means don't consider br0 as an interface for dhcpcd, which should always be true as we only use it for lan. I agree it would be rarely used on rpi; only if one added a usb dongle for external AP or something like that.

holta, 2017-12-30:
Ironically George's suggestion to add "denyinterfaces br0" is critical in 2 other ways:
1) It's a far cleaner way to solve #642
2) It's the 1st confirmed way to trigger this #638 WiFi/routing bug (on different RPi3's, yay finally it's reproducible!)

_In short: at long last I can now remove and re-add "denyinterfaces br0" repeatedly, with "systemctl restart networking; systemctl restart dhcpcd" in between...to force this very mysterious ping-for-10-seconds-only-and-then-it-cycles buggy behavior...until we can finally isolate what's causing it :-)_

holta, 2017-12-30:
For the record, here is example output _(during this #638 bugginess)_ after implementing "denyinterface br0" in /etc/dhcpcd.conf and then "systemctl restart networking" and if nec also "systemctl restart dhcpcd" :

> root@box:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group defaul
t qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOW
N group default qlen 1000
    link/ether b8:27:eb:87:ed:b1 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP g
roup default qlen 1000
    link/ether b8:27:eb:d2:b8:e4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.100/24 brd 192.168.0.255 scope global wlan0
       valid_lft forever preferred_lft forever
14: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN
group default qlen 1000
    link/ether de:80:fb:47:f8:dd brd ff:ff:ff:ff:ff:ff
15: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast st
ate UNKNOWN group default qlen 100
    link/none
    inet 10.8.0.30 peer 10.8.0.29/32 scope global tun0
       valid_lft forever preferred_lft forever

> root@box:~# ip r
default via 192.168.0.1 dev wlan0 src 192.168.0.100 metric 303
10.8.0.0/18 via 10.8.0.29 dev tun0
10.8.0.1 via 10.8.0.29 dev tun0
10.8.0.29 dev tun0 proto kernel scope link src 10.8.0.30
192.168.0.0/24 dev wlan0 proto kernel scope link src 192.168.0.100 metric 303
```

holta, 2017-12-30:
CLARIF: adding **_or_** removing "denyinterfaces br0" (followed by "systemctl restart networking; systemctl restart dhcpcd") seems to trigger #638's glitchy ping cycle.

Certainly for the ensuing many minutes (or many hours) until a reboot &mdash; or until the problem occasionally goes away on its own &mdash; seemingly aided by with a hard WiFi drop/reconnection (as a mere "systemctl restart dhcpcd" does not help!)

georgejhunt, 2017-12-30:
> Certainly for the ensuing many minutes (or many hours) until a reboot — or
> until the problem occasionally goes away on its own — seemingly aided by

Looking at syslog, the dhcp lease period is 180 seconds. It looks like the
dhcpcd daemon is tearing down, and rebuilding the connection, routing, etc
during the process.  That might explain ping dropout every 3 minutes.

Change the lease period to 1 day, and see if the problem goes away

holta, 2017-12-31:
3min DHCP are on purpose to debug.  Longer leases don't help fyi.  A bit
more backstory on tkt.

holta, 2017-12-31:
Clarifications:
- ping shows the exact same pattern[*] after having swapped out the WiFi router
- ping shows the exact same pattern after changing its WiFi SSID/password
- ping shows the exact same pattern after changing ISP's from wireline/cable to mobile/4G/LTE
- ping shows the exact same pattern when lease-time is 3min, 1h, 2h, etc &mdash; as seen by searching for dhcpcd in /var/log/syslog

[*] ping 8.8.8.8 works for almost exactly 10 seconds.  Then ping packets disappear for the next 15-to-30 seconds typically.  And the cycle repeats &mdash; sometimes for hours on end &mdash; sometimes it fixes itself after the building's WiFi completely disappears for a few seconds or such.

[**] The glitchy ping pattern can start on its own, and go on for hours!  It typically starts (and stop) when the RPi3 completely loses WiFi (and hence Internet) access for a brief moment.  Often as a result of "systemctl restart dhcpcd" or similar.

Important: changing the WiFi/DHCP lease time (to 24h, or more!) is not a practical solution here, as it does not empower the broader public, which is the entire goal here.  Worse, this risks papering over the core underlying challenge &mdash; how to make WiFi installations of IIAB hassle-free for individuals across a wide array of WiFi-enabled dorms/buildings worldwide.

(IIAB 6.5's main goal since early October is to help the public install IIAB over ever-flaky WiFi conditions, in situations where they inevitably have little to no access to the router.  Corollary: if the public really could modify their WiFi/DHCP router, they would (certainly should!!) definitely use Ethernet instead :)

holta, 2017-12-31:
Another (Scarier!) Data Point...hinting at the underlying 10-SEC-IS-ALL-THE-WIFI-YOU-GET problem:
- "ping 8.8.8.8" and "ping 192.168.0.1" respond 10 TIMES TOTAL **(and never respond again, until I run "systemctl restart dhcpcd" ...for another 10 PINGS TOTAL)** with the setup below...
- service is restored by running "ip a flush dev wlan0" followed by "systemctl restart dhcpcd" which restored the output of "ip a" and "ip r" to what they are above (instead of this below...)

> root@box:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group defaul
t qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOW
N group default qlen 1000
    link/ether b8:27:eb:87:ed:b1 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP g
roup default qlen 1000
    link/ether b8:27:eb:d2:b8:e4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.100/24 brd 192.168.0.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet 192.168.43.127/24 brd 192.168.43.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet 192.168.0.101/24 brd 192.168.0.255 scope global secondary wlan0
       valid_lft forever preferred_lft forever
44: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN
group default qlen 1000
    link/ether ea:ac:cc:37:50:5e brd ff:ff:ff:ff:ff:ff

> root@box:~# ip r
default via 192.168.0.1 dev wlan0 src 192.168.0.101 metric 303
192.168.0.0/24 dev wlan0 proto kernel scope link src 192.168.0.101 metric 303
192.168.43.0/24 dev wlan0 proto kernel scope link src 192.168.43.127

holta, 2017-12-31:
More Observations coming to light:
- A completely Insane (but effective) way to keep RPi3's WiFi-to-Internet alive _(when faced with this nearly ever-present WiFi failure, on an RPi3 that you need to keep online for long durations, during IIAB content provisioning etc!)_ is to run "systemctl restart dhcpcd" constantly, renewing the WiFi/DHCP lease constantly ~9 seconds or so.
- **Fascinating aside: "systemctl restart dhcpcd" ALSO brings the WiFi network back to life (for 10 seconds anyway) even during WiFi blackout periods (that typically last 15-to-45sec).  _Somehow DHCP's packets are higher priority than ping's packets, and do NOT get dropped!_**
- Strange Pattern I do not understand: WiFi blackout periods are _very_ consistent in their length:
  - occasionally that means 15-to-20 seconds
  - occasionally that means 20-to-30seconds
  - and at other times that means 30-to-45 seconds
- During those times that WiFi is "cured" (working steadily, without the 10SEC disease) there is almost always a slow "422ms" ping approximately every 30 seconds or so, that looks suspiciously like it may be the trigger for this #638 bug (or the antivenom to keep the cycle going?)  In any case the consistency (and similarity) of that pattern is Potentially very suspicious...

jvonau, 2017-12-31:
> root@box:~# ip r
default via 192.168.0.1 dev wlan0 src 192.168.0.101 metric 303
192.168.0.0/24 dev wlan0 proto kernel scope link src 192.168.0.101 metric 303

192.168.43.0/24 dev wlan0 proto kernel scope link src 192.168.43.127  <<-- where is this coming from?

holta, 2017-12-31:
>> 3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP g
    roup default qlen 1000
    link/ether b8:27:eb:d2:b8:e4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.100/24 brd 192.168.0.255 scope global wlan0
    valid_lft forever preferred_lft forever
    inet 192.168.43.127/24 brd 192.168.43.255 scope global wlan0
    valid_lft forever preferred_lft forever
    inet 192.168.0.101/24 brd 192.168.0.255 scope global secondary wlan0
>
> why both .100/.101?

The TL-WR841N WiFi/DHCP router provided new IP address 192.168.0.101 for whatever (normal) reasons &mdash; likely b/c 192.168.0.100 was leased to another client device in the interim &mdash; during that time when the RPi3/IIAB was connected to the **other router** below (router below led to another ISP, via 192.168.43.*)

>>    root@box:~# ip r
    default via 192.168.0.1 dev wlan0 src 192.168.0.101 metric 303
    192.168.0.0/24 dev wlan0 proto kernel scope link src 192.168.0.101 metric 303
>
>192.168.43.0/24 dev wlan0 proto kernel scope link src 192.168.43.127 <<-- where is this coming from?

jvonau, 2017-12-31:
Why the rpi is still retaining the dhcp lease info from an access point that is no longer connected is really interesting.  I would of expected that once the connection to an AP is lost the ip address/mac address(arp) info would be flushed and the whole dhcp lease request be started from the beginning, but that does not appear to be occurring. Does the **other router** share the same SSID?

holta, 2017-12-31:
> Why the rpi is still retaining the dhcp lease info from an access point that is no longer connected is really interesting. I would of expected that once the connection to an AP is lost the ip [...]

No.  All leases seem to retained ad nauseam as files of the form:
```
/var/lib/dhcpcd5/dhcpcd-wlan0-<SSID>.lease
```
...until I run "ip a flush dev wlan0" (perhaps this is harmless, anyway just an observation!)

holta, 2017-12-31:
SUMMARY OF KNOWN FACTS:
- the 10-SEC-SPURTS (IS-ALL-THE-WIFI-YOU-GET) disease is triggered when:
   - "denyinterfaces br0" is **absent** from /etc/dhcpcd.conf
   - 169.254.x.y is mysteriously added to br0 as shown by "ip a", as explained in #642 (and visible in /var/log/syslog ...still unclear why this happens very sporadically!)
   - thereupon, "systemctl restart dhcpcd" (and many similar events, e.g. when WiFi "carrier lost" appears in /var/log/syslog) will often trigger the 10-SEC disease/symptoms.
- the 10-SEC disease continues even after "ip a" and "ip r" are cleaned up &mdash; i.e. even after all br0 references have been removed from "ip r" &mdash; and all 169.254.x.y references have been removed from "ip a" and "ip r"
- the 10-SEC disease can apparently be "cured" with any of these 3 techniques &mdash; though a root-cause cure would be preferred, if we can hopefully later get there:
   - "ip a flush dev wlan0; systemctl restart dhcpcd" (presumably after removing all 169.254.x.y references in "ip a" and "ip r")
   - an accidental WiFi (carrier) drop i.e. when "carrier lost" appears in /var/log/syslog (presumably after removing all 169.254.x.y references in "ip a" and "ip r")
   - a reboot

holta, 2017-12-31:
> - 169.254.x.y is mysteriously added to br0 as shown by "ip a", as explained in #642 (and visible in /var/log/syslog ...still unclear why this happens very sporadically!)

An example of br0's problematic 169.254.x.y spontaneously appearing is pasted in here: https://github.com/iiab/iiab/issues/642#issuecomment-354622097

tim-moody, 2017-12-31:
why is udev involved?

jvonau, 2017-12-31:
> 169.254.x.y is mysteriously added to br0 as shown by "ip a", as explained in #642 (and visible in /var/log/syslog ...still unclear why this happens very sporadically!)

That maybe due to dhcpcd.conf lacking the static ip address info for br0?

> {% if dhcpcd_result == "enabled" and iiab_lan_iface != "none" %}
interface {{ iiab_lan_iface }}
static ip_address={{ lan_ip }}/19
static domain_name_servers=127.0.0.1
{% endif %}

when both conditions are met, recorded in iiab.ini's detected_networks, that stanza should be present.

jvonau, 2017-12-31:
> why is udev involved?

from #642 
> Dec 31 12:26:05 localhost dhcpcd[10035]: dev: loaded udev

That would a call to reload the hardware drivers:
>Dec 31 12:26:05 localhost dhcpcd-run-hooks[10054]: wlan0: starting wpa_supplicant
Dec 31 12:26:05 localhost vmunix: [92548.988717] brcmfmac: power management disabled
Dec 31 12:26:05 localhost dhcpcd[10035]: DUID

holta, 2017-12-31:
> - the 10-SEC-SPURTS (IS-ALL-THE-WIFI-YOU-GET) disease is triggered when:
>    - "denyinterfaces br0" is **absent** from /etc/dhcpcd.conf
>    - 169.254.x.y is mysteriously added to br0 as shown by "ip a", as explained in #642

CORRECTION: I've now witnessed at least one case where the 10SEC disease began even **without** 169.254.x.y and br0 ever appearing in "ip a", "ip r" and /var/log/syslog &mdash; the mystery continues :/

jvonau, 2017-12-31:
Is more than one of those .100/.101/.43.127 addresses present?

holta, 2017-12-31:
> CORRECTION: I've now witnessed at least one case where the 10SEC disease began even **without** 169.254.x.y and br0 ever appearing in "ip a", "ip r" and /var/log/syslog &mdash; the mystery continues :/

Observation: several times I've been able to trigger the 10SEC disease by inserting an Ethernet cable (eth0) into the RPi3 (when "denyinterfaces br0" is **absent** from /etc/dhcpcd.conf) and then removing it a few minutes later.

I don't (yet?) understand what the causality path might be here, as "ip a" and "ip r" show no apparent differences before/after eth0 came and went...

holta, 2017-12-31:
> Is more than one of those .100/.101/.43.127 addresses present?

No.  That 192.168.43.127 was just an isolated test late last night to reconfirm to everybody (and myself!) that this #638 10SEC disease occurs regardless of router, regardless of DHCP lease time, and regardless of ISP.

holta, 2017-12-31:
FYI @jvonau tried "iwconfig wlan0 power off" to see if turning off the Power Management in the RPi3's WiFi module itself might help, similar to OLPC's XO laptops.

The on/off Power Mgmt settings inside the RPi3's WiFi module (and other settings) are visible with ""iwconfig wlan0"

So far the results are inconclusive &mdash; but FYI this new strategy is promising as it's time to seek out new leads towards diagnosing the root cause here.

holta, 2018-01-01:
Here are 2 ways to reproduce the 10-SEC-OF-WIFI-IS-ALL-YOU-GET disease on virgin Raspbian Lite on RPi3, after connecting to WiFi using raspi-config -> 2 Network Options -> N2 Wi-Fi:
1) run "systemctl restart dhcpcd; systemctl restart dhcpd"
2) run "systemctl restart dhcpcd" then insert a live Ethernet cord for fully 5 minutes.

In either case above, observe that WiFi is diseased using "ping 8.8.8.8" or similar &mdash; the pattern is that ping will work for 10 SECONDS and then black out for 15-to-45 seconds typically &mdash; before repeating this cycle.  See the ping traces further above on this ticket.

**Mitigation:** run "ip a flush dev wlan0; systemctl restart dhcpcd" and your WiFi will start working again, for a while anyway ;)

**Observation:** initially WiFi blackout durations are often 10-15 sec each, but grow to be more than 30 sec each later?  Suggests a memory leak or buffer overflow, that gradually worsens over time, hopefully not firmware-related?

**Collateral Damage:** the design goal of Internet-in-a-Box 6.5 (i.e. install IIAB software onto a Pi3 then provision content entirely over WiFi, in dorm-like situations) is not achievable if the OS/hardware cannot reliably sustain WiFi in real-world multi-hour / multi-day situations &mdash; during long downloads of software and even longer downloads of content packs.

Pragmatically, this suggests WiFi installs of IIAB & content are not quite yet realistic, unless the RPi3 is rebooted **_often_** (after any extended period of inactivity and/or insertion of Ethernet cables...)

quozl, 2018-01-01:
I've reviewed this issue from top to bottom because Adam drew my attention to it by posting on OLPC's server-devel@ mailing list!

The issue needs;
* reproducing on distributed Raspbian _without_ using IIAB; possibly this is done in @holta's most recent post here, but the post also mentions IIAB so I'm not so sure,
* reproducing by a developer with investigative skills covering the full stack; Debian, systemd, dhcpcd,
* less speculation and random workaround attempts by developers who don't have the time or willingness to reproduce or dig into source code; it begins to look like prioritisation bullying,

However, I'll speculate a bit too, as I certainly don't have the _data budget_ to fetch IIAB.

* this reminds me of a DHCP server that hasn't noticed an IP address it allocates is in use already on a network,
* this reminds me of multiple DHCP servers on a LAN; which could be proved or disproved using `tcpdump` and DHCP logs on the Raspberry Pi 3, or `tcpdump` on a host elsewhere on the network,
* this does not happen for me with Raspbian Lite, dated 2017-04-10, using Raspberry Pi 3, on my local area network; if @holta could test with different versions of Raspbian Lite they would be able to prove or disprove regression in Raspbian,
* a ping at the same time from another system on the same network to the Raspberry Pi 3's IP address will help isolate whether the loss is associated with ARP or the temporarily programmed MAC address in the wireless chipset,
* applying all pending system updates to the Raspbian system should be tried and the result reported; no change, or not,
* try a different Raspberry Pi 3 fresh from an Element 14 anti-static bag, as they are usually not well enclosed and are easily damaged by electrostatic discharge on an unprotected desk; and the wireless antenna is an obvious path for the damage,
* it's probably `systemd` again. :grin:

holta, 2018-01-01:
> reproducing on distributed Raspbian without using IIAB

> try a different Raspberry Pi 3

Clarif: both above were done earlier, on 3 different Raspberry Pi 3's, running Raspbian Lite from 2017-11-29.

holta, 2018-01-01:
> applying all pending system updates to the Raspbian system should be tried and the result reported; no change, or not

Indeed.  This was tried and there was no change after applying the 22 current "apt update" suggestions to Raspbian Lite from 2017-11-29 (that's the version that IIAB requires).

quozl, 2018-01-01:
> > try a different Raspberry Pi 3
> both above were done earlier, on 3 different Raspberry Pi 3's, ...

Good, thanks.  No change to my other recommendations.

> > apply all pending system updates
> there was no change 

Good.  You have excluded a known fixed problem.  What remains is either a known or unknown unfixed problem.  :grin:  No change to my other recommendations.

>  (that's the version that IIAB requires).

Irrelevant to the problem now, please don't keep bringing up irrelevancies. :smirk_cat: 

I suggest again reproducing the problem with exactly the same ping results but without using @holta's local area network.  You need to exclude that network as the root or contributing cause, in order to find what the actual cause is.  Then go back and find what it is about the network that you need a workaround for.

holta, 2018-01-01:
New Data Point:

This bug was reproduced on Raspbian Desktop 2017-11-29 (as opposed to Raspbian Lite).  For some reason the tester could not get Method 1 ("systemctl restart dhcpcd; systemctl restart dhcpd") to trigger the ping bug.  But Method 2 ("systemctl restart dhcpcd" then insert a live Ethernet cord for exactly 5 minutes) did trigger the 10SEC WiFi bug immediately.  FWIW as is commonly the case, the ON/OFF cycles began at about 30 seconds each, and later on extended to about 50 seconds each.

quozl, 2018-01-01:
> New Data Point

Was this on the same @holta's local area network?  If not, you're probably reporting two different problems that have a similar symptom; packet loss reported by ping.  No change to my other recommendations.

quozl, 2018-01-02:
Problems as you describe with wireless alone do not reproduce for me despite trying with current Raspbian.  Either my network is different (which again points at @holta's network), or my method of configuring wireless is different.  My method is to add lines to `/etc/wpa_supplicant/wpa_supplicant.conf` manually rather than use `raspi-config`.

Problems as you describe on adding or removing ethernet cable do occur, but are _asymmetric_; that is their effect depends on whether you are testing with outbound or inbound ping, or by making TCP connections.

On adding ethernet cable from same network, immediately the route preference properly changed to use ethernet as default.  An existing ping took six seconds to begin using the new route.

On removing ethernet cable, immediately the route preference properly changed to wireless as default.  An existing ping outbound took six to nine seconds before it began to use wireless.  An existing ping inbound took up to 40 seconds.

I don't think these problems are a bug, because removing the ethernet cable is not a common use case; but if you want to explore it the relevant components are kernel driver for wireless, kernel driver for ethernet, kernel networking, dhcpcd, and possibly systemd.

An outbound ping test started _immediately_ after removing the ethernet cable shows no problem.

I don't think outbound ping is a reliable indicator of a properly operating network connection.  Ping is not aware of changing network interfaces, and keeps the same kernel socket open despite the underlying transition.  You cannot draw good conclusions from a ping running on the Rasbian system at the time the network interfaces change.

So to return to your wireless-only problem, useful logs from an _affected_ system would be;
* `journalctl`
* `tcpdump` (for ARP, BOOTP, DNS, and associated protocols).

Add to this your intended use of mDNS to resolve IP from host name, and the cache validity timings for mDNS, and things will get interesting.

holta, 2018-01-02:
>> New Data Point [reproduced with Raspbian Desktop]
>
> Was this on the same @holta's local area network?

Yes but the test was run with all computers and all radios turned off in the entire house (except for WiFi router Num1, naturally, to perform the test...)

Then I turned off that WiFi router too (to be sure) and reproduced this with an entirely different WiFi router (router Num2, to a different ISP...)

Still your idea is an excellent one.  I want to test this problem in more locations now, and with other people, so we can narrow down what exactly's causing this.

> I don't think outbound ping is a reliable indicator

I understand that "ping 8.8.8.8" and "ping 192.168.0.1" aren't the most perfect tests in the world.  However their results perfectly match the sought functionality of browsers (in both directions, to the RPi and from the RPi) and ssh (likewise in both directions) etc, so it seems to be an extremely effective if simple-minded test &mdash; that no doubt has its flaws!

ASIDE: there is a 3rd way this 10SEC syndrome can be triggered (sometimes, not other times!) and that is by running "systemctl restart dhcpcd" repeatedly.  Sometimes (again, this doesn't always work) the 10SEC disease kicks in after the 3rd or 4th or 5th time you run "systemctl restart dhcpcd".  Of course sometimes I have to wait 20-30 seconds, when watching one of these ping streams, as the 10SEC disease/pattern usually kicks in immediately &mdash; but occasionally kicks in 10-20 sec after the trigger &mdash; whichever of the 3 triggers brings on the malignancy.

georgejhunt, 2018-01-02:
On my version of raspbian lite 11/29/17, dhcpd.service is not present in
the image, so method #1 is a noop -- ie. "systemctl restart dhcpd" fails
with a "cannot find unit file error"

Within the wifi environment at my location, I am not able to replicate
intermittent ping results via restart dhcpcd, and then inserting a live
wire.

Unrelated observation:

There seems to be an overlayfs, or some other mechanism, which causes the
/boot partition to accept a "touch /boot/ssh" without error. But then after
reboot, the /boot/ssh has disappeared.  /etc/fstab sets default options for
/boot, and "mount" indicates that /dev/mmcblkp1 has been mounted "rw". So I
do not understand what I have observed.

This may surface as a problem for headless installs where "touch ssh" and
<edit wpa-supplicand.conf> are required in /boot in order to enable a ssh
client connection.

So then I installed IIAB via wifi, rebooted, and did a ping 8.8.8.8 for 546
seconds without any packet loss.

I think we need to find more folks that have Adam's problem, and pin down
the wifi environment which is causing this symptom.

Or maybe I did not understand when the problem surfaces.

But it looks to me like I have a functioning "appliance". Home page works,
as does the admin-console.

holta, 2018-01-02:
> On my version of raspbian lite 11/29/17, dhcpd.service is not present in
the image, so method #1 is a noop -- ie. "systemctl restart dhcpd" fails
with a "cannot find unit file error"

@georgejhunt I'm extremely confused how this is even possible: "systemctl restart dhcpcd" works on both Raspbian Lite 2017-11-29 and Raspbian Desktop 2017-11-29 — from the very moment I install these OS's — I do not run **_any_** other commands, only raspi-config to connect to WiFi.

Profound thx George & James for trying!  I'm starting to wonder if my RPi3's might be a different HW vintage possibly?  If so: any ways I can verify its WiFi module model number or other HW characteristics that might matter?  Note: I have purchased these RPi3's in different stores in the US over the past 18 months, and never had a HW problem with them to date.

(Yes I'll also test in several other locations in coming days...so we can avoid paranormal claims that my home is hexed ;)

quozl, 2018-01-02:
Just a typo. @georgejhunt is typing `dhcpd` but @holta and I are typing `dhcpcd`.  One letter.

But I don't think playing with `dhcpcd` like that is the right approach to this problem.

No change to my recommendations; still think especially that `journalctl` from an affected system may be _useful_.

holta, 2018-01-02:
> Just a typo. @georgejhunt is typing dhcpd but @holta and I are typing dhcpcd. One letter.

Huge thanks @quozl.

Thanks George for persisting with a few dhcpcd experiments where poss.

PS I don't know enough about journalctl and tcpdump, but as soon as others replicate this, we can hopefully shed light here.

quozl, 2018-01-02:
@holta, do some reading!  You're the only one reproducing it at the moment, so you're it.  Reproduce the problem, then type `sudo journalctl > file.txt`, then put the file somewhere for us to see.  It's really easy.

holta, 2018-01-02:
> Reproduce the problem, then type sudo journalctl > file.txt

Done.  See attached journalctl-output.txt
[journalctl-output.txt](https://github.com/iiab/iiab/files/1596815/journalctl-output.txt)

holta, 2018-01-02:
Possibly better than above...please instead see journalctl-output2.txt attached here:

[journalctl-output2.txt](https://github.com/iiab/iiab/files/1596824/journalctl-output2.txt)

quozl, 2018-01-02:
@holta, thanks.

Your Raspbian system during boot acquired wireless connection, lost it within one second, and then reacquired it.  My system acquired connection and kept it.  Pertinent log entry is;

```
Jan 02 01:05:28 raspberrypi dhcpcd[379]: wlan0: carrier lost
```

This points at your wireless access point hardware or firmware, your local radio frequency environment, or a de-authentication attack.

Your Raspbian system some time after boot was offered a DHCP lease by your DHCP server, but the negotiation faltered, and the Raspbian system went on to use a link-local address instead until the DHCP server finally responded.  My system acquires a lease quickly and probes are replied to quickly.  Pertinent log entries are;

```
Jan 02 01:07:32 raspberrypi dhcpcd[1060]: wlan0: offered 192.168.43.165 from 192.168.43.1
Jan 02 01:07:32 raspberrypi dhcpcd[1060]: wlan0: probing address 192.168.43.165/24
Jan 02 01:07:36 raspberrypi dhcpcd[1060]: wlan0: using IPv4LL address 169.254.174.144
...
Jan 02 01:07:37 raspberrypi dhcpcd[1060]: wlan0: leased 192.168.43.165 for 3600 seconds
```

Your network has a very slow DHCP server which is not answering the initial DHCP query quickly enough when the Raspbian system boots, and is not responding quickly enough to address probing when the Raspbian system decides to try re-using the previous lease.

As a result, sometimes a link-local address is being assigned, which you can find in the logs as 169.254.174.144.  This will make ping go silent until the condition resolves several seconds later.

I'm presuming you are using the DHCP server included within a wireless access point.  What is the manufacturing date of the access point?  When was the access point firmware version released?  Does the problem change at all after restarting the access point?  A compromised router can cause this kind of problem, as it may have processes on it mining bitcoin or doing something unusual, stealing resources from the DHCP server.

The `dhcpcd` service is also being stopped and restarted frequently, presumably by you.  I'd like to see a test without that, as restarting `dhcpcd` is not normal use scenario, and will certainly interrupt connections.

One might be tempted to solve this with either;
* manual static IP configuration; which is difficult for ordinary people, or;
* changing the timings used by dhcpcd; but this may violate other parts of the standard and bring new problems from otherwise working configurations.

A `tcpdump` made on a DHCP server would confirm the behaviour observed.

Hope that helps.

holta, 2018-01-02:
I just open a brand new WiFi/DHCP router from its shrinkwrapped plastic, installed its latest firmware _(without connecting it to the Internet)_ and reproduced the exact same WiFi disease on an unadulterated Raspbian Desktop 2017-11-29 on RPi3.  I find use Method 2 to be 100% reliable at reproducing the bug on Raspbian Desktop:

- place the WiFi credential into /etc/wpa_supplicant/wpa_supplicant.conf (e.g. use raspi-config)
- run "systemctl restart dhcpcd"
- insert a live Ethernet-to-Internet cord for exactly 5min
- remove the Ethernet cord
- observe the sickness by running "ping 192.168.0.1" or similar

RECAP: this (3rd of 3) tested routers has never been connected to the Internet (or to _any_ other network), has certainly not been compromised, and the 10-SEC-OF-WIFI-IS-ALL-YOU-GET (about once every minute!) disease continues as always :/

journalctl_output3.txt attached, in case this latest is also helpful:

[journalctl-output3.txt](https://github.com/iiab/iiab/files/1596903/journalctl-output3.txt)

quozl, 2018-01-02:
> run "systemctl restart dhcpcd"

Doesn't seem the right thing to do.  See [Setting WiFi up via the command line](https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md) where it says you should edit `wpa_supplicant` then use;

```
wpa_cli -i wlan0 reconfigure
```

> observe the sickness by running "ping 192.168.0.1" or similar

You are doing this from the Raspberry Pi?  Still doesn't happen for me.

> Method 1 ("systemctl restart dhcpcd; systemctl restart dhcpcd") ...seem to generally work [for me] at reproducing the bug 

And I'm asserting that you're reproducing a different bug when you do that different thing, because restarting dhcpcd twice is not a normal thing to do.  Please stick to one problem at a time?

> journalctl_output3.txt attached

Reviewed, thanks.

Both the wireless router and the ethernet router have the same IP address of 192.168.0.1, yet they are not the same host, and not part of the same network.  Has this always been the case in your failing tests?  What if the ethernet router and the wireless router are on the same network with the same IP address?  That's the configuration I'm testing.

You might see if the problem reproduces with different IP addresses for the different wireless router and ethernet routers.

For the technical guys, have a look at `/proc/sys/net/ipv4/conf/*/arp_filter` which defaults to zero,

```
	0 - (default) The kernel can respond to arp requests with addresses
	from other interfaces. This may seem wrong but it usually makes
	sense, because it increases the chance of successful communication.
	IP addresses are owned by the complete host on Linux, not by
	particular interfaces. Only for more complex setups like load-
	balancing, does this behaviour cause problems.
```

and `arp_announce` which defaults to zero as well.  See ip-sysctl.txt in kernel source.

My understanding of Linux ARP is that connecting to separate networks with the same IP addresses won't work correctly, so don't try it.  You should either configure the system to bring down WiFi when ethernet comes up, or reconfigure those routers.

holta, 2018-01-02:
Surprising New Data:

**Observation 1:** if Raspbian Lite is experiencing 10SEC WiFi disease, it will occur again shortly after reboot if you:
- (immediately after reboot) run "systemctl restart dhcpcd"
- run "ping to 192.168.0.1" (or similar, to verify syndrome)

_Amazingly, this will not work in the other order:_
- (immediately after reboot) run "ping to 192.168.0.1" for at least 1 second
- run "systemctl restart dhcpcd"

_In short, some version of the "Heisenberg Uncertainty Principle" is at work here &mdash; as the very use of "ping" evidently affects the outcome!_

**Observation 2:** Method 2 of inducing this overall bug often seems to work when inserting a live Ethernet-to-Internet cord onto the RPi3 for only 3 minutes (after "systemctl restart dhcpcd" was run) with perhaps 80% certainty) &mdash; whereas inserting the live Ethernet-to-Internet cord for a full 5 minutes seems to induce the syndrome with perhaps 90% certainty.  Finally, inserting the Ethernet cord for just 2 minutes does not trigger the syndrome usually...

quozl, 2018-01-02:
There is saved state.

Ping does fill in the ARP cache of both the Raspberry Pi and the routers.

Restarting dhcpcd _might_ empty the ARP cache of the Raspberry Pi.  It depends on when ARP packets are received.  It won't empty the ARP caches of the routers.

To look at that cache; `sudo arp -n`.

holta, 2018-01-02:
> Both the wireless router and the ethernet router have the same IP address of 192.168.0.1, yet they are not the same host, and not part of the same network. Has this always been the case in your failing tests?

No.  The original test case had 192.168.0.1 as the (same) DHCP router, accessible through both paths WiFi and Ethernet.  Precisely as you described below, yeah.

> What if the ethernet router and the wireless router are on the same network with the same IP address? That's the configuration I'm testing.

> You might see if the problem reproduces with different IP addresses for the different wireless router and ethernet routers.

Worthwhile experiment.  I set the WiFi router to 192.168.13.1 and the Ethernet router remains 192.168.0.1 &mdash; the 10SEC WiFi disease can be reproduced, but it seems possibly a bit harder to reproduce in these conditions:

1) The trick of rebooting an RPi3 that's currently affected by the syndrome (and then immediately running "systemctl restart dhcpcd") no longer works to trigger/retain the syndrome.

2) FWIW I triggered the 10SEC syndrome by running "systemctl restart dhcpcd", then inserting the Ethernet cable for 3-to-5 min, which always/eventually worked (4 times in a row) but it's possible that finally typing "ping 192.168.0.1" from the RPi3 might possibly help bring on the malignancy &mdash; or that could be just a coincidence that occurred twice, I'm not sure.  Anyway the usual symptoms sometimes begin shortly after typing "ping 192.168.0.1", which is not actually reachable after I've disconnected the Ethernet, and so shows this output:
```
PING 192.168.0.1 (192.168.0.1) 56(84) bytes of data.
From 192.168.13.1 icmp_seq=1 Destination Net Unreachable
...
```
journalctl-output4.txt attached:

[journalctl-output4.txt](https://github.com/iiab/iiab/files/1597139/journalctl-output4.txt)

holta, 2018-01-02:
> wpa_cli -i wlan0 reconfigure

I agree that using the above (or `dhcp --rebind` or `dhcpcd --renew` etc...instead of `systemctl restart dhcpcd`) would quite likely avoid this 10SEC bug.  Initial evidence points in that direction anyway.

Context: As we try to get to the bottom of this, we'll also take a step back in the coming week as facts converge here, towards establishing Internet-in-a-Box 6.5 where networking can survive a few serious hits along the way!

The reason: handheld digital library implementer/installer/operator-folk are swapping in+out their Ethernet and WiFi connections **surprisingly often** &mdash; and not just at home and at work.  But also as they travel and provision and demo such RPi3's with many gigabytes of content &mdash; downloaded/updated all along the way (and sometimes in the field too!)

georgejhunt, 2018-01-02:
> often* — and not just at home and at work. But also as they travel and

There may be some race conditions during a dhcpcd restart -- in that
stopping dhcpd, stops wpa_supplicant, which in turn tries to stop dhcpcd
via dhcpcd-run-hooks

But the first thing to work on is having multiple AP which the pi is trying
to associate with. (snippets from output2.txt)
Jan 02 01:05:29 raspberrypi dhcpcd[379]: wlan0: rebinding lease of
192.168.43.165
Jan 02 01:13:37 raspberrypi dhcpcd[1435]: wlan0: rebinding lease of
192.168.0.103

I think trying to associate to two different access points will confuse our
troubleshootiing

holta, 2018-01-02:
Stronger Confirmation of the Bug:

I moved the RPi3 & new router to the Other End Of The House, _with no connections to the Internet at all_.  The 10SEC WiFi bug is fully reproduceable [for me] as follows:
- boot a virgin (untouched) RPi3 Raspbian Lite 2017-11-29 on RPi3
- connect to router's WiFi using `raspi-config` -> "2 Network Options" -> "N2 Wi-Fi" (i.e. set SSID & passphrase)
- run `systemctl restart dhcpcd`
- connect to router's Ethernet for ~5min
- remove Ethernet cord
- "ping 192.168.13.1" (or whatever the router's IP address) confirms 10SEC WiFi malignancy pattern

**(no Internet required, now I just need to find a nearby cave that has electrical outlets ;)**

Please see the journalctl output if it helps: [journalctl-output6.txt](https://github.com/iiab/iiab/files/1598433/journalctl-output6.txt)

jvonau, 2018-01-02:
from journalctl-output4.txt
> Jan 02 09:16:59 raspberrypi systemd[1]: Starting dhcpcd on all interfaces...
Jan 02 09:16:59 raspberrypi dhcpcd[356]: dev: loaded udev
Jan 02 09:17:00 raspberrypi dhcpcd[356]: wlan0: starting wpa_supplicant
Jan 02 09:17:00 raspberrypi dhcpcd-run-hooks[420]: wlan0: starting wpa_supplicant

Ok, 356 spawned 420 to create the wifi link. 

> Jan 02 09:17:00 raspberrypi dhcpcd[356]: eth0: waiting for carrier
Jan 02 09:17:00 raspberrypi dhcpcd[356]: wlan0: waiting for carrier
Jan 02 09:17:00 raspberrypi dhcpcd[356]: wlan0: carrier acquired
Jan 02 09:17:00 raspberrypi dhcpcd[356]: DUID 00:01:00:01:21:b0:d9:e3:b8:27:eb:26:29:fe
Jan 02 09:17:00 raspberrypi dhcpcd[356]: wlan0: IAID eb:26:29:fe
Jan 02 09:17:00 raspberrypi dhcpcd[356]: wlan0: adding address fe80::a30:c77b:abdb:f0ff
Jan 02 09:17:00 raspberrypi dhcpcd[356]: wlan0: carrier lost

the link started and stopped under 420

> Jan 02 09:17:00 raspberrypi dhcpcd[356]: wlan0: deleting address fe80::a30:c77b:abdb:f0ff
Jan 02 09:17:01 raspberrypi dhcpcd[356]: wlan0: carrier acquired
Jan 02 09:17:01 raspberrypi dhcpcd[356]: wlan0: IAID eb:26:29:fe
Jan 02 09:17:01 raspberrypi dhcpcd[356]: wlan0: adding address fe80::90c5:bd6a:776d:ac91
Jan 02 09:17:01 raspberrypi dhcpcd[356]: wlan0: soliciting an IPv6 router
Jan 02 09:17:02 raspberrypi dhcpcd[356]: wlan0: rebinding lease of 192.168.13.101
Jan 02 09:17:02 raspberrypi dhcpcd[356]: wlan0: probing address 192.168.13.101/24
Jan 02 09:17:07 raspberrypi dhcpcd[356]: wlan0: leased 192.168.13.101 for 7200 seconds
Jan 02 09:17:07 raspberrypi dhcpcd[356]: wlan0: adding route to 192.168.13.0/24
Jan 02 09:17:07 raspberrypi dhcpcd[356]: wlan0: adding default route via 192.168.13.1
Jan 02 09:17:07 raspberrypi dhcpcd[356]: forked to background, child pid 529
Jan 02 09:17:07 raspberrypi systemd[1]: Started dhcpcd on all interfaces.
Jan 02 09:17:15 raspberrypi dhcpcd[529]: wlan0: no IPv6 Routers available

Link is up, I'll take 529 as what process will be used to renew the dhcp lease when needed. My question is what became of the 356 and 420 processes? 

> Jan 02 09:17:37 raspberrypi systemd[1]: Stopping dhcpcd on all interfaces...
Jan 02 09:17:37 raspberrypi dhcpcd[598]: sending signal TERM to pid 529
Jan 02 09:17:37 raspberrypi dhcpcd[598]: waiting for pid 529 to exit

hand restarting dhcpcd now killing 529

> Jan 02 09:17:37 raspberrypi dhcpcd[529]: received SIGTERM, stopping
Jan 02 09:17:37 raspberrypi dhcpcd[529]: eth0: removing interface
Jan 02 09:17:37 raspberrypi dhcpcd[529]: wlan0: removing interface
Jan 02 09:17:37 raspberrypi dhcpcd[529]: dhcpcd exited
Jan 02 09:17:37 raspberrypi dhcpcd[598]: sending signal TERM to pid 529
Jan 02 09:17:37 raspberrypi dhcpcd[598]: waiting for pid 529 to exit
Jan 02 09:17:37 raspberrypi systemd[1]: Stopped dhcpcd on all interfaces.

now restarting but what about 356 and 420?

> Jan 02 09:17:38 raspberrypi systemd[1]: Starting dhcpcd on all interfaces...
Jan 02 09:17:38 raspberrypi dhcpcd[608]: dev: loaded udev
Jan 02 09:17:38 raspberrypi dhcpcd[608]: wlan0: starting wpa_supplicant
Jan 02 09:17:38 raspberrypi dhcpcd-run-hooks[624]: wlan0: starting wpa_supplicant
Jan 02 09:17:38 raspberrypi dhcpcd[608]: Too few arguments.
Jan 02 09:17:38 raspberrypi dhcpcd[608]: Too few arguments.

That was not present when dhcpcd first started above.

> Jan 02 09:17:38 raspberrypi dhcpcd[608]: eth0: waiting for carrier
Jan 02 09:17:38 raspberrypi dhcpcd[608]: wlan0: waiting for carrier
Jan 02 09:17:39 raspberrypi dhcpcd[608]: wlan0: carrier acquired
Jan 02 09:17:39 raspberrypi dhcpcd[608]: DUID 00:01:00:01:21:b0:d9:e3:b8:27:eb:26:29:fe
Jan 02 09:17:39 raspberrypi dhcpcd[608]: wlan0: IAID eb:26:29:fe
Jan 02 09:17:39 raspberrypi dhcpcd[608]: wlan0: adding address fe80::90c5:bd6a:776d:ac91
Jan 02 09:17:39 raspberrypi dhcpcd[608]: wlan0: rebinding lease of 192.168.13.101
Jan 02 09:17:39 raspberrypi dhcpcd[608]: wlan0: leased 192.168.13.101 for 7200 seconds

the lease looks to be valid

[... 15 lines cut ...]
Jan 02 09:18:25 raspberrypi dhcpcd[724]: eth0: rebinding lease of 192.168.0.133
Jan 02 09:18:25 raspberrypi dhcpcd[724]: eth0: probing address 192.168.0.133/24
Jan 02 09:18:29 raspberrypi dhcpcd[724]: eth0: leased 192.168.0.133 for infinity

different dhcp servers/APs between wired/wireless correct?
 
> Jan 02 09:18:29 raspberrypi dhcpcd[724]: eth0: adding route to 192.168.0.0/24
Jan 02 09:18:29 raspberrypi dhcpcd[724]: eth0: adding default route via 192.168.0.1
Jan 02 09:18:37 raspberrypi dhcpcd[724]: eth0: no IPv6 Routers available
Jan 02 09:23:37 raspberrypi dhcpcd[724]: eth0: carrier lost
Jan 02 09:23:37 raspberrypi dhcpcd[724]: eth0: deleting address fe80::4ced:af8b:92d1:e557
Jan 02 09:23:37 raspberrypi dhcpcd[724]: eth0: deleting default route via 192.168.0.1
Jan 02 09:23:37 raspberrypi dhcpcd[724]: eth0: deleting route to 192.168.0.0/24

> Observation 2: Method 2 of inducing this overall bug often seems to work when inserting a live Ethernet-to-Internet cord onto the RPi3 for only 3 minutes (after "systemctl restart dhcpcd" was run) with perhaps 80% certainty) — whereas inserting the live Ethernet-to-Internet cord for a full 5 minutes seems to induce the syndrome with perhaps 90% certainty. Finally, inserting the Ethernet cord for just 2 minutes does not trigger the syndrome usually...

[... cut ...] o wired interface but associated to the AP with a valid lease/ip would be to systemctl stop dhcpcd and note if wpa_supplicant was killed off, even on machines that don't have this problem as a data point 

Is the SSID of the AP hidden? How long before the wifi goes away? Thinking dhcpcd may not be seeing the hotplug event when the wired interface is connected. If you don't restart dhcpcd at all, when you plug in the wire does eth0 become active? With no other changes does a reboot bring eth0 up?

quozl, 2018-01-02:
@jvonau, `systemctl restart dhcpcd` _does_ terminate the single `dhcpcd` process and start a new one, and _does not_ terminate or restart the `wpa_supplicant` process, though an `strace` _does_ show a message on a control socket.

@holta, suggest you test with later `dhcpcd` from Debian stable;

```
wget http://ftp.us.debian.org/debian/pool/main/d/dhcpcd5/dhcpcd5_6.10.1-1_armhf.deb
dpkg -i dhcpcd5_6.10.1-1_armhf.deb
```

Details;
* dhcpcd-6.7.1 was [released in 2015-01-30](https://github.com/rsmarples/dhcpcd/releases/tag/dhcpcd-6.7.1),
* Raspbian deploys dhcpcd-6.7.1 with Raspberry Pi special changes, released in 2015-10-17,
* dhcpcd-7.0.0 was [released yesterday](https://roy.marples.name/),
* dhcpcd-6.10.1 is now used in Debian stable; Raspbian is a fair way behind,
* between 6.7.1 and 6.10.1 there were 474 commits,
* dhcpcd-6.10.1 from Debian stable does install cleanly on Raspbian, and seems to work fine for me, including the ethernet cable transitions, so it may be worth a try to see if the problem is fixed in a later version,
* if updating the package works, then raising the problem with Raspbian may help.

[26 more replies]

closed 2021-04-23 by holta