iiab/iiab #641: Document new OER2GO/RACHEL commands in FAQ.IIAB.IO

holta, 2017-12-30:
Commands documented in http://minutes.iiab.io under Dec 28, 2017 Item 2. g.

We're hoping to migrate this to http://FAQ.IIAB.IO soon, per:
- #630 "OER2GO (RACHEL) download support coming to Admin Console"
- iiab/iiab-admin-console#50 "OER2GO (RACHEL) support part I"

tim-moody, 2018-05-17:
the documentation is in the help

holta, 2018-05-17:
Will OER2GO command-line tools continue to be supported going forward?

(If so, where's the best place to get usage details, so I can mention these in 6.5 release announcements?)

tim-moody, 2018-05-18:
draft update to docs.

OER2GO cli is still there, but let's not document for now.

holta, 2018-05-18:
Please see @tim-moody's new section explaining how to download/install RACHEL/OER2GO modules:
https://github.com/iiab/iiab/wiki/IIAB-Installation#oer2go-rachel-modules

As summarized in Paragraph 3 of IIAB 6.5's draft release notes:
https://github.com/iiab/iiab/wiki/IIAB-6.5-Release-Notes

We should add a quick section on the amazing new http://box/admin -> Install Content -> Remove Files (now part of IIAB 6.5 RC5, as a result of https://github.com/iiab/iiab-admin-console/pull/67)

holta, 2018-05-19:
> We should add a quick section on the amazing new http://box/admin -> Install Content -> Remove Files (now part of IIAB 6.5 RC5, as a result of iiab/iiab-admin-console#67)

Done: https://github.com/iiab/iiab/wiki/IIAB-Installation#remove-content

closed 2018-05-19 by holta