iiab/iiab #642: WiFi-installed RPi3's add erroneous br0 route when WiFi drops

holta, 2017-12-30:
This has now happened on 2 separate RPi3's with IIAB 6.5.  After the WiFi-to-Internet connection recovers (on its own, or with "systemctl restart dhcpcd") the routing table blocks access to the Internet:

>root@box:~# ip r
**default dev br0 proto kernel scope link src 169.254.38.92 metric 204**
default via 192.168.0.1 dev wlan0 src 192.168.0.102 metric 303
169.254.0.0/16 dev br0 proto kernel scope link src 169.254.38.92 metric 204
192.168.0.0/24 dev wlan0 proto kernel scope link src 192.168.0.102 metric 303

The short-term workaround is to remove the problematic route bolded above, by running: (for example)
```
ip r del default dev br0 proto kernel scope link src 169.254.38.92 metric 204
```
Thereafter the Internet in once again pingable, and the routing table appears as follows:

> root@box:~# ip r
default via 192.168.0.1 dev wlan0 src 192.168.0.102 metric 303
169.254.0.0/16 dev br0 proto kernel scope link src 169.254.38.92 metric 204
192.168.0.0/24 dev wlan0 proto kernel scope link src 192.168.0.102 metric 303

NOTE: this problem/workaround seems to have no bearing whatsoever on bug #638 "ping works for 10sec, fails for 18-45sec, then repeats (WiFi-installed IIAB/RPi)".  The problem described in #638 (whatever's causing it) seems to:
- remain absent regardless of the above workaround, if #638 was NOT happening a priori ("shortly after boot")
- occur even after the above workaround is applied, on RPi3's that presumably already had the #638 problem ("long after boot")

holta, 2017-12-30:
The line with the lowest number "metric" apparently wins, as the default route.

Question is: how should we should automate this essential workaround &mdash; that allows the default routes like 192.168.0.1 (metric 303) within "ip r" to rescover &mdash; again taking precedence over br0's (metric 204).

So if deleting br0's route really is the best way forward, it could be automated as follows:
```
ip r del `ip r | grep "default dev br0")`
```
...equivalent to:
```
ip r del $(ip r | grep "default dev br0"))
```
Observation: sometimes the above errant route deletes itself "properly", apparently when WiFi's radio signal comes and goes very briefly?

tim-moody, 2017-12-30:
if wlan0 is a gateway it should not be in br0 which is always the lan

holta, 2017-12-30:
> if wlan0 is a gateway it should not be in br0 which is always the lan

Are you talking about 169.254.* within br0 on 10.8.0.30 below?

> root@box:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:87:ed:b1 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:d2:b8:e4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.102/24 brd 192.168.0.255 scope global wlan0
       valid_lft forever preferred_lft forever
4: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN group default qlen 1000
    link/ether a2:b5:49:96:ef:78 brd ff:ff:ff:ff:ff:ff
    inet 169.254.2.87/16 brd 169.254.255.255 scope global br0
       valid_lft forever preferred_lft forever

FWIW there is no LAN on these current machines where eth0 & wlan0 (should) both lead to the Internet during WiFi installation (certainly when PR #635 or its successor is confirmed...)

After installation when "iiab-hotspot-on" has been run, there is of course a hotspot-as-LAN (and br0 presumably, if nec?)

(And if "iiab-hotspot-off" is subsequently run &mdash; when a remote operator wants to connect their IIAB to the WiFi/Internet for updates &mdash; the situation is again reversed, with no LAN effectively.)

holta, 2017-12-30:
See #638 for a far cleaner way to solve this &mdash; i.e. @georgejhunt's [suggestion](https://github.com/iiab/iiab/issues/638#issuecomment-354564289) to add "denyinterfaces br0" to /etc/dhcpcd.conf and its [confirmed](https://github.com/iiab/iiab/issues/638#issuecomment-354567517) [results](https://github.com/iiab/iiab/issues/638#issuecomment-354569464).

holta, 2017-12-30:
**Question:** should "iiab-hotspot-on" remove "denyinterfaces br0" from /etc/hdcpcd.conf if br0 is really needed on RPi somehow?

**Conversely:** would "iiab-hotspot-off" presumably add "denyinterfaces br0" back into /etc/dhcpcd.conf following by another "systemctl restart networking; systemctl restart dhcpcd" ?  (So remote operators can get updates using WiFi-to-Internet, etc!)

As @georgejhunt & @tim-moody have [alluded to](https://github.com/iiab/iiab/issues/638#issuecomment-354568757), trying to get these right!

jvonau, 2017-12-31:
> 4: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN group default qlen 1000
link/ether a2:b5:49:96:ef:78 brd ff:ff:ff:ff:ff:ff
inet 169.254.2.87/16 brd 169.254.255.255 scope global br0
valid_lft forever preferred_lft forever

If br0 is active there should be the 172.18.96.1/19 address present but is not, leading to the fallback use of link-local 169.254.x.x when dhcpcd tried to request an ip address and fails to do so for br0.

holta, 2017-12-31:
Here is an example (pasted in from /var/log/syslog) of br0 adding in the problematic 169.254.x.y to "ip a" and "ip r" &mdash; in case it sheds light as to why this is unfortunately happening sporadically (and what is spontaneously causing it?) &mdash; hence triggering WiFi failures like #638:
```
Dec 31 12:26:04 localhost dhcpcd[10027]: sending signal TERM to pid 9124
Dec 31 12:26:04 localhost dhcpcd[10027]: waiting for pid 9124 to exit
Dec 31 12:26:04 localhost dhcpcd[9124]: received SIGTERM, stopping
Dec 31 12:26:04 localhost dhcpcd[9124]: eth0: removing interface
Dec 31 12:26:04 localhost dhcpcd[9124]: br0: removing interface
Dec 31 12:26:04 localhost dhcpcd[9124]: wlan0: removing interface
Dec 31 12:26:04 localhost dhcpcd[9124]: dhcpcd exited
Dec 31 12:26:05 localhost dhcpcd[10035]: dev: loaded udev
Dec 31 12:26:05 localhost dhcpcd-run-hooks[10054]: wlan0: starting wpa_supplicant
Dec 31 12:26:05 localhost vmunix: [92548.988717] brcmfmac: power management disabled
Dec 31 12:26:05 localhost dhcpcd[10035]: DUID 00:01:00:01:21:b0:d9:dc:b8:27:eb:d2:b8:e4
Dec 31 12:26:05 localhost dhcpcd[10035]: br0: IAID b7:5a:c5:0f
Dec 31 12:26:05 localhost dhcpcd[10035]: br0: adding address fe80::eec9:547a:bad2:1bc9
Dec 31 12:26:05 localhost dhcpcd[10035]: if_addaddress6: Permission denied
Dec 31 12:26:05 localhost dhcpcd[10035]: eth0: waiting for carrier
Dec 31 12:26:05 localhost dhcpcd[10035]: wlan0: waiting for carrier
Dec 31 12:26:06 localhost dhcpcd[10035]: wlan0: carrier acquired
Dec 31 12:26:06 localhost dhcpcd[10035]: wlan0: IAID eb:d2:b8:e4
Dec 31 12:26:06 localhost dhcpcd[10035]: wlan0: adding address fe80::c666:bdb8:6f8:df62
Dec 31 12:26:06 localhost dhcpcd[10035]: if_addaddress6: Permission denied
Dec 31 12:26:06 localhost dhcpcd[10035]: br0: soliciting a DHCP lease
Dec 31 12:26:06 localhost dhcpcd[10035]: br0: soliciting an IPv6 router
Dec 31 12:26:06 localhost dhcpcd[10035]: wlan0: rebinding lease of 192.168.0.101
Dec 31 12:26:06 localhost dhcpcd[10035]: wlan0: leased 192.168.0.101 for 7200 seconds
Dec 31 12:26:06 localhost dhcpcd[10035]: wlan0: adding route to 192.168.0.0/24
Dec 31 12:26:06 localhost dhcpcd[10035]: wlan0: adding default route via 192.168.0.1
Dec 31 12:26:07 localhost dhcpcd[10035]: forked to background, child pid 10162
Dec 31 12:26:07 localhost dhcpcd[10162]: wlan0: soliciting an IPv6 router
Dec 31 12:26:11 localhost dhcpcd[10162]: br0: using IPv4LL address 169.254.186.92
Dec 31 12:26:11 localhost dhcpcd[10162]: br0: adding route to 169.254.0.0/16
```