iiab/iiab #650: Use solr+shine for local search

traverseda, 2018-01-11:
Normally I wouldn't open a ticket for something like this, but shine is poorly document and hard to find, despite it being nearly perfect for this use-case.

https://github.com/ukwa/shine/tree/master/shine

https://www.webarchive.org.uk/shine/search?facet.sort=index&tab=results

holta, 2018-07-18:
@traverseda am moving this Solr question to our [IIAB 6.7](https://github.com/iiab/iiab/milestone/4) milestone as our [IIAB 6.6](https://github.com/iiab/iiab/wiki/IIAB-6.6-Release-Notes) release is now imminent in a few very short weeks, target July 31.

Could you perhaps help make a prototype happen for IIAB 6.7 in coming months?  (If so, let us know what guidance you need making integration with Internet-in-a-Box happen!)

traverseda, 2018-07-18:
I'm afraid I can't, I'm booked solid for the next 4 months.

holta, 2018-11-04:
> I'm afraid I can't, I'm booked solid for the next 4 months.

FYI @traverseda our target for [IIAB 6.7](http://download.iiab.io/6.7/) release is sometime in December, just in case you have a chance to give this a swing and/or describe the task for others!

traverseda, 2018-11-04:
I've been poking around at things related to this for a while. The basic idea is that google is the definitive interface for most people on the web, so we should include something similar. So how best to do this?

We're exposing everything over the web, and web-scraping search spiders are a mature technology. So let's just do that. Google's pagerank algorithm is simple enough to implement, and produces good enough results. Although I'm sure they've moved on to more complex algorithms by now, we don't need to deal with SEO spam so we can rely on older simpler algorithms and still get good results.

So let's include a search engine. The specifics are up in the air, solr seems like a good technology to build on. Xapian would be another good choice.

The internet archive's shine interface looks like it would do most of what we'd want, we we tied it all together.

traverseda, 2018-11-04:
The problem with kiwix+xapian is that you can only search one zim file at a time, it's not merging those indexes. Xapian is probably a better fit than solr, the only reason I was suggesting it was because we could use the internet-archives shine directly.

I think it's going to be hard to rely on upstream search indices, especially if we want to use something like pagerank, which looking at kiwix's results, I think we very much do.

tim-moody, 2019-09-13:
A couple of random comments:

1. IIAB does not really originate any content, so has no control over indexing at source.
2. Some content comes pre-indexed: Kiwix now has an index inside each zim; kalite has builtin search; there was an attempt to add search to RACHEL, but not sure how successful it has been.
3. No two curations are identical, so a global index would have to be generated for each deployment, but individual content packs could be pre-indexed.
4. Zim indices are rather large, so I would not duplicate them.
5. Probably a search result merge function is needed so that results from various indexing schemes can be merged. The problem of how to compare the rank of disparate results is not trivial, however. And the merge can be slow.
6. There is often a need to exclude some content from a given search. For example, Kiwix has cross zim search (which is accessible via the IIAB home page), but some types of zims, especially wiktionary, completely dominate the results, making the results not very useful. I have asked Kiwix for search groups. An advanced search input page could allow check boxes for which sources to include.
7. Kiwix returns formatted html along with headers and footers, which makes aggregation or even integration of the results into another page difficult. I believe they are working on a search api to return raw data.

traverseda, 2019-09-13:
I'll try to get some real benchmarking done soon. I have an odroid-xu4, which is probably the closest analogue to the kind of platforms we want to be running this on. It's not going to give great results, but it should at least help.

Initial results on a powerful laptop, indexing 22 pages on a remote web-server (increased latency compared to a real test) show 118980 kbytes used. That's 118.98mb. I won't bother with the CPU usage because that's not a fair comparison. The memory usage should not go up with more pages, unless you're using more workers to get it done faster.

>IIAB does not really originate any content, so has no control over indexing at source.

> Zim indices are rather large, so I would not duplicate them.

> individual content packs could be pre-indexed.

Are there resources to originate content? That would make a lot of this technical infrastructure quite a bit easier.

>There is often a need to exclude some content from a given search. For example, Kiwix has cross zim search

Indexing everything ourselves gives us a path to solving that, as we can better control the search ranking function. We could even do something like google and only return *one* results from that kind of search, putting it in an infobox near the top. Since Wiktionary results are probably pretty hit-or-miss.

traverseda, 2019-09-13:
I use a pretty heavy summarizer in order to extract a minimal summary of indexed articles, without that it takes 48.084mb. That's more in line with something that could run comfortable on a rpi.

Do you have a more concrete idea of the memory budget, so I can try to keep things under it as I experiment?

holta, 2019-09-13:
> Do you have a more concrete idea of the memory budget, so I can try to keep things under it as I experiment?

During indexing?

FYI 1GB RAM Raspberry Pi 4 is the most common platform for IIAB today, though 2GB models are making their way into the field in some cases.  And in the medical context especially, 512MB Raspberry Pi Zero W's are used &mdash; after the disk image is prepared on a Raspberry Pi 4 typically.

traverseda, 2019-09-13:
Yeah, during indexing. A quick searching test showed 36.632mb used, although the index is still tiny. Searching should be comparable to the existing kiwix implementation, as xapian is doing all the heavy lifting regardless, so I didn't bother testing it.

I'll need to see how xapian deals with bigger indexing, but on-device indexing look reasonable so far.

holta, 2020-01-30:
Please see new/related ticket #2218 about "OpenSearch", thanks to @mitra42.

traverseda, 2020-01-30:
I'm having a hard time using xapian, investigating meilisearch instead.

holta, 2020-01-31:
Please see new/related ticket #2219 ("Search engine API proposal") thanks to @traverseda.

closed 2020-05-31 by traverseda