iiab/iiab #654: usbmount to http://box/usb does not work with modern exFAT sticks

holta, 2018-01-16:
Ref #329

holta, 2018-01-29:
Moving to IIAB 6.5 for discussion.  This is starting to affect teachers in Mexico who can no longer use IIAB to display/share their content, as exFAT-formatted 32GB memory sticks are becoming mainstream.

holta, 2018-02-04:
> [affects] teachers in Mexico who can no longer use IIAB to display/share their [USB stick] content, as exFAT-formatted 32+ GB memory sticks are becoming mainstream.

@georgejhunt are you able to confirm unfortunate behavior?

And/or say if suggest a path forward if so?

holta, 2018-02-08:
Thursday Community/team call excerpt from http://minutes.iiab.io :
- @tim-moody wonders if security of exFAT executables is related to these new store-bought 32+ GB sticks failing to display teacher content?
- @georgejhunt / @floydianslips / others suspect we need to tweak automounting of exfat filesystems &mdash; i.e. http://box/usb functionality might / likely start working as soon as we get exfat-formatted sticks to automount (manually mounting them is not a problem today, since IIAB introduced esFAT support late in 2017)
- @jvonau suggests we might need to add "exfat" to usbmount's own list of filesystems it's supposed to automount
- @georgejhunt / others ask that we verify this we **store-bought exFAT sticks** to be sure (not just our own sticks that we've formatted to exFAT)

holta, 2018-02-08:
Is it possible we just need to add "exfat" to the following line in /etc/usbmount/usbmount.conf ?
```
FILESYSTEMS="vfat ext2 ext3 ext4 hfsplus"
```
_How is this best achieved within [roles/usb-lib](https://github.com/iiab/iiab/tree/master/roles/usb-lib) presumably?_

Note To Self: documentation [roles/usb-lib/README.rst](https://github.com/iiab/iiab/blob/master/roles/usb-lib/README.rst) is very stale and needs to be replaced with http://wiki.laptop.org/go/IIAB/FAQ#Can_teachers_display_their_own_content.3F or similar after we finalize/merge this fix.

holta, 2018-02-08:
Worrisome blog post from someone facing this very same "how to automount exfat" obstacle:
https://unix.stackexchange.com/questions/403591/using-usbmount-with-exfat-drives

This other blog post might possibly be more reassuring...contradicting the above :)
http://malatsblog.blogspot.com/2014/04/usbmount-exfat.html

holta, 2018-02-11:
Sadly adding "exfat ntfs fuseblk" into this line within /etc/usbmount/usbmount.conf does not work:
```
FILESYSTEMS="vfat ext2 ext3 ext4 hfsplus exfat ntfs fuseblk"
```
The error (even after reboot) is:
```
root@box:~# ls -l /media/usb0
ls: cannot access '/media/usb0': Transport endpoint is not connected
```
Confoundingly, even while automounting fails, manually mounting exFAT and NTFS drives *does* in fact work mount the stick's contents &mdash; and also to showcase the USB stick's teacher content (to students) via http://box/usb &mdash; e.g. after typing the following:
```
root@box:~# umount /media/usb0
root@box:~# mount /dev/sda1 /media/usb0
```

jvonau, 2018-02-11:
> NTFS drives *does* in fact work mount the stick's contents — and also to

Time to retire the 10 year old hack that was intended for headless,
gui-less servers and look for a proper integration with the desktop you
want to run with?

holta, 2018-02-11:
> Time to retire the 10 year old hack that was intended for headless,

I need to test headless-mounting of exfat and ntfs as well, good idea.

Certainly exFAT stick and NTFS drive both show false positive confirmation that they've automounted on Raspbian and Ubuntu 16.04:
```
root@box:/media# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    1  63.7G  0 disk
`-sda1        8:1    1  63.7G  0 part /media/usb0
sdb           8:16   0   1.8T  0 disk
`-sdb1        8:17   0   1.8T  0 part /media/usb1
...
root@box:/media# blkid
...
/dev/sda1: UUID="46C3-5AB7" TYPE="exfat" PARTUUID="966a3fdf-01"
...
/dev/sdb1: LABEL="My Passport" UUID="F474B7AA74B76DCC" TYPE="ntfs" PARTUUID="0005f107-01"
```
But the contents of both remain inaccessible (until manual mounting of either stick/drive...)
```
root@box:/media# ls /media/usb0
ls: cannot access '/media/usb0': Transport endpoint is not connected
root@box:/media# ls /media/usb1
ls: cannot access '/media/usb1': Transport endpoint is not connected
```
(Whereas FAT32 drives/sticks indeed do work...)

holta, 2018-02-12:
> I need to test headless-mounting of exfat and ntfs as well, good idea.

Regrettably the same result occurs on headless Raspbian Lite, running IIAB 6.5/master:
- FAT32 and ext4 drives automount successfully, whereas...
- exFAT and NTFS again **appear** to automount (according to `lsblk`) but then fail to show any drive contents...
```
root@box:~# ls /media/usb0
ls: cannot access '/media/usb0': Transport endpoint is not connected
root@box:~# ls /media/usb1
ls: cannot access '/media/usb1': Transport endpoint is not connected
```
Even despite /etc/usbmount/usbmount.conf containing:
```
FILESYSTEMS="vfat ext2 ext3 ext4 hfsplus exfat ntfs fuseblk"
```
_@jvonau suggests [polkit](https://en.wikipedia.org/wiki/Polkit) (formerly PolicyKit) might be affecting fine-grained user-level permissions, causing this failure?  Certainly root can mount exfat and ntfs drives manually!_

_Likewise he recommends examining detailed logs in /var/log/* (journalctl might help) so the precise mechanics of exfat/ntfs automounting failures can be compared side-by-side with the mechanics of vfat/ext4 automount successes._

holta, 2018-02-14:
2015-centric thread reviewing usbmount's failing to automount ntfs: (thanks @jvonau)
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=774149

One alternative approach:
https://ddumont.wordpress.com/2016/04/24/automount-usb-devices-with-systemd/

georgejhunt, 2018-02-14:
This reference seems to do he trick: https://raspberrypi.stackexchange.com/questions/41959/automount-various-usb-stick-file-systems-on-jessie-lite. But more testing before a PR

georgejhunt, 2018-02-15:
I can confirm that ntfs formatted usb sticks also work.

holta, 2018-02-15:
> I can confirm that ntfs formatted usb sticks also work.

Incredible news.

I've now documented that in our [IIAB 6.5 DRAFT Release Notes](https://github.com/iiab/iiab/wiki/IIAB-6.5-Release-Notes).

closed 2018-02-15 by holta