iiab/iiab #658: Add a welcome message for KA Lite

benjaoming, 2018-01-24:
### Expected behavior and actual behavior.

Hi friends,

Thanks for upgrading to 0.17.4!

In this release, we added a new configuration `KALITE_WELCOME_MESSAGE`. There are also other small fixes, see the [release notes](http://ka-lite.readthedocs.io/en/latest/installguide/release_notes.html)

The motivation for this, was to be able to guide users to better understand where to go for support. We frequently get RACHEL questions in the KA Lite channels, and I'm sure you get KA Lite questions or bug reports in your channels, as well? So this might be an option to advertise some IIAB channels..

In your https://github.com/iiab/iiab/blob/master/roles/kalite/templates/settings.py.j2 , just add something like this (our own text displayed below):

```
# Default welcome message
KALITE_WELCOME_MESSAGE = """
<h2>Need help?</h2>
<p>KA Lite is community-driven and relies on help and experience which you can both share and receive in our community.</p>
<br>
<h4>Offline help: </h4>
<ul>
  <li>Use the "Docs" button in the top menu to get help.</li>
</ul>
<h4>Online help: </h4>
<ul>
  <li>Share your implementation of KA Lite on the map and find other KA Lite users in your country
  <a href='https://learningequality.org/ka-lite/map/' target='_blank'>https://learningequality.org/ka-lite/map/</a>
  (Click on "Add your story!")</li>
  <li>Give and receive KA Lite support in the Learning Equality community 
  <a href='http://community.learningequality.org/' target='_blank'>http://community.learningequality.org/</a></li>
  <li>Help program KA Lite at GitHub 
  <a href='http://github.com/learningequality/ka-lite/' target='_blank'>http://github.com/learningequality/ka-lite/</a></li>
  <li>Sign up for emails about new releases <a href='https://groups.google.com/a/learningequality.org/forum/#!forum/dev' target='_blank'>
  https://groups.google.com/a/learningequality.org/forum/#!forum/dev</a></li>
</ul>
"""
```

### Steps to reproduce the problem.

n/a

### Specifications like the version of the IIAB, operating system version, or hardware details.

n/a

benjaoming, 2018-01-24:
Docs:

> `KALITE_WELCOME_MESSAGE = <str>` HTML code to be rendered to admin users on their first login. Default text links to offline documentation and online support forums.

benjaoming, 2018-01-25:
This is how the welcome message displays btw...

[image]

benjaoming, 2018-01-25:
...We display the full URL in case people are offline and need to write it down for later.

holta, 2018-11-04:
Ref: #1285

jvonau, 2025-01-27:
on the way out

closed 2025-01-27 by holta